espressif/esp-idf #9274: ESP32S3 WINBOND W25Q64JVTBIQ: A fatal error occurred: MD5 of file does not match data in flash!

TGiles1998, 2022-07-03:
ESP32S3 custom hardware error when trying to flash W25Q64JVTBIQ.
IDF V5.0

idf.py -p /dev/cu.usbserial-1450 flash monitor
Executing action: flash
Running ninja in directory /Users/thomasgiles/esp/esp-idf/examples/get-started/hello_world/build
Executing "ninja flash"...
[1/5] cd /Users/thomasgiles/esp/esp-id...rted/hello_world/build/hello_world.bin
hello_world.bin binary size 0x2ea80 bytes. Smallest app partition is 0x100000 bytes. 0xd1580 bytes (82%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /Users/thomasgiles/esp/esp-idf/examples/get-started/hello_world/build/bootloader/esp-idf/esptool_py && /Users/thomasgiles/.espressif/python_env/idf5.0_py3.10_env/bin/python /Users/thomasgiles/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/thomasgiles/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x4fb0 bytes. 0x3050 bytes (38%) free.
[2/3] cd /Users/thomasgiles/esp/esp-id...nents/esptool_py/run_serial_tool.cmake
esptool esp32s3 -p /dev/cu.usbserial-1450 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.1
Serial port /dev/cu.usbserial-1450
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:f7:49:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x0003efff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20400 bytes to 12989...
Writing at 0x00000000... (100 %)
Wrote 20400 bytes (12989 compressed) at 0x00000000 in 0.4 seconds (effective 375.0 kbit/s)...
File  md5: e3b50cb3d4a5c41966510952cc170fdf
Flash md5: 5500ad039f388ed660d26deec84cb24a
MD5 of 0xFF is 5500ad039f388ed660d26deec84cb24a

A fatal error occurred: MD5 of file does not match data in flash!
CMake Error at run_serial_tool.cmake:55 (message):
  
  /Users/thomasgiles/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/Users/thomasgiles/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed

FAILED: CMakeFiles/flash /Users/thomasgiles/esp/esp-idf/examples/get-started/hello_world/build/CMakeFiles/flash 
cd /Users/thomasgiles/esp/esp-idf/components/esptool_py && /Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/cmake/data/CMake.app/Contents/bin/cmake -D IDF_PATH=/Users/thomasgiles/esp/esp-idf -D "SERIAL_TOOL=/Users/thomasgiles/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/Users/thomasgiles/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/thomasgiles/esp/esp-idf/examples/get-started/hello_world/build -P /Users/thomasgiles/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

TGiles1998, 2022-07-03:
Esptool Flash_id 

esptool.py -p /dev/cu.usbserial-1450 flash_id
esptool.py v4.1
Serial port /dev/cu.usbserial-1450
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:f7:49:44
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4017
Detected flash size: 8MB
Hard resetting via RTS pin...

moefear85, 2022-07-03:
From my own experience:
either the board is not getting enough power, or the flash has worn out and you need to use a new chip, or a transient usb error occured that is causing corruption in the usb data.

The latter happens to me relatively often infact, although I use long cables and alot of daisy chained hubs and connected peripherals. Even though it "successfully" flashes, I have to reset my PC then reflash again to get the MD5 error to disappear.

moefear85, 2022-07-12:
@TGiles1998 

Are you sure you configured esp-idf correctly? Also, what flash mode did you use? it's best to go with the lowest settings (use 20MHz and try DOUT/DIO/QOUT/QIO). Are you sure the CS line is correctly toggled? I see the chip supports QIO mode.

Specifically, the error you are getting is typical of dead flash. In other words, whatever is wrong, it is making the chip appear like dead flash. The manufacturer and size is getting detected correctly, which means the wiring is probably correct. But perhaps the traces are too long and not enough isolation is present. Keep in mind a standard module has a metal case that shields the components, including flash, from EMI. But you are using 80MHz which is very high, it's not clear if the wiring can handle that. **please try 20MHz first**.

TGiles1998, 2022-07-12:
@moefear85 Hi, I will give the different flash mode a go, I inherently didn't change the menuconfig so hopefully that works to thanks!

TGiles1998, 2022-07-26:
@moefear85 Hi, This worked flawlessly, thank you so much! Much appreciated.

closed 2022-07-26 by TGiles1998