espressif/esp-idf #9277: LVGL Font corruption on SSD1306 display

VikOlliver, 2022-07-03:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-3654-gc2ccc383da
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
Default UNSCII fonts are displayed half height when rendered using the stock example code on a 128x32 SSD1306 monochrome OLED display.

Problem is exhibited by other monospace fonts, but the stock examples are probably better for illustrating the bug.

### Expected Behavior
UNSCII 8 font displayed 8 pixels high, UNSCII 16 font displayed 16 pixels high

### Actual Behavior
UNSCII 8 font displayed 4 pixels high and completely illegible.
UNSCII 16 font displayed 8 pixels high, 16 pixels wide.

### Steps to reproduce

1. Set appropriate LVGL UNSCII font in idf.py menuconfig
2. build/flash esp-idf/examples/peripherals/lcd/i2c_oled

// If possible, attach a picture of your setup/wiring here.
Unable to attach. Github reports "something has gone seriously wrong" when attempting to attach a 294KiB JPEG

### Code to reproduce this issue
Stock example.

## Debug Logs
No errors logged.
I (388) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (398) example: Initialize I2C bus
I (408) example: Install panel IO
I (408) example: Install SSD1306 panel driver
I (508) example: Initialize LVGL library
I (508) example: Register display driver to LVGL
I (508) example: Install LVGL tick timer
I (508) example: Display LVGL Scroll Text

VikOlliver, 2022-07-04:
[image]

OK, managed to upload images of the corrupt and squashed font on the OLED display by using Firefox in debug mode :)

VikOlliver, 2022-07-04:
Playing with this a bit more, it looks like every other line is missed off everything. Creating a 128x32 text area only puts characters on the top of the screen even when the content scrolls. Expanding it to a 128x64 text box will make it fill the whole screen (as you might see from the pictures above, it's definitely a 128x32 display).

andershedberg, 2022-07-19:
We see some similar behavior as well when we tried out the i2c bus and choose this display we had used in a previous Raspberry Pie project - https://www.adafruit.com/product/3527 It also seems we can not use 400kHz i2c clock, but even when the speed is like 10kHz the displayed image is "strange")

Is it related to https://github.com/espressif/esp-idf/pull/9209 ?
(there are also a lot of other i2c issues when searching for issues)

yuanyihan, 2022-11-06:
Hi，you can add some code in init（）：
```
#if SSD1306_LCD_DISP_IS_128X32 == 1
    esp_lcd_panel_io_tx_param(io_handle, 0xa8, (uint8_t[1]){0x1f}, 1); // 0x3F for 128x64, 0x1F for 128x32
    esp_lcd_panel_io_tx_param(io_handle, 0xda, (uint8_t[1]){0x02}, 1); // 0x12 for 128x64, 0x02 for 128x32
#else
    esp_lcd_panel_io_tx_param(io_handle, 0xa8, (uint8_t[1]){0x3f}, 1); // 0x3F for 128x64, 0x1F for 128x32
    esp_lcd_panel_io_tx_param(io_handle, 0xda, (uint8_t[1]){0x12}, 1); // 0x12 for 128x64, 0x02 for 128x32
#endif
```
between `    ESP_ERROR_CHECK(esp_lcd_panel_init(panel_handle));` and `ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, true));`  in https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/i2c_oled/main/i2c_oled_example_main.c#L126

befor:
[image]

after:
[image]

mbelt, 2022-11-07:
`esp_lcd_panel_init(...)` only does the bare minimum setup. 
Set Addressing Mode`0x20 0x00`
Enable Charge Pump `0x8D, 0x14`
Turn On `0xAF`. 

The datasheet describes a longer set of initialization steps. 
Set MUX Ratio `0xA8, 0x3F`
Set Addressing Mode`0x20 0x00` (Not in the datasheet init flowchart, but necessary)
Set Display Offset `0xD2, 0x00`
Set Display Start Line `0x40, 0x00`
Set Segment Re-Map `0xA0, x0A1`
Set COM Output Scan Direction `0xC0`
Set COM Pins Hardware Configuration `0xDA, 0x12` for 128x64 or `0xDA, 0x021` for 128x32px
Set Contrast Control `0x8a, 0x7F`
Disable Display On `0xA4`
Set Normal Display `0xA6`
Set Osc Frequency `0xD5, 0x80`
Enable Charge Pump Regulator `0x8D, 0x14`
Display On `0xAF`

yuanyihan, 2022-11-07:
@mbelt 
THINKS,It‘s very useful
Some minor mistakes:
- `Set Display Offset 0xD2, 0x00` is `Set Display Offset 0xD3, 0x00`
- `Set Segment Re-Map 0xA0, 0xA1` means `Set Segment Re-Map 0xA0 for normal or 0xA1 for rotation180` 
- `Set COM Output Scan Direction 0xC0` is `Set COM Output Scan Direction  0xC0 for normal or 0xC8 for rotation180` 
- `Set COM Pins Hardware Configuration 0xDA, 0x12 for 128x64 or 0xDA, 0x021 for 128x32px`is `Set COM Pins Hardware Configuration 0xDA, 0x12 for 128x64 or 0xDA, 0x02 for 128x32px`

mbelt, 2022-11-07:
@yuanyihan 
No, thank you. 
| Set Segment Re-Map 0xA0 for normal or 0xA1 for rotation180
Fixed an issue I was having.

Edited previous comment with corrections.

yuanyihan, 2022-11-08:
I mean :
Acording SSD1306-DATASHEET
- [10.1.8  Set Segment Re-Map@P36](https://html.alldatasheet.com/html-pdf/1179026/ETC2/SSD1306/4241/36/SSD1306.html)   A0H/A1H
- [10.1.14 Set COM Output Scan Direction@P37](https://html.alldatasheet.com/html-pdf/1179026/ETC2/SSD1306/4359/37/SSD1306.html) C0H/C8H

We know :
- Normal:A0H and C0H ('and' means both)
- rot180: A1H and C8H
```
#if (SSD1306_LCD_DISP_ROT_180 == true)                   //  https://bugxia.com/931.html
    esp_lcd_panel_io_tx_param(io_handle, 0xC8, NULL, 0); // Set COM Output Scan Direction 0xC0 OR 0xC8
    esp_lcd_panel_io_tx_param(io_handle, 0xA1, NULL, 0); // Set Segment Re-Map 0xA0 OR 0xA1
#else
    esp_lcd_panel_io_tx_param(io_handle, 0xC0, NULL, 0); // Set COM Output Scan Direction 0xC0 OR 0xC8
    esp_lcd_panel_io_tx_param(io_handle, 0xA0, NULL, 0); // Set Segment Re-Map 0xA0 OR 0xA1
#endif
```
[image]
[image]
[image]
[image]

storoj, 2023-10-23:
@VikOlliver I had the same issue and I did my best to provide the fix: #12450. Thanks @yuanyihan.