espressif/esp-idf #9278: ESP32 Openthread Router stops advertising OMR prefix after some time

fcgdam, 2022-07-04:
## Environment

- Development Kit:  ESP32-DOIT DevKit v1
- Kit version (for WroverKit/PicoKit/DevKitC): V1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-3208-gdaa950d9ed ( git commit daa950d9ed1cc3cdc85c09b14ddfeb68a2ac6674)
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:    Linux (Arch Linux fully updated)
- Using an IDE?: No
- Power Supply:       USB
- CMAKE version: cmake version 3.23.2

## Problem Description

Running the Openthread border router example on a ESP32 WROOM module. The Openthread network and the OMR (Off Mesh routing) prefix is advertised to the connected WIFI network, and the hosts routing tables are properly updated to allow hosts to access the Openthread devices. Wireshark shows the IPv6 ICMPv6 Router advertisements (RAs) being sent to the network by the ESP32.
Also Neighbor advertisement messages are sent.

The problem is that after a while (around 50/60 minutes?) any type of ICMPv6 messages including RAs stop being sent by the ESP32. This lends to the OT network entries on the hosts routing tables to expire and hosts looses access to the OT network. Still the ESP32 OTBR continues to work, for example is still discoverable by the Tread App through multicast.

### Expected Behavior

The IPv6 ICMv6 Router Advertisement messages should not stop and be periodically sent by the ESP32 OTBR.

### Actual Behavior

The ESP32 OTBR stops sending RA messages and routing tables on the hosts expire meaning that hosts loose access to the OT network.

### Steps to reproduce

1. Flash the ESP32 with the ot_br example while ESP32 is connected to an compatible OT RCP radio.
2. Check after startup that nearby hosts have the necessary routing entries to access the OT network.
3. Wait around 50/60 minutes.
4. Verify that the routing entries have disappeared since the ESP32 OTBR stop announcing them.

During the testing we can use wireshark to filter only the ICMPv6 messages and see that the ESP32 ICMPv6 RA messages stop after a while.

To recover, only resets works. Disabling and enabling the border router does not recover, and also taking the thread network down (thread stop) and bringing it up also does not recover.

### Code to reproduce this issue

None, standard OOTB ot_br example.

gjc13, 2022-07-05:
Hi, thanks for reporting and testing the issue.

We've successfully located the timer overflow bug and has submitted the fix. Once the fix is availabe  on github we will update in the issue.

ble0, 2022-11-03:
Hi @gjc13 
I'm experiencing the same issue in ESP32S3 / tag: v5.1-dev / xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0. Is this issue supposed to be fixed already?