espressif/esp-idf #9283: wifi:AP has neither DSSS parameter nor HT Information, drop it

KonssnoK, 2022-07-04:
Using ESP32S3 DEVKitC

ESP connect to hotspot as a client.

Randomly, the logs start printing 

```
E (936951) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (937984) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (938476) wifi:AP has neither DSSS parameter nor HT Information, drop it
```

forever until the ESP32 is reset.

Switching on/off the access point does not change the situation, also the device appears operational.
I think it's related somehow to AP discovery?

I have no idea on how to replicate, but looking on the internet i found no answer to what this log means and i think it's part of the IP of Espressif (didn't find it in the repo)

Any help?
Thanks

jareklupinski, 2022-07-09:
the same error just started popping up in my logs while working on something for >2 months without seeing this issue

```
Starting Main Loop
E (10033073) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10035225) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10039626) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10050475) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10052733) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10057027) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10059185) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10063586) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10067992) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10074441) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10078839) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10080995) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10083145) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10089696) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10091848) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10091945) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10096255) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10098398) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10100655) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10104954) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10107103) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10113659) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10115807) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10120213) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10124512) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10124615) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10128920) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10131067) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10137622) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10139769) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10141926) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10142025) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10146329) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10148474) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10152876) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10157278) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10159432) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10163730) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10165983) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10168134) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10170289) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10174690) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10176836) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10178991) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10181237) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10185545) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10187696) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10189850) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10189942) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10194251) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10198650) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10200802) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10202954) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10205206) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10207352) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10209506) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10213910) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10216097) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10220361) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10220461) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10222609) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10226913) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10229060) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10237873) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10240014) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (10242164) wifi:AP has neither DSSS parameter nor HT Information, drop it
[... 368 lines cut ...]
E (11422620) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (11424771) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (11426921) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (11429083) wifi:AP has neither DSSS parameter nor HT Information, drop it
[... cut ...]
E (11468290) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (11470546) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (11472694) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (11474850) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (11479247) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (11483548) wifi:AP has neither DSSS parameter nor HT Information, drop it
```

it goes on for much longer... my Core Debug Level is also set to None, so not sure why it is popping up?

clerodri, 2022-07-13:
I have the same problem, My ESP32 cant connect to my wifinetwork, I dont know why. 
[image]

MaxwellAlan, 2022-07-14:
Hi @jareklupinski @KonssnoK 

This log means that ESP STA receive a beacon frame which lack of HT info realted fields.
Could you capture some wireless pkt when this issue happend?
Or could you show me the AP configs?

Hi @clerodri 

Can you share the connection logs and the AP configs?

Thanks.

KonssnoK, 2022-07-18:
do you have any preerred tool for capturing beacons ? @MaxwellAlan 
This happens independently of being connected or not to anything.

MaxwellAlan, 2022-07-20:
Hi @KonssnoK 
If you have MAC book, maybe you can try using MAC's wireless card to enable Monitor mode to capture wireless pkts.
If not, you can try using Ubuntu or Kali OS with support Monitor mode wireless card to capture pkts. Ref to: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

Thanks.

MaxwellAlan, 2022-08-03:
@KonssnoK 
Could you share more details about this Router(tpye, version, config etc.), we need to purchase same AP to reproduce this issue.

KonssnoK, 2022-08-03:
@MaxwellAlan 

The device that is triggering all the logs is connected to a Netgear WAC510.

the device uses MESH networking (even if it's alone in the mesh -> always root)

KonssnoK, 2022-08-03:
@MaxwellAlan another capture, which shows that the issue is in the esp itself.

```
I (15:02:00.477) mqtt_app: sent publish returned msg_id=45182
I (15:02:00.513) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=45182
I (15:02:08.469) mqtt_app: sent publish returned msg_id=38839
I (15:02:08.518) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=38839

E (11444616) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (11449104) wifi:AP has neither DSSS parameter nor HT Information, drop it

I (15:02:40.465) processor: send status... (60000ms)
I (15:02:40.469) mqtt_app: sent publish returned msg_id=0
I (15:03:00.477) mqtt_app: sent publish returned msg_id=3057
I (15:03:00.502) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=3057
I (15:03:40.465) processor: send status... (60000ms)
I (15:03:40.469) mqtt_app: sent publish returned msg_id=0
I (15:04:00.477) mqtt_app: sent publish returned msg_id=50856
I (15:04:00.500) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=50856
I (15:04:40.465) processor: send status... (60000ms)
I (15:04:40.468) mqtt_app: sent publish returned msg_id=0
[capture_2_subtype_08.zip](https://github.com/espressif/esp-idf/files/9252732/capture_2_subtype_08.zip)

E (11599739) wifi:AP has neither DSSS parameter nor HT Information, drop it

I (15:05:00.477) mqtt_app: sent publish returned msg_id=42900
I (15:05:00.505) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=42900
```

In the attached pcap you can see that the only beacons received are from the same device that i'm monitoring via UART

MaxwellAlan, 2022-08-09:
Hi @KonssnoK 

> the device uses MESH networking (even thou it is alone in the mesh -> always root)

You mean ESP32S3 also enable ESP-MESH but not any other mesh device connect to this root, right?

Anyway, thanks for the debug, it's very helpful for us to look into this issue.

KonssnoK, 2022-08-09:
> Hi @KonssnoK
> 
> > the device uses MESH networking (even thou it is alone in the mesh -> always root)
> 
> You mean ESP32S3 also enable ESP-MESH but not any other mesh device connect to this root, right?
> 
> Anyway, thanks for the debug, it's very helpful for us to look into this issue.

Exactly, our system uses ESP-MESH but our development devices are provisioned only with the AP.

There are other esp devices in the room on different mesh id/password, so they shouldn't matter (apart from the beacons)

KonssnoK, 2022-08-11:
@MaxwellAlan could you drop the awaitingreponse state or do you need anything else?

MaxwellAlan, 2022-08-11:
@KonssnoK

Yes, we are looking into this issue.

chewhs00, 2022-08-24:
I have 5 custom boards using ESP32 Wrover-b modules connecting to the same router about 5 feet away for days. One of them suddenly dropped the wifi connection with reason code 2 (AUTH_EXPIRE). Using our app to provision the wifi connection again still gave the same reason. Tried another nearby network from scan yielded the same result.

nmmsv, 2022-09-03:
I'm getting started with ESP32 and ran into this error message as well. I have an Xfinity router, but I went into the settings and disabled 5GHz wifi. Any tips on how to troubleshoot it? 
```
E (85587) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (85680) wifi:AP has neither DSSS parameter nor HT Information, drop it
[ 90583][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
E (93248) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (93314) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (93448) wifi:AP has neither DSSS parameter nor HT Information, drop it
```

AxelLin, 2022-09-03:
> I'm getting started with ESP32 and ran into this error message as well. 

Could you provide wireshark log? see https://github.com/espressif/esp-idf/issues/9283#issuecomment-1189771070

AxelLin, 2022-10-14:
> @KonssnoK
> 
> Yes, we are looking into this issue.

Any progress? @MaxwellAlan

KonssnoK, 2022-10-14:
can you drop the Awaiting response state? @AxelLin @MaxwellAlan

KonssnoK, 2022-10-19:
@shenjun7 can you drop the waiting response state?

KonssnoK, 2022-10-25:
Sometimes it's super spammy..
```
I (14:31:28.106) processor: send status... (60000ms)
E (122951) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (123524) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (123626) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (124377) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (124488) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (124525) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (125326) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (125468) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (125900) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (126097) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (126211) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (127128) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (127325) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (127440) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (127669) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (127948) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (128147) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (128554) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (128669) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (128982) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (129078) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (129176) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (129769) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (130045) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (130179) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (130293) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (130405) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (130624) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (130815) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (130931) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (131635) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (131749) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (131847) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (132159) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (132355) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (132388) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (132765) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (132978) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (133109) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (133291) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (133437) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (133634) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (133895) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (133961) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (134026) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (134157) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (134206) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (134288) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (134502) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (134682) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (134727) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (134944) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (135009) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (135138) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (135387) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (135534) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (135584) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (135619) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (135798) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (136521) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (136788) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (137435) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (137565) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (137631) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (137762) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (138024) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (138155) wifi:AP has neither DSSS parameter nor HT Information, drop it
[... 576 lines cut ...]
E (305812) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (305925) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (305941) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (306563) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (306678) wifi:AP has neither DSSS parameter nor HT Information, drop it
[... cut ...]
E (310119) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (310348) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (310546) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (310643) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (310856) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (310962) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (311167) wifi:AP has neither DSSS parameter nor HT Information, drop it
```

KonssnoK, 2022-10-26:
I'm putting here my setup:

- Take code from https://github.com/espressif/esp-idf/tree/release/v4.4/examples/mesh/ip_internal_network
- Flash 2 devices
- Start access point and connect devices

- SOMETHING HAPPENS HERE -> ?

- the device connected as child will start dumping the log

xueyunfei998, 2022-10-27:
hi @KonssnoK 

Please replace the esp32s3 under the folder in this path “”components/esp_wifi/lib“”with esp32s3 lib, and then reproduce the problem and provide the following capture package. Thank you.

xueyunfei998, 2022-10-27:
[esp32s3.zip](https://github.com/espressif/esp-idf/files/9876887/esp32s3.zip)

KonssnoK, 2022-10-27:
will do!

KonssnoK, 2022-10-27:
@xueyunfei998 should i use master branch? 
I tried on v4.4
```
cmd.exe /C "cd . && C:\src\espressif\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  @CMakeFiles\ip_internal_network.elf.rsp -o ip_internal_network.elf  && cd ."
c:/src/espressif/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_wpa3.c.obj):(.literal.wpa3_build_sae_commit+0x8): undefined reference to `esp_wifi_sta_get_use_h2e_internal'
c:/src/espressif/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_wpa3.c.obj): in function `wpa3_build_sae_commit':
C:/src/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_wpa3.c:27: undefined reference to `esp_wifi_sta_get_use_h2e_internal'
c:/src/espressif/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa_common.c.obj):(.literal.wpa_parse_wpa_ie_rsnxe+0x0): undefined reference to `esp_wifi_sta_get_config_sae_pwe_h2e_internal'
c:/src/espressif/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa_common.c.obj): in function `wpa_parse_wpa_ie_rsnxe':
C:/src/esp-idf/components/wpa_supplicant/src/common/wpa_common.c:156: undefined reference to `esp_wifi_sta_get_config_sae_pwe_h2e_internal'
c:/src/espressif/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj):(.literal.wpa_supplicant_process_3_of_4+0x0): undefined reference to `esp_wifi_sta_disable_wpa2_authmode_internal'
c:/src/espressif/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj): in function `wpa_supplicant_process_3_of_4':
C:/src/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:1258: undefined reference to `esp_wifi_sta_disable_wpa2_authmode_internal'
c:/src/espressif/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj): in function `wpa_sm_set_ap_rsnxe':
C:/src/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:2592: undefined reference to `esp_wifi_sta_get_config_sae_pwe_h2e_internal'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

xueyunfei998, 2022-10-27:
hi @KonssnoK 
I provided wifi lib based on this commit(esp-idf v4.4 https://github.com/espressif/esp-idf/commit/6c5fb29c2c513936ecefa7cdf49fb324e316546e)

KonssnoK, 2022-10-27:
thanks for the explanation, i'll retry

KonssnoK, 2022-10-27:
Probably unrelated,
one device stopped printing the beaconing after the following error:

```
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
n: Received Routing table [2] 7c:df:a1:e1:6a:d8
W (2248399) wifi:I (2248399) mesh: [scan]new scanning time:600ms, beacon interval:300ms
inactive timer: now=8602c5ef last_rx_time=6bdec566 diff=6b127, aid[1]7c:df:a1:e1:ca:90 leaveI (2241889) mesh_mqtt: Other event id:7

I (2248419) wifi:station: 7c:df:a1:e1:ca:90 leave, AID = 1, bss_flags is 658531, bss:0x3fcb3204
E (2248419) esp-tls: [sock=54] connect() error: Host is unreachable
I (2248439) wifi:new:<11,2>, old:<11,2>, ap:<11,2>, sta:<11,0>, prof:11
E (2248439) TRANSPORT_BASE: Failed to open a new connection: 32772
I (2242109) mesh_main: Sending routing table to [1] 7c:df:a1:e1:ca:90: sent with err code: 16391
E (2248449) MQTT_CLIENT: Error transport connect
W (2248449) wifi:I (2248469) mesh_mqtt: MQTT_EVENT_ERROR
inactive timer: now=86039686 last_rx_time=6bfd2c22 diff=6a994, aid[2]7c:df:a1:e1:6a:d8 leaveI (2248469) mesh_mqtt: MQTT_EVENT_DISCONNECTED

I (2248489) wifi:I (2248469) mesh_main: Sending routing table to [2] 7c:df:a1:e1:6a:d8: sent with err code: 16391
station: 7c:df:a1:e1:6a:d8 leave, AID = 2, bss_flags is 658531, bss:0x3fcb35c8
I (2248499) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<11,0>, prof:11
I (2248509) wifi:scan in process, ignore set current channel
W (2248519) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248519) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248529) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248529) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248539) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248549) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248549) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248559) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248559) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248569) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248569) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248579) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248579) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248589) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248599) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248599) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248609) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248609) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248619) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248619) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248629) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248629) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248639) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248639) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248649) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248649) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248659) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248659) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248669) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248669) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248679) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248689) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248689) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248699) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248699) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248709) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248709) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248719) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248719) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2248729) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248729) wifi:inactive timer, bss(0x3fcb35c8) is removed
[... 149 lines cut ...]
W (2249529) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2249529) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2249539) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2249539) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2249549) wifi:inactive timer, bss(0x3fcb35c8) is removed
W (2249549) wifi:inactive timer, bss(0x3fcb3204) is removed
W (2248419) wifi:pp q full: 7-3fcb4128

E (2249559) wifi:fail to post timer, type=12
I (2249569) mesh: [scan]new scanning time:1500ms, beacon interval:1000ms
I (2249579) mesh_main: <MESH_EVENT_CHILD_DISCONNECTED>aid:1, 7c:df:a1:e1:ca:90
I (2249579) mesh: <nvs>write assoc:0
W (2249579) mesh_main: <MESH_EVENT_ROUTING_TABLE_REMOVE>remove 1, new:2
I (2249579) mesh: [scan]new scanning time:300ms, beacon interval:100ms
I (2249609) mesh_main: <MESH_EVENT_CHILD_DISCONNECTED>aid:2, 7c:df:a1:e1:6a:d8
W (2249609) mesh_main: <MESH_EVENT_ROUTING_TABLE_REMOVE>remove 1, new:1
I (2250499) mesh_main: Tried to publish layer:1 IP:192.168.43.144
I (2250499) mesh_mqtt: sent publish returned msg_id=60042
I (2250499) mesh_main: Sending routing table to [0] 7c:df:a1:e0:74:40: sent with err code: 0
I (2250499) mesh_main: Received Routing table [0] 7c:df:a1:e0:74:40
I (2252189) mesh: [wifi]disconnected reason:201(), continuous:5/max:12, root, vote(,stopped)<><>
I (2252189) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201
I (2252509) mesh_main: Tried to publish layer:1 IP:192.168.43.144
I (2252509) mesh_mqtt: sent publish returned msg_id=41645
```

KonssnoK, 2022-10-27:
I think i got the log
```
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieE (7160123) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (7160123) task_wdt:  - IDLE (CPU 0)
E (7160123) task_wdt: Tasks currently running:
E (7160123) task_wdt: CPU 0: wifi
E (7160123) task_wdt: CPU 1: mqtt task
E (7160123) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x42013546:0x3FC955F00x40376BB6:0x3FC95610 0x4004883E:0x3FCE5880  |<-CORRUPTED
0x42013546: protocomm_add_endpoint_internal at C:/src/esp-idf/components/protocomm/src/common/protocomm.c:86

0x40376bb6: cpu_ll_get_core_id at C:/src/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h:34
 (inlined by) esp_intr_enable at C:/src/esp-idf/components/esp_hw_support/intr_alloc.c:734

E (7160123) task_wdt: Print CPU 1 backtrace
e
80B11kprrce:0e403787BD:
ieFC952F0_p403e6BBac0n,6C95
1e 0842015paA:0_bFCABnF0 9
2e0e82210x3pCrBe20e0xo20847D
CAeBe0 0x420aEseD:0acFC,6580 0xe28921B_p0r3e_AeBc0 06420
De3e8023FCpBrCe 0e420nE8A5:
Bi2e 0x420_51Dse0beacAnC48
i2ee504D10p3FCe_Fe0 0x46020
i9:080F1A_F80 exb0a8oF,650x
0i ex4031EEAar0e3FCac0n0 1

ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
senieee80211_parse_beacon,580
```

KonssnoK, 2022-10-27:
[7440.log](https://github.com/espressif/esp-idf/files/9879150/7440.log)
[dev1.log](https://github.com/espressif/esp-idf/files/9879151/dev1.log)
[dev2.log](https://github.com/espressif/esp-idf/files/9879153/dev2.log)

these are the logs collected by the 3 devices while in parallel another device (connected to the same router)
is printing

```

E (7205053) wifi:AP has neither DSSS parameter nor HT Information, drop it
I (12:00:14.887) processor: send status... (60000ms)
I (12:00:14.891) mqtt_app: sent publish returned msg_id=0
E (7209248) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7210918) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7216062) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7218162) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7218684) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7218701) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7219847) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7221308) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7221830) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7221831) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7222994) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7223943) wifi:AP has neither DSSS parameter nor HT Information, drop it
I (12:00:33.887) processor: send offline message...
I (12:00:33.891) mqtt_app: sent publish returned msg_id=10415
E (7224452) wifi:AP has neither DSSS parameter nor HT Information, drop it
I (12:00:34.102) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=10415
E (7224557) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7227597) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7228236) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7230219) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7230743) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7231791) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7236071) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7243325) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7243326) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7243849) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7243850) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7244899) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7248683) wifi:AP has neither DSSS parameter nor HT Information, drop it
I (12:01:00.887) history: buffer info:
  ID: 0
  timestamp: 10.27.2022 12:01:00
  mask: oooooooooo oooooooooo oooooooooo oooooooooo oooooooooo oooooooooo 
  samples: 60
I (12:01:00.899) mqtt_app: sent publish returned msg_id=33627
I (12:01:00.985) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=33627
E (7253876) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7255482) wifi:AP has neither DSSS parameter nor HT Information, drop it
I (12:01:14.887) processor: send status... (60000ms)
I (12:01:14.890) mqtt_app: sent publish returned msg_id=0
E (7276938) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7280025) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (7287368) wifi:AP has neither DSSS parameter nor HT Information, drop it

```

KonssnoK, 2022-10-27:
Another short run in which i had just 2 or 3 prints on the additional device attached to the same router (my hotspot)

[02_dev1.log](https://github.com/espressif/esp-idf/files/9879233/02_dev1.log)
[02_dev2.log](https://github.com/espressif/esp-idf/files/9879234/02_dev2.log)
[02_7440.log](https://github.com/espressif/esp-idf/files/9879235/02_7440.log)

I'm not able to see the same print in the logs, probably because of the amount of spamming.

@xueyunfei998

KonssnoK, 2022-10-27:
@xueyunfei998 
i was able to get the message in one log!

The error occurs when we have `ieee80211_parse_beacon,834`

```
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,834
E (1196568) wifi:AP has neither DSSS parameter nor HT Information, drop it
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,834
E (1196588) wifi:AP has neither DSSS parameter nor HT Information, drop it
[... 137 lines cut ...]
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817
ieee80211_parse_beacon,580
ieee80211_parse_beacon,652
ieee80211_parse_beacon,656
ieee80211_parse_beacon,809
ieee80211_parse_beacon,817

```

xueyunfei998, 2022-10-28:
hi @KonssnoK 
Please continue to test with lib
Please use one esp32s3 b

oard to connect the hot spot test, not two.

[esp32s3.zip](https://github.com/espressif/esp-idf/files/9885074/esp32s3.zip)

xueyunfei998, 2022-10-28:
hi @KonssnoK 
You can use this example to test it, and let the board of esp32s3 connect to the router, instead of using the example of mesh.

examples/wifi/getting_started/station

KonssnoK, 2022-10-28:
@xueyunfei998 

```

recv beacon
eb->data_len is 289
sta input dst is ffffffffffff
rst is 7cdfa1e29ed5
ieee80211_parse_beacon,585
ieee80211_parse_beacon,657
ieee80211_parse_beacon,661
ieee80211_parse_beacon,814
ieee80211_parse_beacon,822
recv beacon
eb->data_len is 181
sta input dst is ffffffffffff
rst is 6e50118838e4
ieee80211_parse_beacon,839
E (2308619) wifi:AP has neither DSSS parameter nor HT Information, drop it
recv beacon
eb->data_len is 279
sta input dst is ffffffffffff
rst is eacbbc919ebc
ieee80211_parse_beacon,585
ieee80211_parse_beacon,657
ieee80211_parse_beacon,661
ieee80211_parse_beacon,814
ieee80211_parse_beacon,822
recv beacon
eb->data_len is 238
sta input dst is ffffffffffff
rst is decbbc919ebc
ieee80211_parse_beacon,585
ieee80211_parse_beacon,657
ieee80211_parse_beacon,661
ieee80211_parse_beacon,814
ieee80211_parse_beacon,822

```

KonssnoK, 2022-10-28:
another note: 
for now replicated only on internal_ip_network.
simple wifi station example still running
[03_7440.zip](https://github.com/espressif/esp-idf/files/9887552/03_7440.zip)

xueyunfei998, 2022-10-28:
hi @KonssnoK 
Can you find out which router's mac address is 6e:50:11:88:38:e4?

KonssnoK, 2022-10-28:
[04_7440.zip](https://github.com/espressif/esp-idf/files/9887677/04_7440.zip)

apparently there is no router with such mac

[image]
[image]
[image]

KonssnoK, 2022-10-28:
also no other device running the code is finding this beacon from this mac address...

KonssnoK, 2022-10-28:
ok, 
i was able to trigger the same log on one of the devices with the wifi_station example and yes it comes from the same MAC 
6e50118838e4

```
sta input dst is ffffffffffff
rst is d6927dc32 1e
ieee80211_parse_beacon,585
ieee80211_parse_beacon,657
ieee80211_parse_beacon,661
ieee80211_parse_beacon,814
ieee80211_parse_beacon,822
recv beacon
eb->data_len is 91
sta input dst is ffffffffffff
rst is 6e50118838e4
ieee80211_parse_beacon,839
E (34857) wifi:AP has neither DSSS parameter nor HT Information, drop it
recv beacon
eb->data_len is 257
sta input dst is ffffffffffff
rst is e0cbbc919ebc
ieee80211_parse_beacon,585
ieee80211_parse_beacon,657
ieee80211_parse_beacon,661
ieee80211_parse_beacon,814
ieee80211_parse_beacon,822
recv beacon
eb->data_len is 323
sta input dst is ffffffffffff
rst is e6cbbc919ebc
ieee80211_parse_beacon,585
```

xueyunfei998, 2022-10-28:
@KonssnoK 
Do you still have this problem when using this wifi_station example  to successfully connect to the router?

KonssnoK, 2022-10-28:
yes,
you can see in the attached log from a wifi_station example running
[05_dev2.zip](https://github.com/espressif/esp-idf/files/9888053/05_dev2.zip)

but i have another device with the same firmware right next to it, and it didn't have any occurrence.
Also, the example with mesh_ip_internal instead has thousands of occurrences.
[05_7440.zip](https://github.com/espressif/esp-idf/files/9888062/05_7440.zip)

xueyunfei998, 2022-10-31:
[esp32s3.zip](https://github.com/espressif/esp-idf/files/9898801/esp32s3.zip)

Please continue to test with this lib,

This lib adds the channel where beacon is located.

Your software can't catch the address 6e:50:11:88:38:e4 because this beacon doesn't have an ssid, so your software won't show it.

Now I need your help. After using this lib to reproduce, the channel of this abnormal beacon will be displayed. After knowing the channel, use wireshark to grab the packet in its channel and catch this abnormal beacon.

xueyunfei998, 2022-10-31:
> @xueyunfei998 yes, you can see in the attached log from a wifi_station example running [05_dev2.zip](https://github.com/espressif/esp-idf/files/9888053/05_dev2.zip)
> 
> but i have another device with the same firmware right next to it, and it didn't have any occurrence.
> 
> Instead, the example with mesh_ip_internal has thousands of occurrences. [05_7440.zip](https://github.com/espressif/esp-idf/files/9888062/05_7440.zip)

in the attached log from a wifi_station example running

This log shows that the router was not successfully connected.

If the router is successfully connected, there will be such a log display.
I (1029) wifi: AP's beacon interval = 102400 us, DTIM period = 3
I (2089) esp_netif_handlers: sta ip: 192.168.77.89, mask: 255.255.255.0, gw: 192.168.77.1
I (2089) wifi station: got ip:192.168.77.89
I (2089) wifi station: connected to ap SSID:myssid password:mypassword‘’

You can configure ssid and passwd in menuconfig.

KonssnoK, 2022-10-31:
yeah they are configured, i guess the amount of logging is actually killing the performances...
the device cannot authenticate because the task can't probably run

KonssnoK, 2022-10-31:
```
recv beacon
eb->data_len is 206
rst is 165ec160a057
rx channel is 6
ieee80211_parse_beacon,586
ieee80211_parse_beacon,658
ieee80211_parse_beacon,662
ieee80211_parse_beacon,815
ieee80211_parse_beacon,823
recv beacon
eb->data_len is 181
rst is 7e14749cbd11
rx channel is 6
ieee80211_parse_beacon,840
E (10537) wifi:AP has neither DSSS parameter nor HT Information, drop it
recv beacon
eb->data_len is 240
rst is bca511ad86a0
rx channel is 6
ieee80211_parse_beacon,586
ieee80211_parse_beacon,658
ieee80211_parse_beacon,662
ieee80211_parse_beacon,815
ieee80211_parse_beacon,823

```

xueyunfei998, 2022-10-31:
@KonssnoK 
You can grab packets on channel 6 and provide the corresponding log.

KonssnoK, 2022-10-31:
@xueyunfei998 
yeah i'm trying to find a way to get a linux wireshark laptop :D 
Meanwhile, please notice that the mac address has changed since Friday, which is suspicious..

KonssnoK, 2022-10-31:
@xueyunfei998 
wireshark is running since a while now, no packet from 7e14749cbd11 was ever seen. 
i'm waiting to see DSSS again in the devices running station FW (currently i see it on another device), then i'll update the wireshark logging.

KonssnoK, 2022-10-31:
Found one device giving the error from another non existing MAC address
```
recv beacon
eb->data_len is 298
rst is d240ef6f60d1
rx channel is 6
ieee80211_parse_beacon,586
ieee80211_parse_beacon,658
ieee80211_parse_beacon,662
ieee80211_parse_beacon,815
ieee80211_parse_beacon,823
recv beacon
eb->data_len is 91
rst is 4ef 47f04c86
rx channel is 6
ieee80211_parse_beacon,840
E (662744) wifi:AP has neither DSSS parameter nor HT Information, drop it
recv beacon
eb->data_len is 298
rst is d240ef6f60d1
rx channel is 6
ieee80211_parse_beacon,586
ieee80211_parse_beacon,658
ieee80211_parse_beacon,662
ieee80211_parse_beacon,815
ieee80211_parse_beacon,823
```

out of 120k packets there is never such mac address, the esp32s3 (only one out of 3) captured 748 instanced instead.

KonssnoK, 2022-10-31:
[logs.pcapng.gz](https://github.com/espressif/esp-idf/files/9900558/logs.pcapng.gz)

@xueyunfei998 here the pcap logs
the mac is always the same in the station FW (as the message above)

xueyunfei998, 2022-10-31:
@KonssnoK 
Upload the log corresponding to this network capture packet.

xueyunfei998, 2022-10-31:
Do you have other test development boards, such as esp32,esp32c3,esp32s2 and so on

> @xueyunfei998 yes, you can see in the attached log from a wifi_station example running [05_dev2.zip](https://github.com/espressif/esp-idf/files/9888053/05_dev2.zip)
>
> but i have another device with [...]

You said this bug didn't appear on the other board?
Do you have any other development boards, such as esp32 esp32c3,esp32s2, etc.
If so, a comparative test can be conducted.

KonssnoK, 2022-10-31:
@xueyunfei998 
the application is the wifi station example.. i just changed the SSID and password..
[08_7440.zip](https://github.com/espressif/esp-idf/files/9900725/08_7440.zip)

i have available
- one ESP32-S2-Sonia-1_V1.2
- one ESP32-C3-DevKitC-02v1.1
but remember that 3 other esp32s3 didn't show the same error at the same time, so we might have to wait a lot before seeing it on these other devices.

KonssnoK, 2022-10-31:
you mean without modified wifi library?

xueyunfei998, 2022-10-31:
> you mean without modified wifi library?

Yes, test ESP32-S2 and ESP32-C3 with normal lib.

KonssnoK, 2022-10-31:
@xueyunfei998 for now i can't see any log on S2 and C3.
Also the S3 with wifi station firmware is not logging anything.
Meanwhile, my mesh esp32s3 is dumping the error (4 devices attached to the same AP)

```
E (17363311) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17363753) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17366374) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17367423) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17367702) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17368346) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17368750) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17369504) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17369521) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17369897) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17370028) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17370569) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17370929) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17371093) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17371125) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17371241) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17371339) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17371441) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17371617) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17371945) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17372354) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17372567) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17372715) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17372977) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17373129) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17373190) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17373945) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17374550) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17374648) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17374861) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17375058) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17375287) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17375369) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17375680) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17375779) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17376073) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17376581) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17376860) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17377253) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17377384) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17377762) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (17378678) wifi:AP has neither DSSS parameter nor HT Information, drop it
```

KonssnoK, 2022-10-31:
ok it is happening also on esp32s2 with internal_ip_network example

```
E (21362) wifi:AP has neither DSSS parameter nor HT Information, drop it
I (22832) mesh_main: Tried to publish layer:1 IP:192.168.43.177
I (22832) mesh_mqtt: sent publish returned msg_id=32760
I (22832) mesh_main: Received Routing table [0] 7c:df:a1:39:4e:2e
I (22832) mesh_main: Sending routing table to [0] 7c:df:a1:39:4e:2e: sent with err code: 0
I (22992) mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=32760
E (23712) wifi:AP has neither DSSS parameter nor HT Information, drop it
I (24842) mesh_main: Tried to publish layer:1 IP:192.168.43.177
I (24842) mesh_mqtt: sent publish returned msg_id=26946
I (24842) mesh_main: Received Routing table [0] 7c:df:a1:39:4e:2e
```

xueyunfei998, 2022-11-02:
hi @KonssnoK 
[s3.txt](https://github.com/espressif/esp-idf/files/9917597/s3.txt)
[esp32s3.zip](https://github.com/espressif/esp-idf/files/9917601/esp32s3.zip)

Please use this lib for testing, and you need to add S3.txtpatch.

Every time you provide a log, you need to provide the corresponding grab packet.

KonssnoK, 2022-11-02:
@xueyunfei998 
is this patch still on top of 6c5fb29c2c513936ecefa7cdf49fb324e316546e ?

xueyunfei998, 2022-11-02:
> @xueyunfei998 is this patch still on top of [6c5fb29](https://github.com/espressif/esp-idf/commit/6c5fb29c2c513936ecefa7cdf49fb324e316546e) ?

yes

[49 more replies]

closed 2023-02-07 by KonssnoK