espressif/esp-idf #9286: BTController freezes, API_vhci_host_check_send_available() always returns false

TVD-Contributor, 2022-07-04:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROVER-IE
- IDF version v4.4.1
- Build System: CMake
- Compiler version (user code) xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Compiler version (libbtdm_app.a) according to text embedded in libbtdm_app.a appears to be GCC: (crosstool-NG crosstool-ng-1.22.0-78-g4b87650) 5.2.0 
- Operating System:     Windows
- (Windows only) environment type: VSCode
- Using an IDE?: VSCode
- Power Supply:         external 3.3V

## Problem Description

When running HFP and SPP profiles I observe that the controller will randomly freeze, typically after 30 minutes or more. The controller is no longer emitting 0x13 HCI_EVENT_NUMBER_OF_COMPLETED_PACKETS event and esp_vhci_host_check_send_available()/API_vhci_host_check_send_available() query always returns false.

Using ESP32-WROVER-IE with Windows 10 PC as peer. Windows initiates the SPP comport as an outgoing connection to the SBIO. HFP profile has ACL connected and SCO audio traffic is active.

I'm using BTStack as host, and https://github.com/espressif/esp32-bt-lib/tree/b877f7e1fc98dccfcf4dbf31f215c5cb44ec3f0d presompiled binary libbtdm_app.a for BTController.

### Expected Behavior

Controller should be able to run HFP and SPP indefinitely without freezing up, or should automatically recover from freezing up.

### Actual Behavior

HFP and SPP traffic from the ESP32 cannot be sent to peer. Controller ceases to emit HCI NUM_PACKETS acknowledgement event (0x13), and API_vhci_host_check_send_available() query always returns false.

### Steps to reproduce

1. Enable SPP and HFP profiles and establish a connection to/from peer
2. Run for a long time (>30 minutes, sometimes >2 hrs)

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
Here I have some ESP_LOGx() messages along with the BTStack logs. The BTStack logs show 0x6E event (generated from within the BTStack) in response to 0x13 event (from BTController). The logs show SPP data can be sent until the 0x13 events cease being emitted from the controller.

I have annotated the logs as C-style comments. The first half of the log shows normal (working) flow, and the second half shows failed traffic flow.

```
// normal (working)
W (2017094) SBIOSPP: RX 40B // receive bytes on ESP32 UART
W (2017094) SBIOSPP: RX 24B
X (2017104) --spp_callback // request callback from BTStack task to ensure thread-safety
X (2017104) --request (from spp_callback) for 64B // callback from BTStack task
I (2017104) L2CAP: l2cap_notify_channel_can_send... // BTStack queries L2CAP layer
I (2017104) L2CAP: l2cap_can_send sent OK!
[00:33:37.104] EVT <= 79 02 42 00 
[00:33:37.104] EVT <= 89 02 02 00 
X (2017104) ----sbiospp_doSendNow of 64B // BTStack has credit so can permit SPP traffic to be sent to controller...
[00:33:37.104] ACL => 81 00 49 00 
I (2017104) L2CAP: l2cap_notify_channel_can_send...
E (2017104) L2CAP: l2cap_can_send not ready! ACL?=0, res=1
I (2017114) RFCOMM: rfcomm_packet_handler ok
E (2017114) L2CAP: l2cap_can_send not ready! ACL?=0, res=1
W (2017114) L2CAP: l2cap_notify_channel_can_send sent 1 channel triggers
[00:33:37.114] EVT <= 6E 00 
I (2017114) L2CAP: l2cap_can_send sent OK!
[00:33:37.114] EVT <= 79 02 42 00 
I (2017114) RFCOMM: rfcomm_packet_handler ok
W (2017114) L2CAP: l2cap_notify_channel_can_send sent 1 channel triggers
[00:33:37.124] EVT <= 13 05 01 81 01 01 00 // ... BTController emit 0x13 HCI NUM_PACKETS event to indicate it can accept more traffic
[00:33:37.124] EVT <= 13 05 01 81 01 01 00 
[00:33:37.124] EVT <= 13 05 01 81 01 01 00 
[00:33:37.124] EVT <= 6E 00 
[00:33:37.124] EVT <= 6E 00 
[00:33:37.124] EVT <= 13 05 01 81 00 01 00 
[00:33:37.124] EVT <= 6E 00 
[00:33:37.124] EVT <= 13 05 01 81 01 01 00 
[00:33:37.134] EVT <= 13 05 01 81 01 01 00 
[00:33:37.144] EVT <= 13 05 01 81 01 01 00 
[00:33:37.144] EVT <= 13 05 01 81 01 01 00 // the last 0x13 NUM_PACKETS event
[00:33:37.154] ACL <= 81 20 09 00 
[00:33:37.154] LOG -- rfcomm.c.1447: RFCOMM data UIH_PF, new credits channel 0x02: 4, now 24
// below here shows failure
// there is an absence of 0x13 events from this point, so HFP traffic probably consumed all outgoing capacity
X (2017164) SPP send to peer 9B
W (2017164) SBIOSPP: RX 9B // receive bytes on ESP32 UART
X (2017164) --spp_callback
X (2017164) --request (from spp_callback) for 9B // callback from main BTStack task 
I (2017164) L2CAP: l2cap_notify_channel_can_send...
E (2017164) L2CAP: l2cap_can_send not ready! ACL?=0, res=1
E (2017164) RFCOMM: rfcomm_request_can_send_now_event err for cid 2 => x6F
X (2017164) --spp_callback ERR: x1F (#0)
X (2017234) SPP send to peer 64B
W (2017234) SBIOSPP: RX 40B
W (2017234) SBIOSPP: RX 24B
X (2017234) --spp_callback
X (2017234) --request (from spp_callback) for 73B
I (2017234) L2CAP: l2cap_notify_channel_can_send...
E (2017244) L2CAP: l2cap_can_send not ready! ACL?=0, res=1
E (2017244) RFCOMM: rfcomm_request_can_send_now_event err for cid 2 => x6F
X (2017244) --spp_callback ERR: x1F (#0)
X (2017304) SPP send to peer 4B
W (2017304) SBIOSPP: RX 4B
X (2017304) --spp_callback
X (2017304) --request (from spp_callback) for 77B
I (2017304) L2CAP: l2cap_notify_channel_can_send...
E (2017304) L2CAP: l2cap_can_send not ready! ACL?=0, res=1
E (2017304) RFCOMM: rfcomm_request_can_send_now_event err for cid 2 => x6F
X (2017304) --spp_callback ERR: x1F (#0)
X (2017374) SPP send to peer 64B
W (2017374) SBIOSPP: RX 40B
W (2017374) SBIOSPP: RX 24B
X (2017374) --spp_callback
X (2017374) --request (from spp_callback) for 141B
I (2017374) L2CAP: l2cap_notify_channel_can_send...
E (2017384) L2CAP: l2cap_can_send not ready! ACL?=0, res=1
E (2017384) RFCOMM: rfcomm_request_can_send_now_event err for cid 2 => x6F
X (2017384) --spp_callback ERR: x1F (#0)
X (2017444) SPP send to peer 2B
W (2017444) SBIOSPP: RX 2B
X (2017444) --spp_callback
X (2017444) --request (from spp_callback) for 143B
I (2017444) L2CAP: l2cap_notify_channel_can_send...
E (2017444) L2CAP: l2cap_can_send not ready! ACL?=0, res=1
E (2017444) RFCOMM: rfcomm_request_can_send_now_event err for cid 2 => x6F
X (2017444) --spp_callback ERR: x1F (#0)
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)

TVD-Contributor, 2022-07-04:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9042174/sdkconfig.txt)

BetterJincheng, 2022-07-05:
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/9043011/libbtdm_app.zip)
Please test with this lib. (replace `/home/jincheng/idf/components/bt/controller/lib_esp32/esp32/libbtdm_app` with the unzipped attachment)
Please let me know if it works. Thanks!

TVD-Contributor, 2022-07-05:
> [libbtdm_app.zip](https://github.com/espressif/esp-idf/files/9043011/libbtdm_app.zip) Please test with this lib. (replace `/home/jincheng/idf/components/bt/controller/lib_esp32/esp32/libbtdm_app` with the unzipped attachment) Please let me know if it works. Thanks!

So far it has been running 1 hr OK, so I will continue to let it run and provide another update later.

TVD-Contributor, 2022-07-05:
It ran for 5 hours. (After this I did have an issue, but it might have been the Windows stack as my Bluetooth Mouse had frozen.)
So it does look promising. I will try continue running further tests.

BetterJincheng, 2022-07-06:
@TVD-Contributor By the way, are your bt-stack and esp-controller running on two different cores? If so, config them into one core and the issue you reported can be solved too.

TVD-Contributor, 2022-07-06:
@BetterJincheng They are on different cores. I have tried in the past on the same core and got very poor performance, like struggling to establish the HFP connection in the first place. However this might have been using v4.3 with the previous binary driver though, which was terrible if I had any ISRs. The latest 4.4 driver was a big improvement in this area. So far I've had another 5-hour run with no trouble, so yesterday's binary seems to be better still.

BetterJincheng, 2022-07-06:
@TVD-Contributor  Running on different cores does have some bugs. And the lib I attached included the fixes. Soon it will be available on github. For now, it is still in the internal review stage.

TVD-Contributor, 2022-07-06:
@BetterJincheng OK - I have switched to using the same core. This build certainly pairs and connects readily, and is running OK. Again I shall leave it for several hours. Thanks!

Alvin1Zhang, 2022-07-11:
Thanks for reporting, fix on master branch is available on https://github.com/espressif/esp-idf/commit/73cfc078a049f1c6bdb60435baba49e7e5ec0b80 and we will back port the fix to release/4.4, thanks.

Alvin1Zhang, 2022-07-20:
Fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/8546527213565e40a17d74039b33d5b8d77217d8, feel free to reopen.

closed 2022-07-20 by Alvin1Zhang