espressif/esp-idf #9287: ESP_ERR_NVS_CORRUPT_KEY_PART with NVS encryption, flash encryption and secure boot enabled

ThanosSiopoudis, 2022-07-04:
## Environment

- Development Kit:      Custom PCB / ESP32-S3-WROOM-1-N8R8
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-S3-WROOM-1-N8R8
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-3654-gc2ccc383da
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     [macOS]
- Using an IDE?: [No|Yes (please give details)] VSCode
- Power Supply:         external 3.3V

## Problem Description
I tried using Secure Boot V2 + Flash encryption on my module with Release mode. While everything works as expected, and the device boots fine after the first reset once encryption is complete, on subsequent boots NVS encryption fails with `ESP_ERR_NVS_CORRUPT_KEY_PART`. The keys are generated from within the ESP32S3 device

### Expected Behavior
NVS encryption should not return `ESP_ERR_NVS_CORRUPT_KEY_PART` once it has generated the keys

### Actual Behavior
`ESP_ERR_NVS_CORRUPT_KEY_PART` thrown after the second device boot

### Steps to reproduce

1. Create a new project with the attached partition table and sdkconfig file
2. Make sure that NVS encryption is enabled in `menuconfig`
3. Write a simple program with the attached code below
4. Flash the secure bootloader and other partitions to the device
5. Boot the device and allow it to enable secure boot and encrypt the paritions
6. The device will reset itself and the NVS keys will be generated. Once done it will report that the NVS parition was encrypted
7. Reboot the device
8. The device will fail with `ESP_ERR_NVS_CORRUPT_KEY_PART` after `nvs_flash_init`

Partition table
```csv
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,   Size, Flags
nvs,      data, nvs,     0xD000,   0xF000
nvs_key,  data, nvs_keys,0x1C000,  0x1000, encrypted
phy_init, data, phy,     0x1D000,  0x1000,
otadata,  data, ota,     0x1E000,  0x2000,
app0,     app,  ota_0,   0x20000,  0x3E0000,
app1,     app,  ota_1,   0x400000, 0x3E0000
```

### Code to reproduce this issue

```cpp
void app_main(void) {
    esp_err_t err_init = nvs_flash_init();
    if (err_init == ESP_ERR_NVS_NO_FREE_PAGES || err_init == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      // NVS partition was truncated and needs to be erased
      // Retry nvs_flash_secure_init
      ESP_ERROR_CHECK(nvs_flash_erase());
      err_init = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err_init );
    ESP_LOGI(kLogPrefix, "NVS Flash Initialised");
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
--- idf_monitor on /dev/cu.usbserial-1433401 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x1a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
[... 316 lines cut ...]
␛[0;32mI (1420) OLED: Display inited: 1␛[0m
␛[0;32mI (1420) [OLEDDisplay]: Setting brightness to 255␛[0m
␛[0;32mI (1420) [MemoryCard]: Initialising I2C Master␛[0m
␛[0;32mI (1424) [MCPROGC SDFS]: Initialising SD Card.␛[0m
␛[0;32mI (1428) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (1438) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (1446) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (1456) gpio: GPIO[7]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (1466) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (1474) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (1548) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
Name: SC16G
Type: SDHC/SDXC
Speed: 40 MHz
Size: 15193MB
CSD: ver=2, sector_size=512, capacity=31116288 read_bl_len=9
SSR: bus_width=4
␛[0;31mE (1560) nvs: Failed to read NVS security cfg: [0x1117] (ESP_ERR_NVS_CORRUPT_KEY_PART)␛[0m
ESP_ERROR_CHECK failed: esp_err_t 0x1117 (ESP_ERR_NVS_CORRUPT_KEY_PART) at 0x4037b8c4
```

## Other items if possible
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/9042258/sdkconfig.zip)

- [x] sdkconfig file (attach the sdkconfig file from your project folder)

ThanosSiopoudis, 2022-07-05:
@mahavirj Sure, I'll give it a go and I'll let you know.
Do you think this could also be affecting OTA updates? I have code that works without flash encryption, updating the app partitions, and once I try it on a flash encrypted device it bricks it with error `invalid header: 0x6e951e03`

ThanosSiopoudis, 2022-07-05:
I can confirm that both NVS and OTA updates work with SPIRAM disabled.
Are there any workarounds?

mahavirj, 2022-07-06:
>  Are there any workarounds?

Not yet. We are working on this, will keep you posted. CC @ginkgm

mythbuster5, 2022-07-18:
Hi guys, you can pick the commit https://github.com/espressif/esp-idf/commit/8538153616af1b651af990c69ebc1b4cb2835c53 to solve this issue

closed 2022-07-19 by espressif-bot