espressif/esp-idf #9289: sdmmc throws "sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107" for 16GB SDHC/SDXC sd card

muratdemirtas, 2022-07-05:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Module or chip used:  [ESP32-WROOM-32|
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:    Cmake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Clion 2021.1
- Power Supply:  external 5V

## Problem Description

I have a question, if I use a 1GB sd card(512-byte sector) then everything is ok but when I plug a 16GB sd card(512-byte sector size) then this error throws. is this 24-bit/32-bit addressing issue?

> [02:21:53:372] <0x1b>[0;31mE (23:21:53.316) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107<0x1b>[0m␍␊
> [02:21:53:372] <0x1b>[0;31mE (23:21:53.316) LFS_FLASH: failed to read block 29520438, size 512, err 263<0x1b>[0m␍␊
> [02:21:53:372] <0x1b>[0;32mI (23:21:53.329) TAG: sqlite3 used size: 0<0x1b>[0m␍␊
> [02:21:53:372] <0x1b>[0;31mE (23:21:53.341) SQLite3DataProvider: SQL ERROR LENGHT:32<0x1b>[0m␍␊

when I limit block count to 50000, again I got this.

> [02:33:37:466] <0x1b>[0;31mE (23:33:37.455) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107<0x1b>[0m␍␊
> [02:33:37:466] <0x1b>[0;31mE (23:33:37.455) LFS_FLASH: failed to read block 7986, size 512, err 263<0x1b>[0m␍␊
> [02:33:37:466] <0x1b>[0;32mI (23:33:37.461) TAG: sqlite3 used size: 0<0x1b>[0m␍␊

my sd card information:

> [02:36:49:165] SD MCC init ok␍␊
> [02:36:49:165] Name: SD␍␊
> [02:36:49:165] Type: SDHC/SDXC␍␊
> [02:36:49:165] Speed: 40 MHz␍␊
> [02:36:49:165] Size: 14922MB␍␊

stable card information

> [03:03:31:799] Name: SD01G␍␊
> [03:03:31:799] Type: SDSC␍␊
> [03:03:31:799] Speed: 40 MHz␍␊
> [03:03:31:799] Size: 922MB␍␊

muratdemirtas, 2022-07-05:
static int littlefs_api_read(const struct lfs_config *c, lfs_block_t block,
                             lfs_off_t off, void *buffer, lfs_size_t size) {
    struct esp_littlefs_sd_ctx *ctx = c->context;
    int retry_count = 0; int32_t start_time = 0;
	esp_err_t err = ESP_FAIL;
	for(retry_count = 0; retry_count < 5; retry_count++){
		start_time= esp_timer_get_time();
		err = sdmmc_read_sectors(&ctx->sd, ctx->non_dma_buf,
										   block  , 1);
		if (err != ESP_OK) {
			printf("[READ]Failed at block %d, err_code, retry_count %d \n",
		  												block,	retry_count );
			continue;
		}
		memcpy(buffer , ctx->non_dma_buf, 512);
		printf("[READ]Read ok block %d, size %d\n", block, size);
		break;
    }
	if(retry_count == 4){
		printf("[READ]Cannot read block %d\n", block);
		return LFS_ERR_IO;
	}
	int32_t stop_time = esp_timer_get_time() - start_time;
	lfs_perf_counter.read_size += size;
	lfs_perf_counter.read_step++;
	lfs_perf_counter.read_time += stop_time;
	sdmmc_benchmark_printf();
    return 0;
}

closed 2022-07-05 by muratdemirtas

cor-of-org, 2023-04-11:
I had a similar issue. Turned out it wasn't the WiFi, but the "Success LED" that was messing with the SD Card writes. When I disabled the flashing LED (on Pin 2), everything worked fine. ... In case this is useful to someone Googling their way here.