espressif/esp-idf #9290: esp-idf v4.3.3 boot: FAILED TO ENABLE SECURE BOOT

huarunlin, 2022-07-05:
## Environment
- Module:  ESP32-WROOM-32
- IDF version: release v4.3.3
- Build System: idf.py
- Compiler version: esp-2021r2-patch3-8.4.0
- Operating System:  Windows
- environment type: ESP Command Prompt
- Power Supply:  external 3.3V

## Problem Description
My program enabled secure boot(v1). I upgraded esp-idf from v4.3.2 to v4.3.3. Recompile the program. Then burn the program through uart0 on the brand new board. After restarting, the log prompts “**boot: FAILED TO ENABLE SECURE BOOT**”, and restarts continuously. This problem did not appear in the use of v4.3.2.

After that, I tried to use the bootloader.bin, ota_data_initial.bin, partition-table.bin compiled by esp-idf v4.3.2, and the app.bin compiled by v4.3.3, and burn it to the board together, and it can start normally.

sdkconfig.defaults:
``` 
# Enable C++ exceptions 
CONFIG_COMPILER_CXX_EXCEPTIONS=n
# Logging
# CONFIG_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_LOG_DEFAULT_LEVEL_WARN=y
# Flash size
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
# Wifi
CONFIG_ESP32_WIFI_NVS_ENABLED=n
# mdns_start
CONFIG_MDNS_TASK_PRIORITY=4
# Default sdkconfig parameters to use the OTA
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x16000
CONFIG_PARTITION_TABLE_MD5=y
# Bootloader#
CONFIG_BOOTLOADER_FACTORY_RESET=n
CONFIG_BOOTLOADER_DATA_FACTORY_RESET="nvs,phy_init,otadata,storage"
CONFIG_BOOTLOADER_OTA_DATA_ERASE=y
CONFIG_BOOTLOADER_NUM_PIN_FACTORY_RESET=21
# Secure
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_SIGNIG_KEY="secure_boot_signing_key.pem"
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y
CONFIG_NVS_ENCRYPTION=n
#
CONFIG_ESP_NETIF_TCPIP_ADAPTER_COMPATIBLE_LAYER=y
CONFIG_ESP_MAIN_TASK_STACK_SIZE=4608

```

partitions.csv:
```
# Name,   Type,   SubType, Offset,   Size
nvs,         data,     nvs,      0x17000,    0x6000
otadata,  data,     ota,         0x1d000,    0x2000
phy_init,  data,     phy,        0x1f000,    0x1000
factory,   app,      factory,   0x20000,   1280K
ota_0,     app,      ota_0,      0x160000, 1280K
ota_1,     app,      ota_1,      0x2A0000, 1280K
storage,  data,     spiffs,      0x3E0000,  128K

```

## Debug Logs

```
I (30) boot: ESP-IDF v4.3.3-dirty 2nd stage bootloader
I (30) boot: compile time 17:38:13
I (30) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00017000 00006000
I (78) boot:  1 otadata          OTA data         01 00 0001d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0001f000 00001000
I (93) boot:  3 factory          factory app      00 00 00020000 00140000
I (100) boot:  4 ota_0            OTA app          00 10 00160000 00140000
I (108) boot:  5 ota_1            OTA app          00 11 002a0000 00140000
I (115) boot:  6 storage          Unknown data     01 82 003e0000 00020000
I (123) boot: End of partition table
I (127) boot: Defaulting to factory image
I (132) boot_comm: chip revision: 3, min. application chip revision: 0
I (139) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=19db8h (105912) map
I (186) esp_image: segment 1: paddr=00039de0 vaddr=3ffb0000 size=03f88h ( 16264) load
I (193) esp_image: segment 2: paddr=0003dd70 vaddr=40080000 size=022a8h (  8872) load
I (197) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=91e18h (597528) map
I (416) esp_image: segment 4: paddr=000d1e40 vaddr=400822a8 size=14748h ( 83784) load
I (451) esp_image: segment 5: paddr=000e6590 vaddr=50000000 size=00010h (    16) load
I (451) esp_image: segment 6: paddr=000e65a8 vaddr=00000000 size=099d8h ( 39384) 
I (471) esp_image: Verifying image signature...
I (822) boot: Loaded app from partition at offset 0x20000
I (822) boot_comm: chip revision: 3, min. application chip revision: 0
I (824) esp_image: segment 0: paddr=00001020 vaddr=3fff00b0 size=02f54h ( 12116) 
I (834) esp_image: segment 1: paddr=00003f7c vaddr=40078000 size=05d6ch ( 23916) 
I (845) esp_image: segment 2: paddr=00009cf0 vaddr=40080400 size=00e68h (  3688) 
I (849) secure_boot_v1: Generating new secure boot key...
I (855) secure_boot_v1: Generating secure boot digest...
I (900) secure_boot_v1: Digest generation complete.
I (900) boot: Checking flash encryption...
I (901) flash_encrypt: Generating new flash encryption key...
I (906) flash_encrypt: Read & write protecting new key...
I (912) flash_encrypt: Setting CRYPT_CONFIG efuse to 0xF
I (918) flash_encrypt: Disable UART bootloader encryption...
I (924) flash_encrypt: Disable UART bootloader decryption...
I (931) flash_encrypt: Disable UART bootloader MMU cache...
I (937) flash_encrypt: Disable JTAG...
I (941) flash_encrypt: Disable ROM BASIC interpreter fallback...
I (948) boot_comm: chip revision: 3, min. application chip revision: 0
I (955) esp_image: segment 0: paddr=00001020 vaddr=3fff00b0 size=02f54h ( 12116) 
I (966) esp_image: segment 1: paddr=00003f7c vaddr=40078000 size=05d6ch ( 23916) 
I (977) esp_image: segment 2: paddr=00009cf0 vaddr=40080400 size=00e68h (  3688) 
I (1533) flash_encrypt: Encrypting partition 1 at offset 0x1d000...
I (1626) boot_comm: chip revision: 3, min. application chip revision: 0
I (1626) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=19db8h (105912) map
I (1669) esp_image: segment 1: paddr=00039de0 vaddr=3ffb0000 size=03f88h ( 16264) 
I (1675) esp_image: segment 2: paddr=0003dd70 vaddr=40080000 size=022a8h (  8872) 
I (1679) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=91e18h (597528) map
I (1899) esp_image: segment 4: paddr=000d1e40 vaddr=400822a8 size=14748h ( 83784) 
I (1930) esp_image: segment 5: paddr=000e6590 vaddr=50000000 size=00010h (    16) 
I (1930) esp_image: segment 6: paddr=000e65a8 vaddr=00000000 size=099d8h ( 39384) 
I (1949) esp_image: Verifying image signature...
I (2289) flash_encrypt: Encrypting partition 3 at offset 0x20000...
E (17162) esp_image: image at 0x160000 has invalid magic byte (nothing flashed here?)
E (17162) esp_image: image at 0x2a0000 has invalid magic byte (nothing flashed here?)
I (17168) flash_encrypt: Setting FLASH_CRYPT_CNT for permanent encryption
I (17175) flash_encrypt: Flash encryption completed
I (17181) boot: Checking secure boot...
I (17185) secure_boot_v1: Read & write protecting new key...
I (17192) secure_boot_v1: blowing secure boot efuse...
I (17198) secure_boot_v1: Disable JTAG...
I (17202) secure_boot_v1: Disable ROM BASIC interpreter fallback...
E (17209) secure_boot_v1: secure boot not enabled for bootloader image, EFUSE_RD_ABS_DONE_0 is probably write protected!
E (17221) boot: FAILED TO ENABLE SECURE BOOT (259).
abort() was called at PC 0x40079adb
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
invalid header: 0x32e1e326
ets Jul 29 2019 12:21:46

```

KonstantinKondrashov, 2022-07-05:
hi @huarunlin!
Could to confirm that this fix resolves your issue? [18886.zip](https://github.com/espressif/esp-idf/files/9045948/18886.zip)
 Thanks.

huarunlin, 2022-07-06:
> hi @huarunlin! Could to confirm that this fix resolves your issue? [18886.zip](https://github.com/espressif/esp-idf/files/9045948/18886.zip) Thanks.

@KonstantinKondrashov ,thanks. This patch allows the bootloader to run the application normally.

mahavirj, 2022-07-12:
Fixed with https://github.com/espressif/esp-idf/commit/7841054063f1268edfab4da26a200428b5f195e6

closed 2022-07-12 by mahavirj