espressif/esp-idf #9291: Failed to wakeup from deepsleep

Mr-Techtron, 2022-07-05:
## Environment

- Development Kit : ESP32-Wrover-Kit
- Kit version : DevKitv1
- Module:  ESP32-WROOM-32
- IDF version : v4.4.1-1329b19fe4945
- Build System : idf.py
- Compiler version : esp-2021r2-patch3-8.4.0
- Operating System : Windows
- Environment type : Eclipse plugin
- IDE : Eclipse
- Power Supply : USB

## Problem Description
The device failed to wake-up from the deep sleep if the RTC clock source is set to Internal 8.5MHz oscillator.

### Steps to reproduce

1. Use the example of deepsleep.
2. Change the RTC clock source to Internal 8.5MHz oscillator.

moefear85, 2022-07-09:
i believe this is the same as Issue-[8899](https://github.com/espressif/esp-idf/issues/8899)

AxelLin, 2022-07-15:
@Mr-Techtron 
This is duplicated of #8007, the v4.4 branch should already include the fix. Please update your esp-idf to verify.

Mr-Techtron, 2022-07-15:
@AxelLin 
I will check, and let you know.

cssword, 2022-07-29:
#9455 
Did you wait more time?  Waked up after more than 3 minutes when I set sleep time is 10s in my case. 
Must something missing ......

Mr-Techtron, 2022-08-22:
I have used the tag v4.4.2, and it works perfectly. Thank you

closed 2022-08-22 by Mr-Techtron