espressif/esp-idf #9293: Unable to restart BLE (NimBLE) after being stopped, WDT timeout

zimacode, 2022-07-05:
## Environment

- Development Kit:      [ESP32C3 custom board]
- Module or chip used:  [ESP32C3]
- IDF version: v4.4-367-gc29343eb94
- Build System:         [Make|CMake|idf.py]
- Compiler version: riscv32-esp-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [Yes (please give details)] VS Code
- Power Supply:         [external 3.3V or Battery]

## Problem Description

Hi! I'm experiencing the issue with the BLE start aftter disabling it for power saving. The issue happens unpredictably, so sometimes BLE starts correctly and without any problems. BLE restart is requested by GPIO interrupt. 
I found that the watchdog is triggered just after `btdm_controller_enable(mode)` is called. 

### Expected Behavior
At startup MCU starts advertising. When the device connects and disconnects, BLE is deinitialized, MCU moves to light sleep mode with Bluetooth low power clock specified as "External 32kHz Crystal". Once the GPIO interrupt occurs (causing the queue to pass a command to the appropriate task) the BLE is initialized again and starts advertising.

### Actual Behavior
MCU starts correctly, during BLE initialization a function `btdm_controller_enable(mode)` (inside `esp_bt_controller_enable`) stucks, cuasing watchdog to trigger.

### Steps to reproduce

1. Initialize NibBLE stack and start advertising;
2. Connect to the ESP32C3 with NimBLE app (or similar);
3. Disconnect from ESP32C3;
4. Initiate an interrupt on GPIO8 (with `GPIO_INTR_LOW_LEVEL`).

## Debug Logs
In case of success:
```
W (39851) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (39851) BTDM_INIT: BT controller compile version [d913766]
I (39895) BTDM_INIT: Bluetooth MAC: 7c:df:a1:bf:eb:c2

I (39901) BLE: BLE Host Task Started
I (39913) NimBLE: GAP procedure initiated: advertise; 
I (39913) NimBLE: disc_mode=2
I (39913) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=1280 adv_itvl_max=1760
I (39914) NimBLE:
```

In case of failure:
```
W (50108) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (50108) BTDM_INIT: BT controller compile version [d913766]
I (50151) BTDM_INIT: Bluetooth MAC: 7c:df:a1:bf:eb:c2

E (55107) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (55107) task_wdt:  - IDLE (CPU 0)
E (55107) task_wdt: Tasks currently running:
E (55107) task_wdt: CPU 0: btController
```

hyu7000, 2022-12-27:
Did you solve it?

IshaESP, 2023-01-11:
Hi @hyu7000 ,
I tried reproducing this issue by initiating an interrupt  on GPIO8 (with GPIO_INTR_LOW_LEVEL). But the issue didn't get reproduced. NimBLE stack re-initializes without any issues.

IshaESP, 2023-01-11:
Hi,
Can you please send a demo application to reproduce this issue? and also sdkconfig if possible.

hyu7000, 2023-01-16:
@IshaESP i use arduino-esp32.

when i add `BLEDevice::init("title");` in my code, below log occur.
`task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:`

like the writer said. this log show me after `btdm_controller_enable(mode)` is called.
but i didnt know what is this function.

are you need a code in arduino-esp32?

IshaESP, 2023-01-17:
Hi,
I don't have an arduino-esp32 setup. Can you send me either one of these things for debugging the issue:
1. Your demo application. OR
2. Existing IDF application with which I will be able to reproduce this at my end.

ChrisEAlfred, 2023-07-06:
@IshaESP you can use the blecent_throughput and modify it to stop and start the blecent_host_task

IshaESP, 2023-07-06:
Hi @ChrisEAlfred ,
I tried blecent application on IDF v4.4 on ESP32C3 to continuously init deinit nimble stack in a loop. It works. Can you refer to that example? 
Or let me know if there is any thing different being done in the application.
Attaching the log for the same.
[blecent_init_deinit_terminal_log.zip](https://github.com/espressif/esp-idf/files/11967557/blecent_init_deinit_terminal_log.zip)

rahult-github, 2024-02-23:
Hi @zimacode  , is this issue still observed , any updates ?

rahult-github, 2024-03-26:
Closing issue. Feel free to reopen with details, if needed.

closed 2024-03-26 by rahult-github