espressif/esp-idf #9294: ESP-IDF crashes when i2c_param_config is called with gpio ports 6 and 11

slmrojo, 2022-07-05:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      none
- Module or chip used:  ESP32
- IDF version: v4.4.1
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Visual Studio
- Power Supply:         USB

## Problem Description

Program crashes when i2c_param_config is called with gpio port 6 as scl and gpio port 11 as sda. When other ports are used, it works.

### Expected Behavior

i2c_param_config sets up an i2c port with gpio ports 6 and 11

### Actual Behavior
the program crashes during i2c_param_config
It also doesn't pass any kind of error

### Steps to reproduce

Run the code below, attempting to setup an i2c port on gpio ports 6 and 11.
### Code to reproduce this issue

[GIST](https://gist.github.com/slmrojo/eae114f036d2d5539055807003db8e23)

## Debug Logs

```
I (310) i2c-simple-example: Program started
I (320) i2c-simple-example: debug 1
I (320) i2c-simple-example: debug 1.a
I (320) i2c-simple-example: debug 1.b
I (330) i2c-simple-example: debug 2
```
[debug logs.txt](https://github.com/espressif/esp-idf/files/9048733/debug.logs.txt)

chegewara, 2022-07-05:
Flash pins conflict:
https://raw.githubusercontent.com/gojimmypi/ESP32/master/images/myESP32%20DevKitC%20pinout.png

slmrojo, 2022-07-05:
Can the flash pins not be used for any gpio? or just i2c?

negativekelvin, 2022-07-05:
Those are the default flash pins.  You have to know that the app runs from flash so those pins are always in use reloading the cache and can't be used for anything.

closed 2022-07-06 by slmrojo