espressif/esp-idf #9295: compile-error when optimization-level=size and default/max log verbosity set to "verbose"

moefear85, 2022-07-05:
## Environment

- target: esp32
- IDF version (run ``git describe --tags`` to find it): v5.0-c2ccc38
- Build System:   idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp-2022r1-RC1 11.2.0
- Operating System:  Linux
- Using an IDE?: VSCode

## Problem Description

for any (new/empty) project, if in sdkconfig, the optimization level is set to "optimize for size" and the default or the maximum log level is set to "verbose", then compilation fails.

## Debug Logs

```
In file included from /home/moefear/source/esp-idf-v5.0-c2ccc38/components/hal/include/hal/ledc_hal.h:18,
                 from /home/moefear/source/esp-idf-v5.0-c2ccc38/components/driver/ledc.c:16:
/home/moefear/source/esp-idf-v5.0-c2ccc38/components/driver/ledc.c: In function 'ledc_set_timer_div':
/home/moefear/source/esp-idf-v5.0-c2ccc38/components/hal/esp32/include/hal/ledc_ll.h:57:27: error: 'glb_clk' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   57 |     hw->conf.slow_clk_sel = slow_clk_sel;
      |     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/moefear/source/esp-idf-v5.0-c2ccc38/components/driver/ledc.c:497:25: note: 'glb_clk' was declared here
  497 |     ledc_slow_clk_sel_t glb_clk;
      |                         ^~~~~~~
```

moefear85, 2022-07-05:
this issue is already being handled in PR-[9025](https://github.com/espressif/esp-idf/pull/9025)

closed 2022-07-05 by moefear85

boarchuz, 2022-10-23:
This is still unresolved (as of ae30c509efde0622c478812d76108f46cb18aa5d).

Any updates? Reopen?

Galaxy88, 2023-03-09:
Still unresolved as of commit: 490216a2ace6dc3e1b9a3f50d265a80481b32f6d

igrr, 2023-03-09:
The fix has been merged into release/v5.0 in c41bb46d928a8562faf4483961233c3114aa865e, it is part of v5.0.1 release.