espressif/esp-idf #9296: Unable to connect to WiFi, ESP32S3 goes in loop

ble0, 2022-07-06:
## Environment
- Development Kit:      ESP32S3-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0 
- Operating System: MAC
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description
Unable to connect to WiFi using a modified example in `/examples/wifi/getting_started/station`

### Expected Behavior
ESP32S3 should connect to AP

### Actual Behavior
ESP32 does not connect to AP. It goes is series of loop

### Steps to reproduce
1. Go to `/examples/wifi/getting_started/station`
2. Compile the example with few modifications: 
```
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_MAXIMUM_RETRY  CONFIG_ESP_MAXIMUM_RETRY

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static const char *TAG = "wifi station";

static int s_retry_num = 0;

static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) {
            esp_wifi_connect();
            s_retry_num++;
            ESP_LOGI(TAG, "retry to connect to the AP");
        } else {
            xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
        }
        ESP_LOGI(TAG,"connect to the AP fail");
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "got ip:" IPSTR, IP2STR(&event->ip_info.ip));
        s_retry_num = 0;
        xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
    }
    else if(event_base == WIFI_EVENT) { // FMOD
        ESP_LOGE(TAG, "Unhandled WIFI_EVENT event_id = %d", event_id);
    }
    else if(event_base == IP_EVENT) { // FMOD
        ESP_LOGE(TAG, "Unhandled IP_EVENT event_id = %d", event_id);
    }
}

void wifi_init_sta(void)
{
    s_wifi_event_group = xEventGroupCreate();

    ESP_ERROR_CHECK(esp_netif_init());

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    esp_event_handler_instance_t instance_any_id;
    esp_event_handler_instance_t instance_got_ip;
    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_any_id));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
                                                        IP_EVENT_STA_GOT_IP,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_got_ip));

    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .password = EXAMPLE_ESP_WIFI_PASS,
            /* Setting a password implies station will connect to all security modes including WEP/WPA.
             * However these modes are deprecated and not advisable to be used. Incase your Access point
             * doesn't support WPA2, these mode can be enabled by commenting below line */
	     .threshold.authmode = WIFI_AUTH_WPA2_PSK,
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE)); // FMOD
    ESP_ERROR_CHECK(esp_wifi_start() );

    ESP_LOGI(TAG, "wifi_init_sta finished.");

    /* Waiting until either the connection is established (WIFI_CONNECTED_BIT) or connection failed for the maximum
     * number of re-tries (WIFI_FAIL_BIT). The bits are set by event_handler() (see above) */
    EventBits_t bits = xEventGroupWaitBits(s_wifi_event_group,
            WIFI_CONNECTED_BIT | WIFI_FAIL_BIT,
            pdFALSE,
            pdFALSE,
            portMAX_DELAY);

    /* xEventGroupWaitBits() returns the bits before the call returned, hence we can test which event actually
     * happened. */
    if (bits & WIFI_CONNECTED_BIT) {
        ESP_LOGI(TAG, "connected to ap SSID:%s password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    } else if (bits & WIFI_FAIL_BIT) {
        ESP_LOGI(TAG, "Failed to connect to SSID:%s, password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
    } else {
        ESP_LOGE(TAG, "UNEXPECTED EVENT");
    }

    #if 0   // FMOD
    /* The event will not be processed after unregister */
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, instance_got_ip));
    ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));
    vEventGroupDelete(s_wifi_event_group);
    #endif
}

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
}
```

// If possible, attach a picture of your setup/wiring here.
No special setup, nothing is connected to the ESP32S3 devboard.

### Code to reproduce this issue
Refer to example code

Other info:

nishanth-radja, 2022-07-12:
@ble0 From the capture,STA is trying to connect to the AP,By sending out the AUTH req.But the AP does not ACK the AUTH and no AUTH response.Hence the connection failed.
Is the client far away from the AP ?
Are other clients able to connect to the AP ?
Can we have a capture of successful connection if any other client connects to the AP?

nishanth-radja, 2022-07-12:
> Is the client far away from the AP ? No, in fact it is very near Are other clients able to connect to the AP ? Yes Can we have a capture of successful connection if any other client connects to the AP? Give me some time to provide it and what capture do you exactly need? .pcap?

yes, .pcap will do.

ble0, 2022-07-13:
Rename to .pcap file
[Connected.log](https://github.com/espressif/esp-idf/files/9098434/Connected.log)

nishanth-radja, 2022-07-13:
@ble0 The issue is that the AP has not cleared the AUTH state of the previous connection.Hence The AP is not able to reply to the AUTH.
In the connected capture,The STA sends a Deauth before sending an auth ,Which helps in clearing the state of the AP and hence the AP is able to respond to the AUTH req from STA.
Sending Deauth before Auth was recently removed as it created other issues.We will provide a better workaround for this problem shortly.

ble0, 2022-07-13:
Hi @nishanth-radja 
Ok, thanks, will look forward to it. You can also mention the commit id or esp-idf version once it is ready.

MaxwellAlan, 2022-07-14:
Hi @ble0 
From your log:https://github.com/espressif/esp-idf/files/9053028/10_WiFi_Disconnect.txt, the firmware version: 63017e0, this version still will send deauth before auth, can you confirm that this version connect fail?
Thanks.

MaxwellAlan, 2022-07-14:
Maybe the root cause is not deauth.
Can you try with the master(271f4e4c40d5acecf4c97770211ab5bb61e703cf) and also capture the wireless pkts.
Thanks.

ble0, 2022-07-14:
Hi @MaxwellAlan 
This issue is not easy to reproduce, so it will take me a while.

AxelLin, 2022-12-25:
> Hi @MaxwellAlan This issue is not easy to reproduce, so it will take me a while.

@ble0 
Any update? Can you reproduce the issue with recent master tree?

Alvin1Zhang, 2023-02-20:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2023-02-20 by Alvin1Zhang

Solar320, 2024-02-10:
I have 3pcs of ESP32SS3 Board recently buyed at Aliexpress and it do may be the same:

The DEV board is connected to an breadboard and has an wifi Testprogram on it.

The serial console:

48:40.868 -> ..ESP-ROM:esp32s3-20210327
21:48:41.881 -> Build:Mar 27 2021
21:48:41.881 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
21:48:41.881 -> SPIWP:0xee
21:48:41.881 -> mode:DIO, clock div:1
21:48:41.881 -> load:0x3fce3808,len:0x44c
21:48:41.881 -> load:0x403c9700,len:0xbe4
21:48:41.881 -> load:0x403cc700,len:0x2a68
21:48:41.881 -> entry 0x403c98d4
21:48:42.023 -> [   151][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
21:48:42.059 -> [   180][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
21:48:42.100 -> [   213][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
21:48:42.100 -> [   217][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
21:48:42.147 -> [   269][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
21:48:42.645 -> ...[  2142][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: Web****, BSSID: XX:bf:XX:64:bd:18, Reason: 2
21:48:44.002 -> [  2143][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
21:48:44.045 -> [  2150][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
21:48:44.045 -> [  2157][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
21:48:44.045 -> [  2177][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
21:48:44.124 -> ....[  3988][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: Web*****, BSSID: XX:bf:XX:64:bd:18, Reason: 2
21:48:45.866 -> [  3988][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
21:48:45.866 -> [  3996][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
21:48:45.866 -> [  4003][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
21:48:45.901 -> [  4023][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
21:48:46.151 -> ....[  5902][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: Web******, BSSID: XX:bf:XX:64:bd:18, Reason: 2
21:48:47.797 -> [  5902][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
21:48:47.797 -> [  5910][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
21:48:47.797 -> [  5917][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
21:48:47.797 -> [  5938][V][WiFiGeneric.cpp:97] set_esp_interface_ip():

If I put my Finger on the shielding plate of the ESP, this happens:

21:41:54.444 -> ESP-ROM:esp32s3-20210327
21:41:54.444 -> Build:Mar 27 2021
21:41:54.444 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
21:41:54.444 -> SPIWP:0xee
21:41:54.444 -> mode:DIO, clock div:1
21:41:54.444 -> load:0x3fce3808,len:0x44c
21:41:54.444 -> load:0x403c9700,len:0xbe4
21:41:54.444 -> load:0x403cc700,len:0x2a68
21:41:54.444 -> entry 0x403c98d4
21:41:54.570 -> [   151][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
21:41:54.601 -> [   181][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
21:41:54.637 -> [   214][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
21:41:54.637 -> [   216][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
21:41:54.637 -> [   224][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
21:41:55.164 -> ...[  2182][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: Web*******, BSSID: xx:bf:b8:xx:bd:18, Channel: 9, Auth: WPA2_PSK
21:41:56.604 -> [  2183][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
21:41:56.638 -> [  2196][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.50.197
21:41:56.638 -> [  2198][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
21:41:56.638 -> [  2205][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.50.197, MASK: 255.255.255.0, GW: 192.168.50.1
21:41:56.677 -> .
21:41:56.677 -> WiFi connected
21:41:56.677 -> IP address: 
21:41:56.677 -> 192.168.50.197
21:41:56.677 -> 
21:41:56.677 -> - Signal Level: -55
21:41:58.673 -> - Signal Level: -55
21:42:00.681 -> - Signal Level: -55
21:42:02.670 -> - Signal Level: -55
21:42:04.654 -> - Signal Level: -57

If I remove my Finger from the shielding plate, the connection remains stable but depending from the Board with much bader receiving signal or with the same as above (-50 -57 and so on) 
I also have an ESP32S3 Board which goes to -70 when I release my Finger from the shielding plate!

The better one of this board is also able to connect, as long it is not connected to the breadboard!
The bad one do not connect independend if it is connected to the Breadboard or not. Only touching the shielding plate let it connect.

**Long Story short, the onboard PCB seems to be broken on my boaards, this is apperently not an problem with the Software.
I used this for testing: https://github.com/Edistechlab/DIY-Heimautomation-Buch/blob/master/ESP32-CAM%20Projekte/ESP32-CAM_Antennen_Test/ESP32-CAM_Antennen_Test.ino**

[image]
[image]
[image]

[image]
I also saw that the ESP chip runs very hot with the internal Antenna, so I mounted an IPEX connector and an external wifi antenna
and voila I receiving with -41 to -47db and i can now connect to my router, in many times with only one or two  "AUTH_EXPIREs" 
And instead of nearly 70°C the ESP runss now at 35°C

sivar2311, 2024-02-15:
Hi @Solar320 !

I have the same board with the same breadboard problem :(
Can you explain your last patch (with the small piece of wire) in more detail? 
Did this also solve the breadboard problem?
How long does the wire need to be and where did you solder it?

Solar320, 2024-02-16:
Hello. I soldered an 8,5mm wire of an resistor, from the end of the PCB wire Antenna in to direction of the input. 
So 1/2 of the wire length is used to connect it to the original PCB Wire. The active part is only what is on the left side.
If I do this It will instantly connect after reset an while connected to the Breadboard. 
I will test it in the next time if it also connect stable if I don't use the Breadboard.
[image]
[image]

This is only experimental and the Signal is Ok but not very good, may be it must be minimal shorter to get an better result.
I will try this in the next time.

sivar2311, 2024-02-16:
@Solar320 Thank you for your reply!
Please let me know the results when you have completed the tests.

Solar320, 2024-02-16:
I allready test it now and it work with or without Breadboard.
It looks like the Breadboard has an little negativ influence to the PCB Antenna but I am sure that this is also the case at other 
ESP32 boards.

The most differences in receiving strenght came from that, because I change the direction of the pcb antenne many times, vertical horizontal. But in the averange it looks like that the metall in the Breadbord is not good for the receiving strength.

But this all is not important, Important is that something like this is gone:

48:40.868 -> ..ESP-ROM:esp32s3-20210327
21:48:41.881 -> Build:Mar 27 2021
21:48:41.881 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
21:48:41.881 -> SPIWP:0xee
21:48:41.881 -> mode:DIO, clock div:1
21:48:41.881 -> load:0x3fce3808,len:0x44c
21:48:41.881 -> load:0x403c9700,len:0xbe4
21:48:41.881 -> load:0x403cc700,len:0x2a68
21:48:41.881 -> entry 0x403c98d4
21:48:42.023 -> [   151][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
21:48:42.059 -> [   180][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
21:48:42.100 -> [   213][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
21:48:42.100 -> [   217][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
21:48:42.147 -> [   269][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
21:48:42.645 -> ...[  2142][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: Web****, BSSID: XX:bf:XX:64:bd:18, Reason: 2
21:48:44.002 -> [  2143][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
21:48:44.045 -> [  2150][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
21:48:44.045 -> [  2157][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
21:48:44.045 -> [  2177][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
21:48:44.124 -> ....[  3988][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: Web*****, BSSID: XX:bf:XX:64:bd:18, Reason: 2
21:48:45.866 -> [  3988][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
21:48:45.866 -> [  3996][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
21:48:45.866 -> [  4003][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
21:48:45.901 -> [  4023][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
21:48:46.151 -> ....[  5902][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: Web******, BSSID: XX:bf:XX:64:bd:18, Reason: 2
21:48:47.797 -> [  5902][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
21:48:47.797 -> [  5910][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
21:48:47.797 -> [  5917][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
21:48:47.797 -> [  5938][V][WiFiGeneric.cpp:97] set_esp_interface_ip():

**After soldering this little wire to the PCB Antenna:

Mounted on the breadboard:**
15:19:16.136 -> ESP-ROM:esp32s3-20210327  **Manual reset**
15:19:16.178 -> Build:Mar 27 2021
15:19:16.178 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
15:19:16.178 -> SPIWP:0xee
15:19:16.178 -> mode:DIO, clock div:1
15:19:16.178 -> load:0x3fce3808,len:0x44c
15:19:16.178 -> load:0x403c9700,len:0xbe4
15:19:16.178 -> load:0x403cc700,len:0x2a68
15:19:16.178 -> entry 0x403c98d4
15:19:16.909 -> ..
15:19:17.407 -> WiFi connected
15:19:17.407 -> IP address: 
15:19:17.407 -> 192.168.50.109
15:19:17.407 -> 
15:19:17.407 -> - Signal Level: -57
15:19:19.404 -> - Signal Level: -57
15:19:21.392 -> - Signal Level: -56
15:19:23.412 -> - Signal Level: -58
15:19:25.396 -> - Signal Level: -58
15:19:27.418 -> - Signal Level: -58
15:19:29.376 -> - Signal Level: -52
15:19:31.412 -> - Signal Level: -59
15:19:33.404 -> - Signal Level: -59
15:19:35.376 -> - Signal Level: -54
15:19:37.416 -> - Signal Level: -55
15:19:39.416 -> - Signal Level: -57
15:19:41.416 -> - Signal Level: -56
15:19:43.405 -> - Signal Level: -59
15:19:45.406 -> - Signal Level: -61
[... 81 lines cut ...]
15:26:13.304 -> - Signal Level: -55
15:26:15.324 -> - Signal Level: -51
15:26:17.349 -> - Signal Level: -54
15:26:19.324 -> - Signal Level: -54
15:26:21.303 -> - Signal Level: -57
15:26:23.334 -> - Signal Level: -54
15:26:25.303 -> - Signal Level: -54
15:26:27.339 -> - Signal Level: -55
15:26:29.228 -> ESP-ROM:esp32s3-20210327  **Manual Reset**
15:26:29.228 -> Build:Mar 27 2021
15:26:29.228 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
15:26:29.268 -> SPIWP:0xee
15:26:29.268 -> mode:DIO, clock div:1
15:26:29.268 -> load:0x3fce3808,len:0x44c
15:26:29.268 -> load:0x403c9700,len:0xbe4
15:26:29.268 -> load:0x403cc700,len:0x2a68
15:26:29.268 -> entry 0x403c98d4
15:26:29.957 -> ..
15:26:30.495 -> WiFi connected
15:26:30.495 -> IP address: 
15:26:30.495 -> 192.168.50.109
15:26:30.495 -> 
15:26:30.495 -> - Signal Level: -54
15:26:32.459 -> - Signal Level: -54
15:26:34.499 -> - Signal Level: -54

sivar2311, 2024-02-16:
I don't think the antenna is the cause. 
The mod seems to improve the signal strength but not solve the root cause.

I have a Freenove ESP32-S3-N8R8 board which does not have these problems. 
This board works both plugged into the breadboard and unplugged.

I have 3 more N16R8s where I haven't soldered the pin headers yet. 
When I put them flat on the breadboard everything works fine.

I therefore suspect that there is more of a grounding problem with the board. 
That would also explain why it works with the finger on the shielding.

What do you think?

zhangyanjiaoesp, 2024-02-21:
@Solar320 Are there any other devices on the breadboard besides the DEV board?
@sivar2311 what's your problem? Can you describe it?

sivar2311, 2024-02-21:
Hi @zhangyanjiaoesp 
Thanks for your reply!
I have the exact same board like Solar320.

If the board is not plugged into the breadboard, it connects to my WiFi network within 0.2 seconds.
<details>
<summary>log</summary>

```
05:37:46.049 > ESP-ROM:esp32s3-20210327
05:37:46.057 > Build:Mar 27 2021
05:37:46.057 > rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
05:37:46.063 > SPIWP:0xee
05:37:46.063 > mode:DIO, clock div:1
05:37:46.063 > load:0x3fce3808,len:0x44c
05:37:46.069 > load:0x403c9700,len:0xbd8
05:37:46.069 > load:0x403cc700,len:0x2a80
05:37:46.069 > entry 0x403c98d0
05:37:46.251 > ARDUINO_EVENT_WIFI_READY 0
05:37:46.288 > ARDUINO_EVENT_WIFI_STA_START 2
05:37:46.316 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:37:46.365 > ARDUINO_EVENT_WIFI_STA_CONNECTED 4
05:37:46.380 > ARDUINO_EVENT_WIFI_STA_GOT_IP 7
05:37:46.396 > 192.168.2.188
```
</details>

As soon as I plug it into an (empty) breadboard, no WiFi connection is possible. 

<details>
<summary>log</summary>

```
05:35:56.188 > ESP-ROM:esp32s3-20210327
05:35:56.197 > Build:Mar 27 2021
05:35:56.197 > rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
05:35:56.202 > SPIWP:0xee
05:35:56.202 > mode:DIO, clock div:1
05:35:56.202 > load:0x3fce3808,len:0x44c
05:35:56.208 > load:0x403c9700,len:0xbd8
05:35:56.208 > load:0x403cc700,len:0x2a80
05:35:56.208 > entry 0x403c98d0
05:35:56.389 > ARDUINO_EVENT_WIFI_READY 0
05:35:56.425 > ARDUINO_EVENT_WIFI_STA_START 2
05:35:57.531 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:35:58.554 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:35:59.577 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:00.600 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:01.624 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:02.647 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:03.671 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:04.694 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:05.717 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:06.740 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:07.764 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:08.787 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:09.810 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:10.834 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:11.857 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:12.880 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:13.903 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:14.927 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:15.950 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:16.973 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:17.996 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:19.019 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:20.043 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:21.066 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:22.091 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:23.114 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:24.137 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:25.160 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:26.184 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:27.206 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:28.232 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:29.253 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:30.277 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:31.299 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:32.323 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:33.347 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:34.369 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:35.393 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:36.418 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:37.439 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:38.464 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:39.486 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:40.510 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:41.534 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:42.556 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:43.580 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:44.603 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:45.627 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:46.650 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:47.673 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:48.696 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:49.720 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:50.743 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:51.766 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:52.789 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:53.813 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:54.836 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:55.859 > ARDUINO_EVENT_WIFI_STA_DISCONNECTED 5
05:36:56.427 > 0.0.0.0
```
</details>

The sketch I'm using for testing: [wifi_test.cpp](https://gist.github.com/sivar2311/70930b252a9ede7ffb3b3f58be09b3a5#file-wifi_test-cpp)

I am aware that this is a hardware problem.

zhangyanjiaoesp, 2024-02-21:
@sivar2311 @Solar320 we will test locally and feedback ASAP.

Solar320, 2024-02-24:
> @Solar320 Are there any other devices on the breadboard besides the DEV board? @sivar2311 what's your problem? Can you describe it?

Hello, I were away for one week. In my case it is equal if other components are conected to the breadboard (and ESP). The metall in the breadboards clamps are enough to bring the Antenna out of balance. Every antenna can be missmatched if you bring big metall parts to near to it. But this should not be the case when you only connect the ESP to the breadboard, unless the Antenna is allready missmatchend and the additionally metal from the breadboard give them the rest. 

My guess is that in some cases they use the wrong or components with an to big tolerance between the ESP and the Antenna. (matching circuit) This can explain why only a few people have this problem.

My antennamod is working good enough to bring the ESP in an useable state. 
We can only wait if there are other peoples which have the same problem and can confirm the problem. 

As I see this now... https://github.com/schreibfaul1/ESP32-audioI2S/issues/530   I have installed the ESP32-miniwebradio from "schreibfaul1" a few days ago!!!

It had also this problem, it won't connect to my router in 3m distance. After the antennamod the radio is working as it should.
As i was away in Germany I also get the DAC PCM5102A in my hands to test if there audio came out and it was working as expected. Also it has an good Wifi signal (3 to 4 bars) given to the mod with an IPEX connector and an antenna. The distance was now 6 - 7 m through 2 walls.

[image]
[image]
[image]

sivar2311, 2024-02-25:
The RSSI is not affected in my case.
I used a second ESP32 as an access point for testing. 
Both ESP's "see" each other with an RSSI value greater than -10, so in my case it cannot be due to the antenna.

If the faulty ESP is plugged into the breadboard, I get a **Reason: 2 - AUTH_EXPIRE** error:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
06:15:00.272 > ESP-ROM:esp32s3-20210327
06:15:00.281 > Build:Mar 27 2021
06:15:00.281 > rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
06:15:00.286 > SPIWP:0xee
06:15:00.286 > mode:DIO, clock div:1
06:15:00.286 > load:0x3fce3808,len:0x44c
06:15:00.292 > load:0x403c9700,len:0xbd8
06:15:00.292 > load:0x403cc700,len:0x2a80
06:15:00.292 > entry 0x403c98d0
06:15:00.449 > [   174][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
06:15:00.481 > [   207][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
06:15:00.520 > [   244][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
06:15:00.526 > [   245][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
06:15:00.531 > [   246][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
06:15:01.616 > [  1341][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ESP32APESP32AP, BSSID: 84:cc:a8:5f:19:01, Reason: 2
06:15:01.627 > [  1342][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
06:15:01.633 > [  1350][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
06:15:01.639 > [  1356][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
06:15:01.644 > [  1364][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
```

If the ESP is not in the breadboard, this error does not occur:
```
06:16:37.449 > ESP-ROM:esp32s3-20210327
06:16:37.458 > Build:Mar 27 2021
06:16:37.458 > rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
06:16:37.463 > SPIWP:0xee
06:16:37.463 > mode:DIO, clock div:1
06:16:37.463 > load:0x3fce3808,len:0x44c
06:16:37.469 > load:0x403c9700,len:0xbd8
06:16:37.469 > load:0x403cc700,len:0x2a80
06:16:37.469 > entry 0x403c98d0
06:16:37.626 > [   174][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
06:16:37.659 > [   208][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
06:16:37.698 > [   245][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
06:16:37.704 > [   247][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
06:16:37.709 > [   247][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
06:16:41.739 > [  4285][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ESP32APESP32AP, BSSID: 84:cc:a8:5f:19:01, Reason: 15
06:16:41.750 > [  4287][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
06:16:41.755 > [  4295][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
06:16:41.761 > [  4302][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
06:16:41.772 > [  4310][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
06:16:41.804 > [  4352][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: ESP32APESP32AP, BSSID: 84:cc:a8:5f:19:01, Channel: 1, Auth: WPA2_PSK
06:16:41.815 > [  4354][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
06:16:42.208 > [  4753][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.4.2
06:16:42.213 > [  4756][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
06:16:42.219 > [  4759][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.4.2, MASK: 255.255.255.0, GW: 192.168.4.1
```

sivar2311, 2024-02-25:
I wanted to determine the RSSI from the router to the connected client and compare it with the RSSI value from the faulty ESP32-S3 to the router. Unfortunately, my router does not show me the RSSI of the connected clients.
So I used a second ESP32 as an access point to determine the RSSI to the faulty ESP32-S3. 

I get the RSSI from both ESP's (from the access point ESP32 and from the faulty "client" ESP32-S3) and they both pretty much match. In my case it is irrelevant whether the ESP32-S3 is in the breadboard or not. The values always match and even when the faulty ESP32-S3 is plugged into the breadboard, the value does not get worse. It should be a few dBm less here, but it doesn't in my case.

Solar320, 2024-02-25:
Ok, I now tested the ESP32C6 in all 3 variations. Without Pins, with Pins and withpins and connected to the Breadboard.
All tests in the same distance to the router.

In the first 2 tests this board shows no problems.  When I connect it to the breadboard it has an very bad Influence to the antenna and signal. But this time there are no Auth Errors.

When I touch the shielding plate with my finger the biggest Influence to the Antenna happens when I do this when the ESP is connected to the board. If the ESP not connected the Influence is minimal.

My Router is able to show the Signal strength, in theorie it should show the same differences as the ESP itself.  I can test this also with an ESP AP, but I think that the test should show the same Auth error when It is at the same distance like my router currently is. (e.g. 3m)

 In some weeks I will get one new ESP32 S3 dev board looking like this.
[image]

There is an additionally "2022-V1.3" writing on the backside, my modules do not have this. 
On my modules there is only the writing "ESP32-S3" nothing else.  I don't know if this will made an difference..I will see.

zhangyanjiaoesp, 2024-02-28:
@sivar2311 @Solar320 I didn't reproduce your problem.

I'm using the following ESP32-S3 dev board to test, and I'm on the IDF master branch, using the [station example ](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station). 

[image]

When I only connect the ESP32-S3 board to the breadboard, it can connect to AP quickly. 

[image]

Can you show me a picture about how you connect the ESP with the breadboard ? Also, please tell me the Arduino version you are using,  I will try to test with Arduino again.

Solar320, 2024-02-28:
"@sivar2311 @Solar320 I didn't reproduce your problem."

You may didn't understand the problem, we have not this 2022 V1.3 Version of this board.
You have already the Version V1.3!  
But it sounds good, so it is possible that **my second purchase of the ESP32S3 2022 V1.3**  will working without errors. **I can't test it till now because it is still on the way.**

The connections to the display are in this case irrelevant, because my modules do not work eighter, if it is connectet on the bord or/and if there any additionally wiress connected. 

My boards looking like this, the hve no "2022 v1.3" on it!

[image]
[image]
[image]

zhangyanjiaoesp, 2024-02-29:
> My boards looking like this, they have no "2022 v1.3" on it!
> 
>  ![IMG_20240228_175456](https://private-user-images.githubusercontent.com/111298158/308644465-cf1d3c0b-1562-4699-811e-00141a6346b5.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxNzM3NzMsIm5iZiI6MTcwOTE3MzQ3MywicGF0aCI6Ii8xMTEyOTgxNTgvMzA4NjQ0NDY1LWNmMWQzYzBiLTE1NjItNDY5OS04MTFlLTAwMTQxYTYzNDZiNS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyOVQwMjI0MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lZGZjMGMwNGM0NTAwMmZmODc5NWFmMzNkM2RmYTQwYzUzYThmNDViNzk0YTg5Y2QxOTBkOWRmMmRiNWIwZTBiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.--PiJfPoyPlYzBGJwNgske4CWVs-ckmdsfmzPwR2y04)

@Solar320 I knew your board is this, and I have purchased it and it's on the way.  I saw in one of your comments that you wanted to test the **ESP32-S3 2022 v1.3** board, and I happen to have one, so I tested with it and the issue can't be reproduced, then I share the test results with you.

My other purpose is to hope you to confirm whether my connection between the ESP and the breadboard is correct.

zhangyanjiaoesp, 2024-02-29:
@Solar320 Can you provide me with a purchase link for the board you are using? I find that the one I bought is a little different from yours.

sivar2311, 2024-02-29:
@zhangyanjiaoesp I have the exact same board as Solar320.
I bought it on aliexpress: https://de.aliexpress.com/item/1005006266375800.html (N16R8 version)

Solar320, 2024-02-29:
> @Solar320 I knew your board is this, and I have purchased it and it's on the way. I saw in one of your comments that you wanted to test the **ESP32-S3 2022 v1.3** board, and I happen to have one, so [...]

Ok then I missunderstood you. Your connection should be correct to supply  the Board direct with 3,3V.
You should be carefull when you also connect the USB  the same time, it should not ave an influence but you never know...

@sivar This link should be the right one.

I bought my board there:
https://de.aliexpress.com/item/1005004639132852.html?spm=a2g0o.order_list.order_list_main.82.50cc5c5fSSVejr&gatewayAdapt=glo2deu

If the link do not work search for TZT-FIVE-STARS Store and Tzt ESP32-S3-DevKitC-1 ESP32-S3 wifi
I hope this is working.

captaincarmnlg, 2024-03-01:
if you connect the right side (com port side) of the board wifi does not connect

Solar320, 2024-03-01:
I get my module now, it look little different than on the product photo.
And it has clearly another Antenna! 

This module was connected to the right COM-USB-C and is working "without Pins", with soldered Pins and also when it is connected to the breadboard. The Label is different, my old modules has the Label N16R8 and P2N8 the productphoto has only one label MON16R8 and the modul I get now, has also only one label N16R8 

The Antenna is now super stable.

[image]
[image]
[image]

[image]

sivar2311, 2024-03-01:
@Solar320 My ordered modules and boards should arrive within 10 days. 
Thanks for the link to the correct version of the board!

Solar320, 2024-03-02:
I hope they deliver the same version with the new antenna form, at least I can't quarantee it to 100%.

zhangyanjiaoesp, 2024-03-04:
@Solar320 The antenna of my ESP32-S3 2022 v1.3 board is as follows, is different from yours, but it can work well.
 
[image]

Xiehanxin, 2024-03-04:
hi @sivar2311 @Solar320 , thanks for your feedback. in the hardware design guidelines, we need to ensure there is sufficient clearance around the antenna. Therefore, compared to the module, the RF performance of the development board may be affected to some extent, but it should not completely disrupt the WiFi connection. Additionally, based on the product images you provided earlier, it appears that this product, whether it's the development board or the module, is not an official product. There are even printing errors on the module's silk screen. If you could provide more detailed information about the purchasing channel for this product, we would greatly appreciate it.

Xiehanxin, 2024-03-04:
hi @sivar2311 @Solar320 could you read the MAC of the module? you can use **_esptool.py chip_id_** by esptool 
 https://github.com/espressif/esptool
This is very helpful for us to track the chip's manufacturing information.

sivar2311, 2024-03-04:
Hi @Xiehanxin many thanks for the feedback!

I was aware that the board was a clone. However, I would not have guessed that the module itself is also a clone :/

The link to where I bought it is here: https://github.com/espressif/esp-idf/issues/9296#issuecomment-1970398549

The module on my board looks like exactly the same as in the picture you posted https://github.com/espressif/esp-idf/issues/9296#issuecomment-1975933792

Unfortunately I'm a noob when it comes to python. I can't manage to start the script (error message that pyserial is not available). 
I'm using PlatformIO and the Arduino framework. I don't know what esptool.py outputs but the following sketch

```
#include <Arduino.h>

void setup() {
    Serial.begin(9600);
    uint64_t efuseMac = ESP.getEfuseMac();
    while (efuseMac) {
        Serial.printf("%02X ", efuseMac & 0xFF);
        efuseMac >>= 8;
    }
    Serial.println();
}

void loop() {}
```

outputs `80 65 99 A1 AA 28`

Xiehanxin, 2024-03-04:
hi @sivar2311 thanks for your feedback, we will check the MAC address you provide.

Solar320, 2024-03-04:
> hi @sivar2311 @Solar320 , thanks for your feedback. in the hardware design guidelines, we need to ensure there is sufficient clearance around the antenna. Therefore, compared to the module, the RF performance of the development board may be affected to some extent, but it should not completely disrupt the WiFi connection. Additionally, based on the product images you provided, it appears that this product, whether it's the development board or the module, is not an official product. There are even printing errors on the module. ![image](https://private-user-images.githubusercontent.com/37785066/309675069-a2d182fd-642e-4235-850d-947ff2792789.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1NzgzNzAsIm5iZiI6MTcwOTU3ODA3MCwicGF0aCI6Ii8zNzc4NTA2Ni8zMDk2NzUwNjktYTJkMTgyZmQtNjQyZS00MjM1LTg1MGQtOTQ3ZmYyNzkyNzg5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA0VDE4NDc1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ1N2Q4MzcyYmM1M2VhOTkxMGRiZDNmMzdjNDk1ZTAwOGNmMTg5MjUzYjE2MDBjYzllNWM5MjBiZDFjY2M1OTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.LM4FLkl1c0OkmcMXv7Y2p6_yW2abGne2V3RlK_ZBhJU)
> 
> If you could provide more detailed information about the purchasing channel for this product, (you can send email to [sales@espressif.com](mailto:sales@espressif.com) and add the issue link ) we would greatly appreciate it.
> 
> I have seen that you have bought two different devkitboard, are they all from the https://de.aliexpress.com/item/1005004639132852.html?spm=a2g0o.order_list.order_list_main.82.50cc5c5fSSVejr&gatewayAdapt=glo2deu ?

Yes I got the working Boards from this TZT Store and they have the new antenna form.
I bought an ESP3S3 Board and an the same as chipmodule, I only test the normal ESP32S3 board but I think the chipmodule also will work, both have the new Antenna form.

The not working one's are from:  
Store-Infos
Name des Geschäfts:Shop-ID:Standort:Seit geöffnet:
AliSupplier Store 5017177 Shenzhen Longgang District, ChinaMay 5,2019

They now changed their Product picture, now they has also have the Label "2022 V1.3" on it. 
But this is no guarantee that they also will send this modell.

https://de.aliexpress.com/item/1005005767192637.html?spm=a2g0o.order_list.order_list_main.106.70925c5fDX06Yb&gatewayAdapt=glo2deu

Solar320, 2024-03-04:
[image]

MAC 30:30:F9:77:0c:c8  this should be the MAC

sivar2311, 2024-03-07:
> I get my module now, it look little different than on the product photo. And it has clearly another Antenna!
>
> This module was connected to the right COM-USB-C and is working "without Pins", with [...]

My boards arrived today. 
I can confirm that these v1.3 boards work perfectly!

Solar320, 2024-03-07:
This are great news, since the Alisupplier store seems still to sell the old modules, I don't buy anymore there. I will now buy all modules at TZT-FIVE-STARS Store. I also bought an ESP32C6 module from TENSTAR ROBOT Store, this module has the old antenna design but seems to work without Auth errors and seems to be stable. But I prefer the new antenna design, delivered from TZT.

Sorry at [Xiehanxin](https://github.com/Xiehanxin) the espressive Team has mailed me, telling me that this module from Alisupplier was not genuie and gave me an link to their esspressive store. This store is INSANE in the negative meaning! They want over 20€ for an normal ESP32S3 module, NO WAY!!! https://de.aliexpress.com/item/1005003979778978.html?gatewayAdapt=4itemAdapt

sivar2311, 2024-03-10:
> ![image](https://private-user-images.githubusercontent.com/37785066/309675069-a2d182fd-642e-4235-850d-947ff2792789.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwODU5MzEsIm5iZiI6MTcxMDA4NTYzMSwicGF0aCI6Ii8zNzc4NTA2Ni8zMDk2NzUwNjktYTJkMTgyZmQtNjQyZS00MjM1LTg1MGQtOTQ3ZmYyNzkyNzg5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEwVDE1NDcxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFlMjQ0NTZjYWI3ZTA1MDA2MjYyMmZlMzY3MDI1OTJlZWM1MTIzMjgxY2E1OWQ3NDM1YjdmMDQxNjBmMjc1NDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RrIIMUc0yjHw-DSItPn05J_RMpVUEshJM5NrKl0u360)

@Xiehanxin 
Here is a picture from the module of the new v1.3 board:
[image]

Is this a genuine module? What are the characteristics by which one can distinguish original from fake? 
Unfortunately, only renderings of the modules can be seen on the official Espressif AliExpress store.

Solar320, 2024-03-10:
This is like my new modules look like. Original or not, but you won't pay over 20€ for only one board at espressive store at least I don't. The module looks like it is mounted a bit to high. on the upper left edge there is very few space between the first 2 pins.
It can also 

Espressive seems to sell this chips in big ammounts, who say that all modules which don't come directly from Espressive are not genuie? Every manufacturer can create it's own modules, based on this espressive chips and are compatible to the espressive 's own boards. The only different is that this boards do not come directly from espressive may be also the quality is different. If espressive attaches so much importance to it, they should play on any protected program which shows were it is manufactured.

Some professionals may need the boards, coming directly from espressive, to be sure to have the same quality over a high quantity.