espressif/esp-idf #9299: Not able to connect App in BLE-MESH network

hipal31, 2022-07-06:
## Environment

- Module or chip used: ESP32-C3
- IDF version : v4.3.3
- Build System: idf.py
- Compiler version : 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: external 3.3V

## Problem Description

I am developing product using Ble-mesh. I am using app which is developed using [Android-nRF-Mesh-Library](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library). Problem occurs when I add more then 50 devices in a BLE-MESH network then sometimes I am not able to connect with device in BLE-MESH network though mobile app. May be that's because of so many packets transferred in mesh network. In app, device is available to connect but device doesn't respond connect request.
 
### Expected Behavior
App connects with device any time we open the app

### Actual Behavior
When testing with more then 50 devices then App not able to connect with device

### Steps to reproduce

1. Implement Light CTL server model in BLE Mesh
2. Provision more then 50 devices in single network using nRF Mesh application
3. connect to network using nRF Mesh app ( try multiple times)
## Other items if possible

-  sdkconfig file ([sdkconfig.txt](https://github.com/espressif/esp-idf/files/9054358/sdkconfig.txt))
-  elf file ([elf-file.txt](https://github.com/espressif/esp-idf/files/9054359/elf-file.txt))

JiaLWang, 2022-07-07:
Hi @hipal31，

>`sometimes I am not able to connect with device`

Can you reconnect the node after disconnecting?

Did the node print any error log? It's better to provide the log of the node.

hipal31, 2022-07-07:
Yes I am able to re-connect to device.

No node didn't print any error log.

Actually I don't know in which file connection request is received.
I tries ble-mesh log level to debug level but I am not able to identify in it.

JiaLWang, 2022-07-07:
Can you find the node connected to the phone among the 50 devices？
If you can find it. You can monitor the log of the device through the serial port.

hipal31, 2022-07-08:
hey @JiaLWang 
Can you tell me in which layer I will get ble-mesh connection request?

[image]

hipal31, 2022-07-14:
Hey @JiaLWang 
I have attached log file here [inc-relay-buffer.txt](https://github.com/espressif/esp-idf/files/9111425/inc-relay-buffer.txt) in which after getting error app is not able to connect to device and if app connect with device then device doesn't receive command from app.
Please look into logs and help me with this, As production is on halt because of this problem.

hipal31, 2022-09-22:
Hey @JiaLWang 
Can you tell me an approximate date for solving the mesh stack stuck issue?

hipal31, 2022-12-20:
Hey @JiaLWang any update on this?

hipal31, 2024-06-13:
Hey @JiaLWang Any update?

JiaLWang, 2024-06-13:
Hi @hipal31  ,
Sorry for the long delay in getting back to you and many thanks for being patient.

I will arrange for colleagues to support the issue ASAP.

JiaLWang, 2024-06-13:
Hi @forx157 ,

Please support this issue with high priority. Thanks.

hipal31, 2024-06-17:
Hi @forx157 Please help me with the issue

forx157, 2024-06-17:
Hi, @hipal31 
Please select a node as the test node, in the `bt_mesh_proxy_server_init` function in the `proxy_server.c` file, set the device name of this node to a different name from the other nodes, for example, "ESP-BLE-MESH-NO-RELAY", after finishing the setup, turn off the relay function of this node in menuconfig according to the following path `(Top) → Component config → ESP BLE Mesh Support → Relay support`. After the above settings, please test whether the node is unable to reconnect after disconnection.

closed 2025-03-31 by Alvin1Zhang