espressif/esp-idf #9300: esp-idf programming guide for v4.4.1 (stable) "Get Started" fails on Mac

buckybots, 2022-07-06:
followed simple directions on new Mac.  going in circles between export.sh and install.sh

1. completed the Mac tools setup
2. installed esp-idf :  git clone -b v4.4.1 --recursive https://github.com/espressif/esp-idf.git
3. tried to complete tools setup, but can not get idf.py to be recognized, as tools will not set up due to missing xtensa elf files etc. as seen below.  
4. I tried and retried this setup a couple of times.  
5. The instructions are straightforward, but there is a problem with this 'stable' release on Mac.   I also run on Windows machine, and that setup works fine. 

here are the errors:
eric@erics-Mac-mini esp-idf % ./install.sh
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.9.12
"python3" has been detected
Installing ESP-IDF tools
Selected targets are: esp32s3, esp32c3, esp32h2, esp32s2, esp32
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
WARNING: directory for tool xtensa-esp32-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
Installing xtensa-esp32-elf@esp-2021r2-patch3-8.4.0
file xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /Users/eric/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0
WARNING: directory for tool xtensa-esp32s2-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
Installing xtensa-esp32s2-elf@esp-2021r2-patch3-8.4.0
file xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /Users/eric/.espressif/dist/xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/eric/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0
WARNING: directory for tool xtensa-esp32s3-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
Installing xtensa-esp32s3-elf@esp-2021r2-patch3-8.4.0
file xtensa-esp32s3-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /Users/eric/.espressif/dist/xtensa-esp32s3-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/eric/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0
WARNING: directory for tool riscv32-esp-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
Installing riscv32-esp-elf@esp-2021r2-patch3-8.4.0
file riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /Users/eric/.espressif/dist/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/eric/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0
WARNING: directory for tool esp32ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
Installing esp32ulp-elf@2.28.51-esp-20191205
file binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /Users/eric/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/eric/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205
WARNING: directory for tool esp32s2ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
Installing esp32s2ulp-elf@2.28.51-esp-20191205
file binutils-esp32s2ulp-macos-2.28.51-esp-20191205.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /Users/eric/.espressif/dist/binutils-esp32s2ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/eric/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205
WARNING: directory for tool openocd-esp32 version v0.11.0-esp32-20211220 is present, but tool was not found
Installing openocd-esp32@v0.11.0-esp32-20211220
file openocd-esp32-macos-0.11.0-esp32-20211220.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /Users/eric/.espressif/dist/openocd-esp32-macos-0.11.0-esp32-20211220.tar.gz to /Users/eric/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220
Installing Python environment and packages
Python 3.9.12
pip 22.0.4 from /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pip (python 3.9)
Installing Python packages from /Users/eric/esp/esp-idf/requirements.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 4)) (62.1.0)
Requirement already satisfied: click>=7.0 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 8)) (8.1.3)
Requirement already satisfied: pyserial>=3.3 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 12)) (37.0.4)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 18)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 19)) (0.28)
Requirement already satisfied: idf-component-manager~=1.0 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 20)) (1.1.4)
Requirement already satisfied: gdbgui==0.13.2.0 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 22)) (0.13.2.0)
Requirement already satisfied: pygdbmi<=0.9.0.2 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 25)) (0.9.0.2)
Requirement already satisfied: python-socketio<5 in /Users/eric/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (from -r /Users/eric/esp/esp-idf/requirements.txt (line 28)) (4.6.1)
[... 178 lines cut ...]

eric@erics-Mac-mini hello_world % . /Users/eric/esp/esp-idf/export.sh
Detecting the Python interpreter
Checking "python" ...
Checking "python3" ...
Python 3.9.12
"python3" has been detected
Adding ESP-IDF tools to PATH...
WARNING: directory for tool xtensa-esp32-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
ERROR: tool xtensa-esp32-elf has no installed versions. Please run '/Users/eric/esp/esp-idf/install.sh' to install it.
WARNING: directory for tool xtensa-esp32s2-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/Users/eric/esp/esp-idf/install.sh' to install it.
WARNING: directory for tool xtensa-esp32s3-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
ERROR: tool xtensa-esp32s3-elf has no installed versions. Please run '/Users/eric/esp/esp-idf/install.sh' to install it.
WARNING: directory for tool riscv32-esp-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
ERROR: tool riscv32-esp-elf has no installed versions. Please run '/Users/eric/esp/esp-idf/install.sh' to install it.
WARNING: directory for tool esp32ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
ERROR: tool esp32ulp-elf has no installed versions. Please run '/Users/eric/esp/esp-idf/install.sh' to install it.
WARNING: directory for tool esp32s2ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
ERROR: tool esp32s2ulp-elf has no installed versions. Please run '/Users/eric/esp/esp-idf/install.sh' to install it.
WARNING: directory for tool openocd-esp32 version v0.11.0-esp32-20211220 is present, but tool was not found
ERROR: tool openocd-esp32 has no installed versions. Please run '/Users/eric/esp/esp-idf/install.sh' to install it.
eric@erics-Mac-mini hello_world % idf.py
zsh: command not found: idf.py
eric@erics-Mac-mini hello_world %

dobairoland, 2022-07-07:
Hi @buckybots. What chip architecture is used in your Mac?

Could you please try to run one of the tools manually and post here the output? For example:

```sh
~/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version

xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

buckybots, 2022-07-08:
Thanks for looking at this!  Here is what I get:

eric@erics-Mac-mini hello_world % ~/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version
zsh: bad CPU type in executable: /Users/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
eric@erics-Mac-mini hello_world % 

using the apple help I get :
Hardware Overview:

  Model Name:	Mac mini
  Model Identifier:	Macmini9,1
  Chip:	Apple M1
  Total Number of Cores:	8 (4 performance and 4 efficiency)
  Memory:	16 GB
  System Firmware Version:	7429.61.2
  OS Loader Version:	7429.61.2

This is macOS Monterey
Version 12.1

Mac mini(M1,2020)
Chip Apple M1
Memory 16GB

I've tried several versions of esp-idf, none of them work, master, release 4.4, release 4.3, stable 4.4.1

buckybots, 2022-07-08:
I am trying to use the visual studio code setup.  I tried the express setup.  I think you are correct that there is a CPU error.  Here is the part of the output window during the visual studio code setup :

idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
/bin/sh: /Users/eric/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc: Bad CPU type in executable
Command failed: xtensa-esp32s3-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc: Bad CPU type in executable
Error checking xtensa-esp32s3-elf version
Error: Command failed: xtensa-esp32s3-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc: Bad CPU type in executable

/bin/sh: /Users/eric/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc: Bad CPU type in executable
Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc: Bad CPU type in executable
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc: Bad CPU type in executable

/bin/sh: /Users/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Bad CPU type in executable
Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Bad CPU type in executable
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Bad CPU type in executable

/bin/sh: /Users/eric/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc: Bad CPU type in executable
Command failed: riscv32-esp-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc: Bad CPU type in executable
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc: Bad CPU type in executable

/bin/sh: /Users/eric/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin/esp32s2ulp-elf-as: Bad CPU type in executable
Command failed: esp32s2ulp-elf-as --version
/bin/sh: /Users/eric/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin/esp32s2ulp-elf-as: Bad CPU type in executable
Error checking esp32s2ulp-elf version
Error: Command failed: esp32s2ulp-elf-as --version
/bin/sh: /Users/eric/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin/esp32s2ulp-elf-as: Bad CPU type in executable

/bin/sh: /Users/eric/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/openocd: Bad CPU type in executable
Command failed: openocd --version
/bin/sh: /Users/eric/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/openocd: Bad CPU type in executable
Error checking openocd-esp32 version
Error: Command failed: openocd --version
/bin/sh: /Users/eric/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/openocd: Bad CPU type in executable

/bin/sh: /Users/eric/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as: Bad CPU type in executable
Command failed: esp32ulp-elf-as --version
/bin/sh: /Users/eric/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as: Bad CPU type in executable
Error checking esp32ulp-elf version
Error: Command failed: esp32ulp-elf-as --version
/bin/sh: /Users/eric/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as: Bad CPU type in executable

idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
/bin/sh: /Users/eric/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as: Bad CPU type in executable
Command failed: esp32ulp-elf-as --version
/bin/sh: /Users/eric/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as: Bad CPU type in executable
Error checking esp32ulp-elf version
Error: Command failed: esp32ulp-elf-as --version
/bin/sh: /Users/eric/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin/esp32ulp-elf-as: Bad CPU type in executable

/bin/sh: /Users/eric/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc: Bad CPU type in executable
Command failed: riscv32-esp-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc: Bad CPU type in executable
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
/bin/sh: /Users/eric/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc: Bad CPU type in executable

[... cut ...]
/bin/sh: /Users/eric/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin/esp32s2ulp-elf-as: Bad CPU type in executable
Command failed: esp32s2ulp-elf-as --version
/bin/sh: /Users/eric/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin/esp32s2ulp-elf-as: Bad CPU type in executable
Error checking esp32s2ulp-elf version
Error: Command failed: esp32s2ulp-elf-as --version
/bin/sh: /Users/eric/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin/esp32s2ulp-elf-as: Bad CPU type in executable

dobairoland, 2022-07-08:
Thank you for the additional info. I've updated the title to reflect the cause of the issue. We will get back to you....

antmak, 2022-07-08:
Hi @buckybots. It seems something wrong with Rosetta environment on the mac m1.

Could you check and share output follow up commands, please:

```bash
arch -x86_64 ~/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version
```

```bash
arch -arm64 ~/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version
```

```bash
lsbom -f /Library/Apple/System/Library/Receipts/com.apple.pkg.RosettaUpdateAuto.bom

```

Normally, all IDF Tools run with Rosetta translator on M1 computers. Rosetta should be installed in system.

buckybots, 2022-07-08:
eric@erics-Mac-mini ~ % arch -x86_64 ~/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version
arch: posix_spawnp: /Users/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Bad CPU type in executable
eric@erics-Mac-mini ~ % arch -arm64 ~/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version
arch: posix_spawnp: /Users/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Bad CPU type in executable
eric@erics-Mac-mini ~ % lsbom -f /Library/Apple/System/Library/Receipts/com.apple.pkg.RosettaUpdateAuto.bom

can't open /Library/Apple/System/Library/Receipts/com.apple.pkg.RosettaUpdateAuto.bom: No such file or directory
**** Can't open /Library/Apple/System/Library/Receipts/com.apple.pkg.RosettaUpdateAuto.bom.
eric@erics-Mac-mini ~ %

buckybots, 2022-07-08:
I don't know about Rosetta translator.  I did try IDF Tools v4.3 and master, they did not work either.

buckybots, 2022-07-08:
Googled Roseeta on Mac, got the following, I will try to install it and try again.

If you need to install Rosetta on your Mac
Rosetta 2 enables a Mac with Apple silicon to use apps built for a Mac with an Intel processor.

How to install Rosetta
Rosetta 2 is available only for [Mac computers with Apple silicon](https://support.apple.com/kb/HT211814).

You're asked to install Rosetta the first time you open an [app that needs Rosetta](https://support.apple.com/en-us/HT211861#needsrosetta). Click Install, then enter your user name and password to allow installation to proceed.

Alert: To open app, you need to install Rosetta. Do you want to install it now?

If you click Not Now, you will be asked again the next time you open an app that needs Rosetta.

If Rosetta is already installed, you are not asked to install it again.

How to use Rosetta
Rosetta is not an app that you open or interact with. Rosetta works automatically in the background whenever you use an app that was built only for Mac computers with an Intel processor. It translates the app for use with Apple silicon.

In most cases, you won't notice any difference in the performance of an app that needs Rosetta. But you should [contact the app developer](https://support.apple.com/kb/HT201777) to inquire about a version that can natively use the full power and performance of Apple silicon.

Which apps need Rosetta?
To identify apps that need Rosetta or can use Rosetta:

Select an app in the Finder.
From the File menu in the menu bar, choose Get Info.
See the information labeled Kind:
Application (Intel) means the app supports only Intel processors and needs Rosetta to work on a Mac with Apple silicon.
Application (Universal) means the app supports both Apple silicon and Intel processors, and uses Apple silicon by default. Universal apps don't need Rosetta.
Safari Info window with "Open using Rosetta" selected

For apps labeled Application (Universal), the Info window includes the setting “Open using Rosetta.” This setting enables a universal app such as a web browser to use plug-ins, extensions, or other add-ons that haven't been updated to support Apple silicon. If a universal app doesn't recognize an add-on that you installed for the app, you can quit the app, select this setting, and try again.

Published Date: May 17, 2022

buckybots, 2022-07-08:
I went to install Rosetta, and I see that the app, that is somewhere in the esp-idf scripts that are run, it should note that this is an M1 and ask for Rosetta to be installed.  Then the user (me) would opt in, and then I guess the would work.

Here is the explanation from the above help that seems to suggest that approach.

_You're asked to install Rosetta the first time you open an_ [app that needs Rosetta](https://support.apple.com/en-us/HT211861#needsrosetta). Click Install, then enter your user name and password to allow installation to proceed.
        
Alert: To open app, you need to install Rosetta. Do you want to install it now?
        
If you click Not Now, you will be asked again the next time you open an app that needs Rosetta.

let me know if I should switch back to my pc now, and at some point when the scripts are fixed to ask for Rosetta 2,  try again.

Thanks.

closed 2022-07-09 by buckybots

antmak, 2022-07-11:
@buckybots thanks a lot for that cli recipe!