espressif/esp-idf #9301: New Features

I-AM-Tushar-chauhan, 2022-07-06:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

I-AM-Tushar-chauhan, 2022-07-06:
Hello sir,
can you please send me the documentation of changes between version esp-idf-4.0 and version  esp-idf-4.4? because sometimes my data will erase automatically. Mainly I need to know the changes between NVS, spiffs and flash memory.

Dazza0, 2022-07-08:
@I-AM-Tushar-chauhan Please see our [release notes](https://github.com/espressif/esp-idf/releases) for changes made by each iterative release. If there's something missing from the release notes, please feel free to reopen this issue.

closed 2022-07-08 by Dazza0