espressif/esp-idf #9303: WPA-enterprise reauthentication is not working

raysvmc, 2022-07-06:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version v4.4.1
- Build System:        idf.py
- Compiler version xtensa-esp32-elf esp-2021r2-patch3-8.4.0
- Operating System:     Windows
- Using an IDE?: VS Code 1.68.1
- Power Supply:         USB

## WPA enterprise authenticates reliably but never reauthenticates

We can authenticate through phase 1 and 2 and the 4 way handshake, but the reauth gets through to the very last handshake and fails.  We've collected debug logs from the esp32 and logs from the router. We've tried a variety of versions, code, and hardware, including the enterprise example code. This example is for simple test code based on the example code.

### Expected behavior: Reauthentication completes

### Actual behavior: The fourth and last handshake, the response to the third handshake, is never sent.

### Steps to reproduce:

1. Configure an esp32 with code to connect to an enterprise network
2. Configure the network to reauthenticate periodically
3. Allow it to connect and authenticate, and wait for the reauth.
4. Observe  sending 3/4 msg of 4-Way Handshake from the router and corresponding entries from esp-idf, near the line wifi:recv deauth, reason=0x2

We are using a Netgear WAX620 router and freeRADIUS 3.0.25 running on linux mint.

### Code to reproduce this issue

```cpp

//After this is called there is just an infinite loop along with a simple event handler function. If needed,
//I will share that too.

bool WIFI_Connect(void)
{
    esp_err_t err;

    printf("WIFI_Connect: connecting\n");

    err = esp_netif_init();

[... 194 lines cut ...]

"6/1/2022 11:51:50 AM",D (312095) wifi:stop CSA timer
"6/1/2022 11:51:50 AM",W (312105) wifi:<ba-del>idx
"6/1/2022 11:51:50 AM",D (312105) wifi:remove 6c:cd:d6:5f:a9:c1 from rc list
"6/1/2022 11:51:50 AM",I (312105) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
"6/1/2022 11:51:50 AM",D (312115) wifi:filter: set rx policy=8
"6/1/2022 11:51:50 AM",D (312115) wifi:Send disconnect event, reason=2, AP number=0
"6/1/2022 11:51:50 AM",D (312125) event: running post WIFI_EVENT:5 with handler 0x400e89d0 and context 0x3ffbde88 on loop 0x3ffbcc54[0m
"6/1/2022 11:51:50 AM",D (312135) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbdb80 from event_id=5[0m
"6/1/2022 11:51:50 AM",D (312145) esp_netif_lwip: check: remote, if=0x3ffbdb80 fn=0x400dd494
"6/1/2022 11:51:50 AM",[0m
"6/1/2022 11:51:50 AM",D (312155) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbdb80[0m
"6/1/2022 11:51:50 AM",D (312155) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbdb80[0m
"6/1/2022 11:51:50 AM",D (312165) esp_netif_lwip: if0x3ffbdb80 start ip lost tmr: no need start because netif=0x3ffbdc00 interval=120 ip=0[0m
"6/1/2022 11:51:50 AM",D (312175) esp_netif_lwip: check: local, if=0x3ffbdb80 fn=0x400ddc6c
"6/1/2022 11:51:50 AM",[0m
"6/1/2022 11:51:50 AM",D (312185) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbdb80[0m
"6/1/2022 11:51:50 AM",V (312185) esp_netif_objects: esp_netif_next_unsafe 0x0[0m
"6/1/2022 11:51:50 AM",V (312195) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbdb80[0m
"6/1/2022 11:51:50 AM",V (312195) esp_netif_objects: esp_netif_next_unsafe 0x3ffbdb80[0m
"6/1/2022 11:51:50 AM",D (312205) esp_netif_lwip: call api in lwip: ret=0x0, give sem[0m
"6/1/2022 11:51:50 AM",D (312215) event: running post WIFI_EVENT:5 with handler 0x400d67f4 and context 0x3ffc4e74 on loop 0x3ffbcc54[0m
"6/1/2022 11:51:50 AM",[wifi] Event ................: STA_DISCONNECTED

```

hpdhabe, 2022-07-12:
Hi @raysvmc,
Can you please attach the debug logs and a sniffer capture to this issue if possible? It will be really helpful in debugging the issue.
Please make sure the flag `WPA_DEBUG_PRINT` is enabled before capturing the debug logs. This can be found under the `Supplicant` component in `Component Config` menu of `menuconfig`.
If you need more help with capturing 82011 packets, refer to the following link: https://github.com/espressif/esp-idf/issues/8687#issuecomment-1105070843

raysvmc, 2022-07-12:
Thank you @hpdhabe for looking into this. We're trying to get set back up to make another test but it might take a week or so for various reasons I won't bore you with. I have set WPA_DEBUG_PRINT and configured a laptop for monitor mode sniffing.

AxelLin, 2022-08-10:
@raysvmc  Any update regarding this issue?

raysvmc, 2022-08-10:
Apologies for the delay @hpdhabe and @AxelLin , we've had a series of other problems in the way, but it seems like we're almost ready to make another test. I will push for that.

AxelLin, 2022-09-06:
@raysvmc 
Just wondering if this is still an issue or not.
If you still hit the issue, please provide debug logs and a sniffer capture. see https://github.com/espressif/esp-idf/issues/9303#issuecomment-1181643113

raysvmc, 2022-09-06:
@AxelLin this is very much an issue for us and our customers. I'm unable to perform the test myself as I am remote and don't have the system setup here or the knowledge to set it up. Two other engineers have been performing the tests, and I have been asking repeatedly that it get done ASAP. They have been working on it, but between other emergencies, vacations, and trouble with the router configuration, it's not quite done. I will continue pushing for it and I apologize for the delay. I'm hoping this week!

raysvmc, 2022-09-09:
We have a test finally @hpdhabe @AxelLin and I will share:
capture5-9.9.2022.1555.txt from the debug port
messages.txt from the router
The packet capture .pcap is not being allowed - I will investigate.
The behavior is a bit different but it's failing the same way as far as I can tell.
[capture5-9.9.2022.1555.txt](https://github.com/espressif/esp-idf/files/9539212/capture5-9.9.2022.1555.txt)
[messages.txt](https://github.com/espressif/esp-idf/files/9539213/messages.txt)

raysvmc, 2022-09-09:
[(Acrylic) VMC-ENTERPRISE-1_9.9.2022.1555.csv](https://github.com/espressif/esp-idf/files/9539220/Acrylic.VMC-ENTERPRISE-1_9.9.2022.1555.csv)

raysvmc, 2022-09-09:
[WAX620-NETGEAR-VMC br-lan-2020-03-03__164958-capture.csv](https://github.com/espressif/esp-idf/files/9539224/WAX620-NETGEAR-VMC.br-lan-2020-03-03__164958-capture.csv)
This is a capture of the router's network. I don't know if this is at all helpful.

Shreyas0-7, 2022-09-22:
Hey @raysvmc 
Can you please retry with this patch 
[wpa_reauth.zip](https://github.com/espressif/esp-idf/files/9625282/wpa_reauth.zip)

raysvmc, 2022-09-22:
Shreyas0-7, thanks for the suggestion, we tried that and had about the same results I'm afraid.
[AuthIssue092222.zip](https://github.com/espressif/esp-idf/files/9629657/AuthIssue092222.zip)

Shreyas0-7, 2022-09-23:
Hey @raysvmc is it possible for you to log hostapd logs as well?
You can do it by passing arguments -dd and -K while starting hostapd.

raysvmc, 2022-09-23:
It looks like we have that log. Is this it?
[hostapd.txt](https://github.com/espressif/esp-idf/files/9634894/hostapd.txt)

Shreyas0-7, 2022-09-28:
Hey @raysvmc thanks for your patience and your co-ordination, can you please retry with this patch.
[wpa2_reauth.zip](https://github.com/espressif/esp-idf/files/9668973/wpa2_reauth.zip)

raysvmc, 2022-09-29:
Thanks @Shreyas0-7, I will try to compile that today and it might be tested tomorrow.

se-k32, 2022-10-02:
Hi, I am joining this conversation because I think we have the same problem (apologize if not; in that case, I will create a new post). One of our clients has a problem that all ESP32 devices gets disconnected precisely 3 hours after being connected to AP. AP is using WPA2-PSK authentication (different as used by @raysvmc). The disconnect reason is WIFI_REASON_AUTH_EXPIRE (2) or sometimes WIFI_REASON_NOT_AUTHED (6). After that, devices reconnect to AP and the same thing happens again after 180 minutes. We think that the AP requests some kind of re-authentication, but ESP fails to respond to it, so AP kicks him off. We are using esp-idf 4.4.3. We tried the patch from wpa2_reauth.zip, but sadly it did not help. We don't have precise logs, but we are happy to help testing it.

AxelLin, 2022-10-02:
@se-k32 
Could you create a new issue?
Please provide sniffer log and device verbose log if possible, and what is the AP brand/model?
BTW, esp-idf-v4.4.3 is not yet available, please provide correct version (git describe --tags).

se-k32, 2022-10-02:
Yes no problem, I have created it (https://github.com/espressif/esp-idf/issues/9896#issue-1393698360). I thought that the solution/problem is the same as this one. Also sorry for the wrong esp-idf version, it should be 4.4.2.

Shreyas0-7, 2022-10-03:
Thanks for retry @raysvmc. By applying this patch we were able to fix this bug in our test environment but as this was also seen again in [9896](https://github.com/espressif/esp-idf/issues/9896). To investigate it furthermore we will require sniffer capture on the particular channel on which AP is turned on. Also in sniffer capture it is observed that only beacon frames are seen. Please make sure that you are able to capture the handshake also. Thank you.

raysvmc, 2022-10-04:
Thanks @Shreyas0-7 and I apologize for not checking the sniffer capture. We're working on that.

AxelLin, 2022-10-14:
> Thanks @Shreyas0-7 and I apologize for not checking the sniffer capture. We're working on that.

Any update regarding the sniffer log? @raysvmc

raysvmc, 2022-10-14:
@AxelLin unfortunately not yet. There has been some kind of difficulty in making this work and I will see if I can help. Thanks for asking!

raysvmc, 2022-10-27:
@Shreyas0-7 @AxelLin this last test seems to be working now, but I will share what we got on this one in case it's of interest. We'll have to see if this is repeatable. There are two tests, each of which reauths successfully with a few handshake resends. I summarized them in NotesFromSerialCapture.txt. Thanks!
[AuthIssue102722.zip](https://github.com/espressif/esp-idf/files/9883240/AuthIssue102722.zip)

raysvmc, 2022-10-28:
@Shreyas0-7 @AxelLin we've run two tests, and the second one was fine. I'm sharing the first one which worked sometimes but had different problems authenticating. There are two reboots so it's kind of three tests in one and I wrote up a text file cross-referencing the logs. I hope you can follow it, as I had a difficult time. There's a LAN capture but I didn't share it - please let me know if that would help.
[AuthIssue102722a.zip](https://github.com/espressif/esp-idf/files/9891040/AuthIssue102722a.zip)

AxelLin, 2022-11-08:
@Shreyas0-7 Any update on this issue? Did you check above logs?

Shreyas0-7, 2022-11-09:
@raysvmc @AxelLin As seen in logs and sniffer capture AP sends EAPOL message 3 of 4 twice with different replay counters, we reply to the first one encrypting with older keys but AP expects 4 of 4 for the latest replay counter to which we reply encrypting using new keys. This situation cannot be handled at this moment, also you can observe the same behavior in upstream.

AxelLin, 2022-11-10:
> Thanks for retry @raysvmc. By applying this patch we were able to fix this bug in our test environment but as this was also seen again in [9896](https://github.com/espressif/esp-idf/issues/9896). 

@Shreyas0-7 
https://github.com/espressif/esp-idf/issues/9896#issuecomment-1272018890 set esp_wifi_set_inactive_time(WIFI_IF_STA, 10); to avoid the issue.
Is that a reasonable change?  Any bad side-effect with setting inactive time to 10 sec?
Or is that actually a different issue?

Shreyas0-7, 2022-11-11:
@AxelLin Issue you mentioned is completely different and setting esp_wifi_set_inactive_time(WIFI_IF_STA, 10); will have no effect here. 
@raysvmc Yes AP should have waited longer to send message 3 of 4 again. By upstream behavior, I meant wpa_supplicant which is used by Linux.

AxelLin, 2022-11-13:
@Shreyas0-7 

Then is this https://github.com/espressif/esp-idf/issues/9303#issuecomment-1261472280 a vaild fix? (since you already confirm it fixed the bug in your test environment.)
The fix is not yet available in any branches, so it's not clear to me if you need more testing or it's not a good fix?

Shreyas0-7, 2022-11-13:
@AxelLin The patch I provided contained the fix. I have raised the corresponding MR and it's under testing, it will be available once it gets merged.

raysvmc, 2022-11-14:
@Shreyas0-7 these might be dumb questions, but, first, I applied both patches wpa_reauth and wpa2_reauth; should I have only applied one at a time?
Second, I understand that the AP is not waiting long enough, but is there another problem with the wpa_supplicant used by Linux, and is this the RADIUS server? Thanks for your patience!

AxelLin, 2022-11-17:
@Shreyas0-7 
Fix for v4.4:  24efd4a505c27d97c8bccfa9e8c7a6258536b852 
I think v4.3 also need the fix.

kapilkedawat, 2022-11-18:
> @Shreyas0-7 these might be dumb questions, but, first, I applied both patches wpa_reauth and wpa2_reauth; should I have only applied one at a time? 

Yes, Please use final fix (https://github.com/espressif/esp-idf/commit/24efd4a505c27d97c8bccfa9e8c7a6258536b852)

> Second, I understand that the AP is not waiting long enough, but is there another problem with the wpa_supplicant used by Linux, and is this the RADIUS server? Thanks for your patience!
What we mean is, same issue is present in upstream linux based distributions as well. the solution will be to keep both new key and old key in hardware till device confirms that the new key will be used for further connection which cannot be supported.

> I think v4.3 also need the fix.
Yes, the fix has been backported and will be available shortly.

raysvmc, 2022-11-18:
@kapilkedawat Thank you!

AxelLin, 2022-12-03:
@kapilkedawat @Shreyas0-7 
Looks like v5.0 tag does not include this fix?

Shreyas0-7, 2022-12-03:
The fix is backported to v5.0 and is merged recently

closed 2023-04-27 by nachiketkukade