espressif/esp-idf #9305: seem like have memory leakage when call "i2s_write" at version release/4.4

zhangoneone, 2022-07-06:
## Environment

- Development Kit:      [none]
- Kit version (DevKitC): [none]
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version (run ``git describe --tags`` to find it):
    v4.4.1-117-g8c8f700c1d
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
call function i2s_write, there are memory leakage 24 bytes every time that I observed.
free heap[248824 bytes],minimum heap[248824 bytes]
free heap[248800 bytes],minimum heap[248776 bytes]
free heap[248776 bytes],minimum heap[248752 bytes]

### Expected Behavior
heap memory not leak

### Actual Behavior
heap memory leak

### Steps to reproduce
repeat to call i2s_write and ensure it return successful.
show the heap memory info everytime.

### Code to reproduce this issue

please refer from 
https://github.com/espressif/esp-idf/blob/c2ccc383dae2a47c2c2dc8c7ad78175a3fd11361/components/driver/deprecated/i2s_legacy.c   line 1630,

according to review the code, maybe there are queue content need to be free, by checking the content data structure,
it is located at lldesc.h ,"struct lldesc_s", which is just right 24 bytes length. so I guess maybe that the queue content was not be free on a suitable time.
 could anyone help check the issue ? thanks!

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

L-KAYA, 2022-07-07:
Hi @zhangoneone , before looking into this issue, may I ask why the reference link is a file only exist in v5.0?

L-KAYA, 2022-07-07:
I have tested the `i2s_write` on both master(v5.0) and v4.4, seems no memory leak here. Could you check again if the v4.4 environment is clean and transmit data in right way?

zhangoneone, 2022-07-07:
thank your @L-KAYA  reply,  supplement more info

this is debug output log:

I (881) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=8
I (882) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=8
I (885) I2S: I2S0, MCLK output by GPIO0
I (915) ES8311: ES8311 in Slave mode and I2S format
free heap[248776 bytes],minimum heap[248776 bytes]
E (1381) I2S: dma error, interrupt status: 0x00002200
free heap[248752 bytes],minimum heap[248752 bytes]
free heap[248728 bytes],minimum heap[248728 bytes]

zhangoneone, 2022-07-07:
debug log indicate there are dma transmit error occur,  not sure how it impact the i2s_write. would there hit  some unpopular code path ?

L-KAYA, 2022-07-07:
Seems you have modified the `i2s_es8311` example. The error log you mentioned is not a common error, it should never occur. I need to a further test for it. Could you provide the details about how you modified that example?

zhangoneone, 2022-07-07:
[test_demo.zip](https://github.com/espressif/esp-idf/files/9062869/test_demo.zip)
Hi @L-KAYA , I detached a test demo that reproduced the issue, if any problem please let me know,thanks

zhangoneone, 2022-07-07:
this test demo base on idf sdk es8311 example but have a little modify

zhangoneone, 2022-07-07:
fixed, root cause is freertos msg queue "send operation" do not status check. it lead to memory have been reserved, but send fail, then it has not opportunity to be free.

thank your reply again! I'll close the issue

closed 2022-07-07 by zhangoneone