espressif/esp-idf #9308: rtl8201 no mem for receive buffer

chrisovergaauw, 2022-07-07:
## Environment

Intellij IDE
IDF 4.1.1 (relevant code is exactly the same code as in latest)
micropython latest
ESP-32-WROOM-32
AsyncTCP Server example
lan phy: rtl8201

## Problem Description
Exact same issue as #9083 but then for `emac_esp32_rx_task` in `esp_eth_mac_esp.c`
after "no mem for receive buffer" it seems like no memory will never be freed. Any socket interaction in MP will hang after this, but the code is clearly an ESP-IDF issue.

//Detailed problem description goes here.

### Expected Behavior
malloc for rx received bytes instead (similar to #9083) of ETH_MAX_PACKET_SIZE (this will reduce how often the error occurs)
and memory should be freed in order to recover from this scenario.

kostaond, 2022-09-01:
@chrisovergaauw could you please confirm you mean https://github.com/me-no-dev/AsyncTCP by "AsyncTCP Server example"?

thiagohd, 2022-09-15:
I have the same problem, but with my own firmware.

I have implemented a UDP logging functionality, where every logging from the ESP-32 will be broadcasted to the network.

And the same firmware, but disabling the UDP logging system, doesn't have this issue. 

It looks like EMAC phy is trying to pass all the UDP messages received, and it fills up the memory faster than it can consume it.

```
static int init_udp_socket(){
    int b_sock, ret;
    if ((b_sock = socket(PF_INET, SOCK_DGRAM, IPPROTO_UDP)) < 0){
        printf("ERR socket(): %d\n", b_sock);
        return 0;
    }
    
    int broadcastPermission = 1;
    if ((ret = (setsockopt(b_sock, SOL_SOCKET, SO_BROADCAST, (void *) &broadcastPermission, sizeof(broadcastPermission)))) < 0){
        printf("ERR setsockopt() = %d\n", ret);
        return 0;
    }

    int opts;
    opts = fcntl(b_sock, F_GETFL);
    if(opts < 0){
        printf("ERR fcntl() = %d\n", opts);
        return 0;
    }

    opts = (opts | O_NONBLOCK);
    if((ret = (fcntl(b_sock,F_SETFL,opts))) < 0){
        printf("ERR fcntl() = %d\n", ret);
        return 0;
    }
    return b_sock;
}
static int check_udp_socket(int b_sock){
    int opts;

    if(b_sock == 0) return 0;

    opts = fcntl(b_sock, F_GETFL);
    if(opts < 0){
        printf("fcntl get failed\n");
        return 0;
    }
    return 1;
}
static int send_udp_packet(int b_sock, char *str, int len){
    int ret;
    struct sockaddr_in tx_addr;

    tx_addr.sin_family = AF_INET;                 
    tx_addr.sin_addr.s_addr = inet_addr("255.255.255.255");
    tx_addr.sin_port = htons(UDP_LOG_PORT);
    if((ret = sendto(b_sock, str, len, 0, (struct sockaddr *)&tx_addr, sizeof(tx_addr))) < 0)
        printf("ERR sento(): %d\n", ret);

    return ret;
}
int send_udp(const char *fmt, ...){
    if(!check_udp_socket(__xudp_socket))
        __xudp_socket = init_udp_socket();
    
    char udp_log_buff[UDP_LOG_LINE_SIZE] = {0};
    va_list va;
	va_start(va, fmt);
	int xlen = vsnprintf(udp_log_buff, UDP_LOG_LINE_SIZE, fmt, va);
	va_end(va);
    if(xlen < 0){
        printf("Erro ao formatar mensagem de log UDP\n");
    }else{
        xlen = (xlen >= UDP_LOG_LINE_SIZE) ? UDP_LOG_LINE_SIZE : xlen;
    }
    return send_udp_packet(__xudp_socket, udp_log_buff, xlen);
}
```

kostaond, 2022-10-31:
Could you please check if [this WIP update](https://github.com/kostaond/esp-idf/tree/feature/emac_buff_optimization) helped to resolve your issue?

kostaond, 2022-11-01:
@chrisovergaauw thank you for very quick response and testing! Just to summarize, you now don't see `no mem for receive buffer` anymore? Once HTTPS communication hangs, are you able to receive other traffic? It would be great if you were able to describe steps (or provide code snippet) so I could reproduce the issue.

chrisovergaauw, 2022-11-01:
I don't see it currently, but that might be due to the fact that my Micropython code hangs on a socket write action.
That means that the ESP-IDF code underneath probably does not do much either.
I cannot tell you whether receiving other traffic still works, but the `esp.emac receive` message do keep on coming in, so that's a good sign. Sorry, I have not yet been able to reproduce this without our company's micropython code base, which is far too complex to serve as a debug example. I am still working on creating a reproducible example.

kostaond, 2022-11-01:
Try to ping the device once the HTTPS gets stuck.

chrisovergaauw, 2022-11-01:
it still responds to pinging properly 👍

Alvin1Zhang, 2023-01-12:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/785f154f5652b353c460ec47cb90aa404207918c, feel free to reopen.

closed 2023-01-12 by Alvin1Zhang