espressif/esp-idf #9309: libsodium component fails with new idf-component-manager on esp-idf v4.4

dot4qu, 2022-07-07:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      none
- Module or chip used: N/A
- IDF version (run ``git describe --tags`` to find it):
  v4.4
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    8.4.0
- Operating System:    macOS
- Using an IDE?: No
- Power Supply: N/A

## Problem Description

idf-component-manager has an issue with libsodium component on esp-idf v4.4 but not v5.0. No board/chip necessary, error happens in build/compilation.

### Expected Behavior

Component downloaded and compiled in both.

### Actual Behavior

Compiles fine when using latest master branch of esp-idf, but `idf build` fails on v4.4.

The libsodium dependency was added with the command specified in the component manager registry (https://components.espressif.com/component/espressif/libsodium): 
`idf.py add-dependency espressif/libsodium==1.0.20`

This is the v4.4 error output:
```
CMake Error at /Users/user/Developer/esp/esp-idf/tools/cmake/component.cmake:238 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "libsodium" candidates: "libsodium, espressif__libsodium"

Call Stack (most recent call first):
  /Users/user/Developer/esp/esp-idf/tools/cmake/build.cmake:491 (__component_get_requirements)
  /Users/user/Developer/esp/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:14 (project)
```

Here is the idf_component.yml file located in the main/ directory:

```
## IDF Component Manager Manifest File
# This file is only required since ESP-IDF v5.
dependencies:
  espressif/libsodium: "==1.0.20"
  ## Required IDF version
  idf:
    version: ">=4.1"
  # # Put list of dependencies here
  # # For components maintained by Espressif:
  # component: "~1.0.0"
  # # For 3rd party components:
  # username/component: ">=1.0.0,<2.0.0"
  # username2/component2:
  #   version: "~1.0.0"
  #   # For transient dependencies `public` flag can be set.
  #   # `public` flag doesn't have an effect dependencies of the `main` component.
  #   # All dependencies of `main` are public by default.
  #   public: true
```

Here is the generated depencies.lock file:
```
dependencies:
  espressif/libsodium:
    component_hash: 36e9832f40c909c0a9135a3e775fcbdd7a2a13d47a1c052ee5b21499eb3eddb5
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.0.20
  idf:
    component_hash: null
    source:
      type: idf
    version: 5.0.0
manifest_hash: 733ca58a2c360c30a6a892a31857491a3db091336ff54fabba4a5fc066b6af06
target: esp32
version: 1.0.0
```

kumekay, 2022-07-11:
@dot4qu Could you please try manifest like this:

```
dependencies:
  espressif/libsodium:
    version: "==1.0.20"
    rules:
      - if: "idf_version >=5.0"
  idf:
    version: ">=4.1"
```

`libsodium` was a part of ESP-IDF till 4.4 and was removed in 5.0. To avoid solver confusion, you can add it conditionally.

dot4qu, 2022-07-12:
Ah I spoke too soon. This fixes the issue for esp-idf v4.4, but not for the v4.3 release container. What's odd is that if I check out the release/v4.3 tag locally I can build, but when our CI is building with the release-4.3 container it fails. This is the error after running `idf.py build`:
```
Your environment is not configured to handle unicode filenames outside of ASCII range. Environment variable LC_ALL is temporary set to C.UTF-8 for unicode support.
[66](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:68)
Executing action: all (aliases: build)
[67](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:69)
Running cmake in directory /app/examples/mpd_status/build
[68](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:70)
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 /app/examples/mpd_status"...
[69](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:71)
-- Found Git: /usr/bin/git (found version "2.17.1") 
[70](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:72)
4 problems were found in the manifest file /app/examples/mpd_status/main/idf_component.yml:
[71](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:73)
Invalid manifest format
[72](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:74)
Invalid dependency format
[73](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:75)
Dependency version spec format is invalid
[74](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:76)
Unknown keys in dependency details: rules
[75](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:77)
CMake Error at /opt/esp/idf/tools/cmake/project.cmake:209 (message):
[76](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:78)
Call Stack (most recent call first):
[77](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:79)
  /opt/esp/idf/tools/cmake/project.cmake:296 (__project_init)
[78](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:80)
  CMakeLists.txt:5 (project)
[79](https://github.com/vroland/epdiy/runs/7307004385?check_suite_focus=true#step:3:81)
```

kumekay, 2022-07-12:
@dot4qu IDF release-v4.3 docker image wasn't rebuilt for a while due to another issue. It's already fixed and now passing our internal CI. Should be updated soon

dot4qu, 2022-07-12:
@kumekay ah got it, thanks! I'll keep an eye out for the update over the next couple of days.

dot4qu, 2022-07-25:
Problem is fixed with the new image pushed 2 days ago, thanks!

closed 2022-07-25 by dot4qu

law-ko, 2023-01-27:
@dot4qu Can you provide more insights on what you did? I did not have any `idf_component.yml` file under the project directory until I performed `idf.py reconfigure` but `idf_component.yml` is only shown under `managed_components/espressif__libsodium` directory.

I am using `release/v4.4` and version is `ESP-IDF v4.4.3-347-g9ee3c8337d`

Thank you.

kumekay, 2023-01-27:
Hello @dot4qu 

Could you please explain the problem you are facing? If you are using `libsodium` with ESP-IDF v4.4 you don't need `idf_component.yml` in your project's main component. You only need it if you want to make a project for ESP-IDF >=v5.0

law-ko, 2023-01-27:
Hi @kumekay 

I am using ESP-IDF v4.4 but having issues when trying to build the project:
```
Processing 2 dependencies:
[1/2] espressif/libsodium (1.0.20)
[2/2] idf (4.4.3)
-- DEBUG: Use esp-modbus component folder: /home/espuser/esp/esp-idf/components/freemodbus.
CMake Error at /home/espuser/esp/esp-idf/tools/cmake/component.cmake:237 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "libsodium" candidates: "libsodium, espressif__libsodium"

Call Stack (most recent call first):
  /home/espuser/esp/esp-idf/tools/cmake/build.cmake:486 (__component_get_requirements)
  /home/espuser/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:20 (project)

-- Configuring incomplete, errors occurred!
See also "/home/espuser/esp/esp-idf/examples/example_project/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

I was previously on ESP-IDF v4.4.2 and did not have this issue until v4.4.3, which is the one from `release/v4.4` branch now

Thank you

kumekay, 2023-01-27:
@law-ko I've tried to reproduce the problem with ESP-IDF v4.4.3-347-g9ee3c8337d, but without luck. Do you have any components in the `components` directory or [EXTRA_COMPONENT_DIRS](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/api-guides/build-system.html#optional-project-variables) set for your project?

law-ko, 2023-01-27:
@kumekay Yes, the components directory from `esp-homekit-sdk` is included in the [EXTRA_COMPONENT_DIRS](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/api-guides/build-system.html#optional-project-variables)

```
set(EXTRA_COMPONENT_DIRS
						 ["${CMAKE_CURRENT_LIST_DIR}/../../../../libraries/homekit/homekit"](https://github.com/espressif/esp-homekit-sdk/tree/master/components/homekit)
   )
```

kumekay, 2023-01-27:
@law-ko What version of the homekit are you using?

`esp_hap_core` used to have a manifest causing problems with ESP-IDF v4.4. It was fixed back in May in this commit:
https://github.com/espressif/esp-homekit-sdk/commit/df56006edf48c2c7304cf8873fb7cbac104f3879