espressif/esp-idf #9310: WIFI_PROV_MGR code "qrcode.h" file directory error

Noumeer, 2022-07-07:
fatal error: qrcode.h: No such file or directory
#include "qrcode.h"
Unable to compile the code.

mahavirj, 2022-07-08:
@Noumeer 

`qrcode` is now moved to component manager, please see [here](https://components.espressif.com/component/espressif/qrcode). It is used in example through following configuration:

https://github.com/espressif/esp-idf/blob/495d35949d50033ebcb89def98f107aa267388c0/examples/provisioning/wifi_prov_mgr/main/idf_component.yml#L2

Which example you are using? Please share your environment details.

Noumeer, 2022-07-08:
I am using "wifi_prov_magr" example in vscode esp idf.

brianignacio5, 2022-07-14:
Are you using the ESP-IDF extension ? Did you enable the component manager?

Noumeer, 2022-07-18:
Its working now

Alvin1Zhang, 2022-07-19:
Glad to hear that it is working, feel free to reopen.

closed 2022-07-19 by Alvin1Zhang