espressif/esp-idf #9312: BLE_MESH: No matching TX context for ack when debug relay node esp-ble-mesh

xuanchibkd14, 2022-07-08:
Hi,Im a new esp32, I'm testing esp-ble-mesh sensor server node with relay node enable, my config is:
~~~
static esp_ble_mesh_cfg_srv_t config_server = {
    .relay = ESP_BLE_MESH_RELAY_ENABLED,
    .beacon = ESP_BLE_MESH_BEACON_ENABLED,
#if defined(CONFIG_BLE_MESH_FRIEND)
    .friend_state = ESP_BLE_MESH_FRIEND_ENABLED,
#else
    .friend_state = ESP_BLE_MESH_FRIEND_NOT_SUPPORTED,
#endif
#if defined(CONFIG_BLE_MESH_GATT_PROXY_SERVER)
    .gatt_proxy = ESP_BLE_MESH_GATT_PROXY_ENABLED,
#else
    .gatt_proxy = ESP_BLE_MESH_GATT_PROXY_NOT_SUPPORTED,
#endif
    .default_ttl = 7,
    /* 3 transmissions with 20ms interval */
    .net_transmit = ESP_BLE_MESH_TRANSMIT(2, 20),
    .relay_retransmit = ESP_BLE_MESH_TRANSMIT(2, 20),
};
~~~

but i only get:
~~~
I (296465) BLE_MESH: Settings store, waiting 0 seconds
I (296473) BLE_MESH: send, app_idx 0xfffe src 0x0004 dst 0x0001
I (296477) BLE_MESH: send, len 9: 803e00040000000011
I (296484) BLE_MESH: Settings store, waiting 0 seconds
I (00:04:53.744) ble_mesh: ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND
I (00:04:53.745) ble_mesh: elem_addr 0x0004, app_idx 0x0000, cid 0xffff, mod_id 0x1100
I (306990) BLE_MESH: net_idx 0x000 iv_index 0x00000000 flags 0x00
W (307963) BLE_MESH: No matching TX context for ack
W (308256) BLE_MESH: No matching TX context for ack
W (313032) BLE_MESH: No matching TX context for ack
W (313130) BLE_MESH: No matching TX context for ack
W (314740) BLE_MESH: No matching TX context for ack
W (315714) BLE_MESH: No matching TX context for ack
W (315861) BLE_MESH: No matching TX context for ack
W (316153) BLE_MESH: No matching TX context for ack
~~~
please help me
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ]x I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.4
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 8.4
- Operating System:     [macOS]
- Using an IDE?: [No]
- Power Supply:         [USB|external 5V]

xuanchibkd14, 2022-07-13:
any can help me?

JiaLWang, 2022-07-15:
Hi @xuanchibkd14 

Please refer to `https://github.com/espressif/esp-idf/blob/master/docs/en/api-guides/esp-ble-mesh/ble-mesh-faq.rst#54-what-is-the-meaning-of-warning-no-matching-tx-context-for-ack`.

xuanchibkd14, 2022-08-15:
> Hi @xuanchibkd14
> 
> Please refer to `https://github.com/espressif/esp-idf/blob/master/docs/en/api-guides/esp-ble-mesh/ble-mesh-faq.rst#54-what-is-the-meaning-of-warning-no-matching-tx-context-for-ack`.

Hi, i need the solution, please don't explain any information i can google

JegadeeshXaglabs, 2022-08-18:
Hi @xuanchibkd14 
Did u find the solution.

xuanchibkd14, 2022-08-19:
@JegadeeshXaglabs not yet, but if you turn on relay node by 
~~~
.relay = ESP_BLE_MESH_RELAY_ENABLED
~~~
it will work fine without any log debug, i tested.
[Change BT_DBG-> BT_INFOR](https://github.com/espressif/esp-idf/blob/4d14c2ef2d9d08cd1dcbb68a8bb0d76a666e2b4b/components/bt/esp_ble_mesh/mesh_core/net.c#L1230) you will see this work

JiaLWang, 2022-08-22:
Hi @xuanchibkd14 

Do you mean that the node receives the message and does not forward it?

JegadeeshXaglabs, 2022-08-24:
> @JegadeeshXaglabs not yet, but if you turn on relay node by
>
> ```
> .relay = ESP_BLE_MESH_RELAY_ENABLED
> ```
>
> it will work fine without any log debug, i tested. [Change BT_DBG-> [...]

Still it shows the same issue

xuanchibkd14, 2022-08-24:
> Hi @xuanchibkd14
> 
> Do you mean that the node receives the message and does not forward it?

sorry, i want to know how to relay work, but this warning is stupid

forx157, 2022-08-25:
hi, I've tested this example, and the relay function is normal and doesn't have to do anything.
you can add a log in the file `components/bt/esp_ble_mesh/mesh_core/net.c : bt_mesh_net_relay`。That function is used to relay the message.

JegadeeshXaglabs, 2022-08-25:
> > > @JegadeeshXaglabs not yet, but if you turn on relay node by
> > > ```
> > > .relay = ESP_BLE_MESH_RELAY_ENABLED
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > it will work fine without any log debug, i tested. [Change BT_DBG-> BT_INFOR](https://github.com/espressif/esp-idf/blob/4d14c2ef2d9d08cd1dcbb68a8bb0d76a666e2b4b/components/bt/esp_ble_mesh/mesh_core/net.c#L1230) you will see this work
> > 
> > 
> > Still it shows the same issue
> 
> can you show your models test

Vendor module In ESP32 Ble mesh

JegadeeshXaglabs, 2022-08-25:
> It should also be noted that the log level in the menuconfig is greater than the log level you use, and the setting path of log level in the menuuconfig is: component config -> ESP BLE Mesh Support -> BLE Mesh STACK DEBUG LOG LEVEL, which is set to greater than your log level.

Do i have to edit the log level or ?

forx157, 2022-08-25:
> 
I'm just saying this to make sure your logs are output properly :)

forx157, 2022-08-25:
> Do i have to edit the log level or ?

I'm just saying this to make sure your prints are output properly. :)

JegadeeshXaglabs, 2022-08-25:
> I'm just saying this to make sure your prints are output properly. :)

yes i am sure if i disconnect 2nd node i can see the msg in 1st node form a client.

forx157, 2022-08-25:
> yes i am sure if i disconnect 2nd node i can see the msg in 1st node form a client.

So can you see that the relay function is working properly ?

JegadeeshXaglabs, 2022-08-25:
> 

How can i check the relay function is working properly

forx157, 2022-08-25:
> > 
> 
> How can i check the relay function is working properly

Start by preparing three nodes. Then add a log to the `bt_mesh_net_relay` function, such as `BT_INFO ("Relay")`. Finally, download the program to the chip,and two of them are used as server nodes and one as the client node, and then after the mesh net is successfully established, you can try to restart a server (the purpose is to make this node send a message), so that the relay log can be seen in the other server, so that you can prove that the relay function is working properly.

xuanchibkd14, 2022-08-25:
> > 
> 
> How can i check the relay function is working properly

you need three nodes, two of them have client/server vender program, the other have client on/off program (it has relay node enable so you don't need to edit the code)(need provisioner in a network)
First you need to let two nodes vender models transmit data in close distance, then move them away from each other until the data can no longer be transmitted. After you can put the other node in between them and turn it on, you will see 2 vender model nodes continue to transmit data

JegadeeshXaglabs, 2022-08-26:
> you need three nodes, two of them have client/server vender program, the other have client on/off program (it has relay node enable so you don't need to edit the code)(need provisioner in a network) [...]

I can t send any msg from on/off client to vendor server. But i tried on/off server and client program.it relays msg perfectly.

forx157, 2022-08-29:
> 

have you let them form a network in your on/off example?

forx157, 2022-08-29:
> I can t send any msg from on/off client to vendor server. But i tried on/off server and client program.it relays msg perfectly.

have you let them form a network in your on/off example?

JegadeeshXaglabs, 2022-08-29:
> > 
> 
> have you let them form a network in your on/off example?

yes i form mesh network on/off example.

forx157, 2022-08-29:
> > > 
> > 
> > 
> > have you let them form a network in your on/off example?
> 
> yes i form mesh network on/off example.

Could you give me a detailed description of how you did the onoff experiment?

xuanchibkd14, 2022-08-29:
> I can t send any msg from on/off client to vendor server. But i tried on/off server and client program.it relays msg perfectly.

do you custom for vendor server with on/off client to a model like client/server, there default not same model so can't send is normal. but relay node is working so good

JegadeeshXaglabs, 2022-08-29:
> do you custom for vendor server with on/off client to a model like client/server, there default not same model so can't send is normal. but relay node is working so good

did you mean combination of client and server in one program?

xuanchibkd14, 2022-08-29:
> did you mean combination of client and server in one program?

i mean on/off client with on/off server, sensor model client with sensor model server. they can talk to each. other. and other model cant talk with, except you custom in code

JegadeeshXaglabs, 2022-08-29:
I am using vendor module client and vendor module server (2 nodes).With that i cant receive clients msg in both the server nodes.

xuanchibkd14, 2022-08-29:
@JegadeeshXaglabs can you show your log, im test it work on example

JegadeeshXaglabs, 2022-08-29:
[image]

xuanchibkd14, 2022-08-29:
@JegadeeshXaglabs and your log client?

JegadeeshXaglabs, 2022-08-29:
[image]
Right one is client

JegadeeshXaglabs, 2022-08-29:
[image]
if i connect second node This is what i got from first node (left side)

JegadeeshXaglabs, 2022-08-30:
Two of address

JegadeeshXaglabs, 2022-08-30:
Even i tried in sensor module. Faced same problem

xuanchibkd14, 2022-08-30:
you should try in vendor example, and do not enable ble mesh storage

JegadeeshXaglabs, 2022-08-30:
> you should try in vendor example, and do not enable ble mesh storage

Where can i disable ble mesh storage

JegadeeshXaglabs, 2022-08-30:
disabled ble mesh storage faced same problem

forx157, 2022-08-31:
Could you share your code with us?

JegadeeshXaglabs, 2022-08-31:
<script src="https://gist.github.com/JegadeeshXaglabs/0532bf9afca329eed938e6e7a47e6f04.js"></script>

JegadeeshXaglabs, 2022-08-31:
https://gist.github.com/JegadeeshXaglabs/b1fc5089d40ff961cf99ef5879082b37

JegadeeshXaglabs, 2022-09-07:
> your 2 server have address is 0x0005 and 0x0006, do you send to two of address, or only one?

Can you tell me how to send a msg to two server. Do i have to add these address in vendor client side ?