espressif/esp-idf #9313: Issue when porting project for ESP32 to ESP32-S3

krupis, 2022-07-08:
ESP-IDF version used: 4.4.1

Problem description:

I have initially built my project on ESP32. The project involves BLE and WiFi, I have also used .cpp for some components just to learn c++.

I have changed the build target to ESP32-S3 and tried to build my project. I had some confilcts about the GPIO pins that I have managed to fix easily, however there are a few errors that I am not sure about:
[image]

There seems to be a lot of issues regarding touch_pad_intr_mask_t. Could someone point me in the right direction how this can be solved.

0xjakob, 2022-07-13:
@krupis Could you share the command line output of the compiler? The full build log output for the file in question would be great. Alternatively, a little example that reproduces the issue would be great, too.

krupis, 2022-07-13:
> @krupis Could you share the command line output of the compiler? The full build log output for the file in question would be great. Alternatively, a little example that reproduces the issue would be great, too.

The same issue is described here:
https://github.com/espressif/esp-idf/issues/5699

Tommorow I will try to share the output logs

0xjakob, 2022-07-25:
@krupis Are you including any IDF header files inside an `extern "C" { ... }` block?

krupis, 2022-07-25:
> @krupis Are you including any IDF header files inside an `extern "C" { ... }` block?

Yes I definately am. At the beggining of my project, I have decided to write my code in cpp. This probably was not the best idea. Every component that I have in my project is .cpp component apart from a few components that I could not get to build if I name them cpp ( one of those is my WiFi component where I have all wifi functions from the esp-idf. 

What is the correct way to mix c and cpp components? For example if I have ADC component which is cpp and then I have WiFi component which is C. If both components are dependant on each other, what is correct way to include headers to one another?

krupis, 2022-07-25:
When I try to make my wifi component cpp,  I get the following error:
```
    wifi_config_t wifi_config = {
        .ap = {
            //.ssid = EXAMPLE_ESP_WIFI_SSID_AP,
            //.ssid_len = strlen(EXAMPLE_ESP_WIFI_SSID_AP),
            .password = EXAMPLE_ESP_WIFI_PASS_AP,
            .ssid_len = strlen(device_info.ID),
            .channel = EXAMPLE_ESP_WIFI_CHANNEL,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK,
            .max_connection = EXAMPLE_MAX_STA_CONN,
        },
    };
```

```
../components/WIFI/WIFI.cpp:231:5: error: C99 designator 'password' outside aggregate initializer
```

0xjakob, 2022-07-25:
@krupis Please try to not include IDF headers inside `extern "C" {...}` blocks. Outside these blocks, but inside your C++ code, it should be fine. If you are normally including IDF header files from your C++ code but still get an error, it's probably a bug.

The reason why we do this is that some of our header files contain C++ code which won't work if *both* they are included from C++ code *and* they are declared as `extern "C"`.

0xjakob, 2022-07-25:
If you are able to switch to master, the code shown above should probably work. We enabled C++20 recently which allows "aggregate initialization". For v4.4 and below, C++11 is default which doesn't allow "aggregate initialization".

krupis, 2022-07-25:
Can you confirm if I understand this correctly?:
This is my custom NVS component that I have created. It is NVS.cpp and NVS.h

Not correct:

```
#ifndef NVS_H
#define NVS_H

#ifdef __cplusplus
extern "C" {
#endif

#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "nvs.h"
#include "stdio.h"
#include "Thermostat.h"
#include "string.h"

#ifdef __cplusplus
}
#endif

#endif
```

Correct:

```
#ifndef NVS_H
#define NVS_H

#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "nvs.h"
#ifdef __cplusplus
extern "C" {
#8732 

#include "stdio.h"
#include "Thermostat.h"
#include "string.h"

#ifdef __cplusplus
}
#endif

#endif
```

0xjakob, 2022-07-25:
@krupis yes, this is correct. If you have problems with the "Correct" when including the IDF headers, please let us know!

BTW., there is a [C++ API for NVS](https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/include/nvs_handle.hpp).

ESP-Marius, 2025-01-17:
Seems like the initial question here was answered. Marking this one as closed.

closed 2025-01-17 by ESP-Marius