espressif/esp-idf #9314: esp32-cam, send images to webserver continuously

anhay1999, 2022-07-08:
Hello!
I am currently writing code to make my esp32-cam send images to web continuously but I encounter this error:
W (54) boot.esp32: PRO CPU has been reset by WDT.
W (54) boot.esp32: WDT reset info: PRO CPU PC=0x40083cc3
W (56) boot.esp32: WDT reset info: APP CPU PC=0x4009089d
The error keeps appearing, I've been fixing it for days but couldn't find out a solution. I can show you all my code if it is necessary, but for brevity, i have figured out this is from the while loop, for example, if my boot.py looks like this:

import machine
from machine import Pin
import camera
import uos
import shutil

uos.mount(machine.SDCard(), "/sd") #mount the SD card
camera.init(0, format=camera.JPEG)
camera.quality(10)
camera.framesize(9)
file = open("sd/"+"a"+".jpg", "wb")
file.write(camera.capture())
file.close()
shutil.copy("sd/a.jpg","static/a.jpg")

#you can find shutil.py here:[https://github.com/micropython/micropyt ... /shutil.py](https://github.com/micropython/micropython-lib/blob/master/python-stdlib/shutil/shutil.py). But actually I have add a little function called "copy" to my shutil.py that I would show you later. This function copies file from a folder to another folder
#the reason why I need to store image in sd folder/SD card then copy it static folder is that I want my web take images from static folder. If I directly store it to static folder without store it to the sd folder first, the error would appear!

The above code would work without any errors, but if I want to continuously capture picture so that I put my take-a-picture-code-lines in a while loop, the error would appear, just like this:

import machine
from machine import Pin
import camera
import uos
import shutil

uos.mount(machine.SDCard(), "/sd") #mount the SD card
camera.init(0, format=camera.JPEG)
camera.quality(10)
camera.framesize(9)
while True:
file = open("sd/"+"a"+".jpg", "wb")
file.write(camera.capture())
file.close()
shutil.copy("sd/a.jpg","static/a.jpg")

Here's my shutil.py if you need:

import os

def rmtree(top):
for path, dirs, files in os.walk(top, False):
for f in files:
os.unlink(path + "/" + f)
os.rmdir(path)

def copyfileobj(src, dest, length=512):
if hasattr(src, "readinto"):
buf = bytearray(length)
while True:
sz = src.readinto(buf)
if not sz:
break
if sz == length:
dest.write(buf)
else:
b = memoryview(buf)[:sz]
dest.write(b)
else:
while True:
buf = src.read(length)
if not buf:
break
dest.write(buf)
def copy(src, dest):
with open(src, 'rb') as src_file:
with open(dest, 'wb') as dest_file:
copyfileobj(src_file, dest_file)

I would absolutely appreciate if someone could help me, it's a bit frustrating when working for days but couldn't find a solution. Thanks in advance!

NathanJPhillips, 2022-07-14:
This code doesn't look like it uses ESP-IDF so your problem probably isn't caused by an issue in the IDF repository. I think your request would be better served in a forum specific to your programming environment.

anhay1999, 2022-07-15:
> This code doesn't look like it uses ESP-IDF so your problem probably isn't caused by an issue in the IDF repository. I think your request would be better served in a forum specific to your programming environment.

thank you! my bad.

devosavchenko, 2022-07-15:
Hi, @anhay1999,
The Core 0 was reset because you did not reset the watchdog counter before it end. In C code to prevent this need to call esp_task_wdt_reset_user(...) function, or disable watchdog (not recommended). 
Please check your configuration and try to understand wich function executed longer than watchdog counting end.

Alvin1Zhang, 2022-07-18:
Thanks for reporting and help share the notes, feel free to reopen.

closed 2022-07-18 by Alvin1Zhang

anhay1999, 2022-07-21:
> Hi, @anhay1999, The Core 0 was reset because you did not reset the watchdog counter before it end. In C code to prevent this need to call esp_task_wdt_reset_user(...) function, or disable watchdog [...]

Thank you! i will try it.