espressif/esp-idf #9316: ble50_security_server example ESP32-S3 device not showing on LightBlue

krupis, 2022-07-08:
Hello. I have programmed the ESP32-S3 development board using the ble50_security_server example. I can see that the device programmed successfully and started:
```
I (24) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (25) boot: compile time 13:38:15
I (25) boot: chip revision: 0
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=1ed90h (126352) map
I (115) esp_image: segment 1: paddr=0002edb8 vaddr=3fc939f0 size=01260h (  4704) load
I (116) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=74c0ch (478220) map
I (207) esp_image: segment 3: paddr=000a4c34 vaddr=3fc94c50 size=01e3ch (  7740) load
I (208) esp_image: segment 4: paddr=000a6a78 vaddr=40374000 size=0f9f0h ( 63984) load
I (226) esp_image: segment 5: paddr=000b6470 vaddr=50000000 size=00010h (    16) load
I (233) boot: Loaded app from partition at offset 0x10000
I (233) boot: Disabling RNG early entropy source...
I (246) cpu_start: Pro cpu up.
I (247) cpu_start: Single core mode
I (255) cpu_start: Pro cpu start user code
I (255) cpu_start: cpu freq: 160000000
I (255) cpu_start: Application information:
I (257) cpu_start: Project name:     ble50_sec_gatts_demo
I (263) cpu_start: App version:      v4.4.1-dirty
I (269) cpu_start: Compile time:     Jul  8 2022 13:37:43
I (275) cpu_start: ELF file SHA256:  1d52bee3247301f4...
I (281) cpu_start: ESP-IDF:          v4.4.1-dirty
I (286) heap_init: Initializing. RAM available for dynamic allocation:
I (294) heap_init: At 3FC9A8B8 len 00045748 (277 KiB): D/IRAM
I (300) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (307) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (313) spi_flash: detected chip: generic
I (318) spi_flash: flash io: dio
W (322) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (335) sleep: Configure to isolate all GPIO pins in sleep state
I (342) sleep: Enable automatic switching of GPIO sleep configuration
I (349) coexist: coexist rom version e7ae62f
I (354) cpu_start: Starting scheduler on PRO CPU.
W (365) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (365) BT_INIT: BT controller compile version [d913766]
I (375) phy_init: phy_version 501,94697a2,Apr  7 2022,20:49:08
I (475) system_api: Base MAC address is not set
I (475) system_api: read default base MAC address from EFUSE
I (475) BT_INIT: Bluetooth MAC: 34:b4:72:70:00:9a

I (485) SEC_GATTS_DEMO: app_main init bluetooth
I (505) SEC_GATTS_DEMO: ESP_GATTS_REG_EVT
I (505) SEC_GATTS_DEMO: The number handle = 8
I (505) SEC_GATTS_DEMO: ESP_GAP_BLE_SET_LOCAL_PRIVACY_COMPLETE_EVT, tatus = 0
I (515) SEC_GATTS_DEMO: ESP_GAP_BLE_EXT_ADV_SET_PARAMS_COMPLETE_EVT status 0
I (515) SEC_GATTS_DEMO: ESP_GAP_BLE_EXT_ADV_DATA_SET_COMPLETE_EVT status 0
I (525) SEC_GATTS_DEMO: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0

```

I have downloaded LightBlue app and I expect to see a device with a name "ESP_BLE50_SERVER" because that is configured on the raw_adv_data:
```
static uint8_t ext_adv_raw_data[] = {
        0x02, 0x01, 0x06,
        0x02, 0x0a, 0xeb, 0x03, 0x03, 0xab, 0xcd,
        0x11, 0X09, 'E', 'S', 'P', '_', 'B', 'L', 'E', '5', '0', '_', 'S', 'E', 'R', 'V', 'E', 'R',
};
```

However, there is no such device on the LightBlue.

I have decided to test if BLE is working properly and tried another example gatt_server_service_table and that worked without any issues. I am able to see the device on the LighBlue.

Could someone point me in the right direction? What are the possible causes of 5.0 example not working?

Weijian-Espressif, 2022-07-12:
@krupis Please check whether the phone and app supports Bluetooth 5.0

krupis, 2022-07-12:
> @krupis Please check whether the phone and app supports Bluetooth 5.0

My phone is Samsung galaxy s8 edge, from what I know it supports ble 5. I havent been able to find any information on the LightBlue app though

Weijian-Espressif, 2022-07-12:
You can use nRF connect APP. The latest version of the APP supports Bluetooth 5.0 and displays hardware information in the Device Information option.

Weijian-Espressif, 2022-07-12:
@krupis  https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp

krupis, 2022-07-12:
> @krupis https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp

Sadly it does not seem to show up on my phone NRF connect app.

However, I can see this packet being transmitted on wire shark:
[image]

Does that mean that my phone cannot capture BLE 5?

krupis, 2022-08-02:
> I'm not sure, the phone and APP need to support Bluetooth 5.0 to scan ADV, you can try other apps

Hello. Its been a while. I have been further testing this with LightBlue and NRF connect app with different phones and there is my results:

1. **Using wire shark connected to NRF52840 :**

[image]

2. Using LightBlue app on the Samsung galaxy S8 edge Android version 9. The ESP32 BLE does not show up in the scanned  device list. 
[image]

3. Using NRF connect app on the Samsung galaxy S8 edge Android version 9. The ESP32 BLE does not show up in the scanned device list

[image]

4. Using Lightblue app on the Pocophone F1 Android version 9
[image]

5. Using NRF connect on Pocophone F1 android version 9
[image]

6. Using Lightblue on Huawei P10 android version 9
[image]

7. Using NRF Connect on Huawei P10 android version 9
[image]

8. Using NRF Connect on IPhone 12 pro max
[image]

According to my results, only the Iphone 12 pro max and Pocohpone F1 were able to capture the ESP32 5.0 BLE device.

Is BLE 5.0 not backwards compatible with BLE 4? I was under impression that if the phone does not support BLE 5.0 I would still be able to connect to this device

What if I am developing code with ESP32-S3 and ble 5.0 and the client develop app for it. There is no possibility to connect to my device via phones that do not support ble 5.0?

krupis, 2022-08-02:
Temporary closed moved to #9480

krupis, 2022-08-02:
Temporary closed moved to https://github.com/espressif/esp-idf/issues/9480

closed 2022-08-02 by krupis