espressif/esp-idf #9317: esp_wifi_sta_get_ap_info does not return connected SSID

NathanJPhillips, 2022-07-08:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32 & ESP32-S2
- IDF version (run ``git describe --tags`` to find it): v4.4-3-g5402e26ace
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt.
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

`esp_wifi_sta_get_ap_info` does not return connected SSID but configured SSID instead.

### Expected Behavior

Running the minimal reproducer below should produce the following output:
```
I (2025) test-esp_wifi_sta_get_ap_info: Configured with network 'Number24'
I (2025) test-esp_wifi_sta_get_ap_info: Currently connected to 'Number24'
I (2075) test-esp_wifi_sta_get_ap_info: Configured with network ''
I (2075) test-esp_wifi_sta_get_ap_info: Currently connected to 'Number24'
```

### Actual Behavior

Running the minimal reproducer below produces the following output:
```
I (2025) test-esp_wifi_sta_get_ap_info: Configured with network 'Number24'
I (2025) test-esp_wifi_sta_get_ap_info: Currently connected to 'Number24'
I (2075) test-esp_wifi_sta_get_ap_info: Configured with network ''
I (2075) test-esp_wifi_sta_get_ap_info: Currently connected to ''
```

### Steps to reproduce

1. Run the minimal reproducer below.
2. Observe output.

### Code to reproduce this issue

https://gist.github.com/NathanJPhillips/84d08ae24878838d396e569c95d15d70

mhdong, 2022-07-11:
hi @NathanJPhillips  You called `esp_wifi_set_config`, did not call `esp_wifi_connect` to make the setting take effect, so ssid obtained by `esp_wifi_sta_get_ap_info` is incorrect. You can use `esp_wifi_sta_get_ap_info` after `esp_wifi_connect` to obtain ap info.

NathanJPhillips, 2022-07-12:
Hi @mhdong - you have exactly described the problem. The documentation says `esp_wifi_sta_get_ap_info` should "Get information of AP which the ESP32 station is associated with." but it doesn't when `esp_wifi_set_config` has been called without a call to `esp_wifi_connect`. In fact it returns an invalid mix of information about the AP the ESP32 is associated with (e.g. its RSSI) and information provided to `esp_wifi_set_config` (e.g. the SSID). To fix this you need to store and return the valid SSID of the currently connected AP separately to the information altered by `esp_wifi_set_config`. I would happily provide a PR but I believe this code is not included in the IDF source.

mhdong, 2022-07-13:
@NathanJPhillips yes, this code is not included in the IDF source. We will consider to fix this issue in a later release.

MaxwellAlan, 2022-07-26:
Hi @NathanJPhillips

> Will the fix only be applied to new devices? Is it part of the code in the ROM of the chip?

No, all chips can be fixed later.

NathanJPhillips, 2022-10-19:
Will this ticket be left open to track this issue or is there somewhere else I can check for progress on this?

NathanJPhillips, 2022-11-15:
It seems this may take some time to fix. Could someone provide some pointers in the meantime about where I can find the relevant code that I would need to change to fix this?

mhdong, 2022-11-15:
Hi @NathanJPhillips  I'm sorry I didn't deal with this problem in a timely manner.
You can save ssid when you handler a WIFI_EVENT_STA_CONNECTED event.
`/** Argument structure for WIFI_EVENT_STA_CONNECTED event */
typedef struct {
    uint8_t ssid[32];         /**< SSID of connected AP */
    uint8_t ssid_len;         /**< SSID length of connected AP */
    uint8_t bssid[6];         /**< BSSID of connected AP*/
    uint8_t channel;          /**< channel of connected AP*/
    wifi_auth_mode_t authmode;/**< authentication mode used by AP*/
} wifi_event_sta_connected_t;`

I have prepared relevant fixes internally. When the fix is merged, fix commit will be synchronized here.

Alvin1Zhang, 2023-01-12:
Thanks for reporting and sorry for slow turnaround, fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/1c0d0e505d9fc50045bce4625a8c95aaf55af362, feel free to reopen.

closed 2023-01-12 by Alvin1Zhang