espressif/esp-idf #9318: serial output using ESP32-S3 usb controller is unreliable

davidzuhn, 2022-07-09:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:     ESP32-S3-DevKit-C-1
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version (run ``git describe --tags`` to find it):  v5.0-dev-4001-g495d35949d

- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:   ac OS 12
- Using an IDE?:  No
- Power Supply:  USB

## Problem Description

My application uses the console component, and provides a number of commands.   When I run 'help' at the console when using the UART for the console (via the CP2102N), the output is completely repeatable and formatted as expected.   When I use the USB Serial/JTAG controller for the console (as I expect to do when my "real" hardware arrives), the output is janky [that's the technical term :)].   Some newlines seem to be eaten, and some lines get omitted altogether.   

### Expected Behavior

When I use the UART console (connected via /dev/cu.SLAB_USBtoUART), this is what I get every time I run 'help'

```
>help
fs 
  data fs command

haptic 
  Haptic feedback

state 
  Display current connection state values

throttle 
  throttle support

wifi 
  wifi support

battery 
  notify battery level

tasks 
  Show tasks

event 
  Print event loop debug

reset 
  Reset the device (optionally to factory default)

version 
  Display version information

board 
  read board values

knob 
  View/set the knob position levels

help 
  Print the list of registered commands
```

### Actual Behavior

When I run using the USB Serial/JTAG interface (via /dev/cu.usbmodem1101), and run the command again (and again), I see results like this:

```
>help
fs 
  data fs command

haptic 
  Haptic feedback

state 
  Display current connection state values

throttle 
  throttle support

wifi 
  wifi support

battery 
  notify battery level
tasks 
  Show tasks

event 
  Print event loop debug

reset 
  Reset the device (optionally to factory default)

version 
  Display version information

board 
  read board values

knob 
  View/set the knob position levels

help 
  Print the list of registered commands
>help
fs 
  data fs command

haptic 
  Haptic feedback

state 
  Display current connection state values

throttle 
  throttle support

  wifi support

battery 
  notify battery level

tasks 
  Show tasks

event 
  Print event loop debug

reset 
  Reset the device (optionally to factory default)

version 
  Display version information

board 
  read board values

knob 
  View/set the knob position levels

help 
  Print the list of registered commands

>help
fs 
  data fs command

haptic 
  Haptic feedback

  Display current connection state values

throttle 
  throttle support

wifi 
  wifi support

battery 
  notify battery level
tasks 
  Show tasks

event 
  Print event loop debug

reset 
  Reset the device (optionally to factory default)
  Display version information

board 
  read board values
  View/set the knob position levels

help 
  Print the list of registered commands

>help
fs 
  data fs command

haptic 
  Haptic feedback
state 
  Display current connection state values

throttle 
  throttle support

wifi 
  wifi support

  notify battery level

tasks 
  Show tasks

event 
  Print event loop debug

reset 
  Reset the device (optionally to factory default)

version 
  Display version information

board 
  read board values

  View/set the knob position levels

  Print the list of registered commands
```

Note these are not identical results for each time, and none of them match the "correct" text as seen on the UART.   

### Steps to reproduce

1. Run my program on ESP32-S3-DevKitC, using (alternately) UART & USB console.  I've tried to reproduce this on a small example, but haven't managed to see the same problems.   Two binaries are in the attached zip file, both will run on the ESP32-S3-DevKitC board with no further hardware attached.

### Code to reproduce this issue

## Debug Logs

## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
- [X] elf file in the ``build`` folder 
- [ N/A ] coredump (This provides stacks of tasks.)

[bug-report.zip](https://github.com/espressif/esp-idf/files/9075556/bug-report.zip)

moefear85, 2022-07-09:
i'm not an expert on this but did you try latest master? there was an issue with usb-cdc lines getting eaten, which was fixed about 2 weeks ago.

davidzuhn, 2022-07-09:
The version listed **is** the latest master (even still as of today - 09 July 2022).

moefear85, 2022-07-10:
ok, i have a hunch. can you apply this patch? it might solve the problem.

remove the .txt extension, place it in
> $IDF_PATH/components/tinyusb/tinyusb
and run this command:
>git apply usb.patch
[usb.patch.txt](https://github.com/espressif/esp-idf/files/9078763/usb.patch.txt)

igrr, 2022-07-10:
@moefear85 I'm afraid the file you are suggesting to modify is related to the software USB CDC driver for the USB_OTG peripheral. The reported issue is related to the USB_SERIAL_JTAG peripheral which has a hardware CDC implementation.

(I have modified the issue title to avoid confusion.)

moefear85, 2022-07-10:
oh ok, i don't have any experience with esp32-s3 yet. I have some modules lying around, but haven't broken them out yet (haven't found the time).

egnor, 2022-07-14:
To reduce variables, instead of your application, can you try the standard "basic console" example project? (At `esp-idf/examples/system/console/basic`.) Make sure to run `idf.py menuconfig` and set `ESP_CONSOLE_SERIAL_JTAG`.

That has its own "help" output, which gives me:

```
esp32s3> help
free 
  Get the current size of free heap memory

heap 
  Get minimum size of free heap memory that was available during program
  execution

version 
  Get version of chip and SDK

restart 
  Software reset of the chip

tasks 
  Get information about running tasks

deep_sleep  [-t <t>] [--io=<n>] [--io_level=<0|1>]
  Enter deep sleep mode. Two wakeup modes are supported: timer and GPIO. If no
  wakeup option is specified, will sleep indefinitely.
  -t, --time=<t>  Wake up time, ms
      --io=<n>  If specified, wakeup using GPIO with given number
  --io_level=<0|1>  GPIO level to trigger wakeup

light_sleep  [-t <t>] [--io=<n>]... [--io_level=<0|1>]...
  Enter light sleep mode. Two wakeup modes are supported: timer and GPIO.
  Multiple GPIO pins can be specified using pairs of 'io' and 'io_level'
  arguments. Will also wake up on UART input.
  -t, --time=<t>  Wake up time, ms
      --io=<n>  If specified, wakeup using GPIO with given number
  --io_level=<0|1>  GPIO level to trigger wakeup

join  [--timeout=<t>] <ssid> [<pass>]
  Join WiFi AP as a station
  --timeout=<t>  Connection timeout, ms
        <ssid>  SSID of AP
        <pass>  PSK of AP

nvs_set  <key> <type> -v <value>
  Set key-value pair in selected namespace.
Examples:
 nvs_set VarName i32 -v
  123 
 nvs_set VarName str -v YourString 
 nvs_set VarName blob -v
  0123456789abcdef 

         <key>  key of the value to be set
        <type>  type can be: i8, u8, i16, u16 i32, u32 i64, u64, str, blob
  -v, --value=<value>  value to be stored

nvs_get  <key> <type>
  Get key-value pair from selected namespace. 
Example: nvs_get VarName i32
         <key>  key of the value to be read
        <type>  type can be: i8, u8, i16, u16 i32, u32 i64, u64, str, blob

nvs_erase  <key>
  Erase key-value pair from current namespace
         <key>  key of the value to be erased

nvs_namespace  <namespace>
  Set current namespace
   <namespace>  namespace of the partition to be selected

nvs_list  <partition> [-n <namespace>] [-t <type>]
  List stored key-value pairs stored in NVS.Namespace and type can be specified
  to print only those key-value pairs.
Following command list variables stored
  inside 'nvs' partition, under namespace 'storage' with type uint32_tExample:
  nvs_list nvs -n storage -t u32 

   <partition>  partition name
  -n, --namespace=<namespace>  namespace name
  -t, --type=<type>  type can be: i8, u8, i16, u16 i32, u32 i64, u64, str, blob

nvs_erase_namespace  <namespace>
  Erases specified namespace
   <namespace>  namespace to be erased

help 
  Print the list of registered commands
```

It doesn't seem corrupt, but I haven't done a line by line check.

moefear85, 2022-07-14:
good idea. although that wouldn't catch any hidden variables that manifest themselves under stress, (which may be at play here, otherwise everybody would have faced the issue), it can rule out systematic problems, such as bad connectors/cables or buggy drivers.

egnor, 2022-07-14:
Don't be so sure it would have come up already! (Though somehow the bootloader works...?) I've reported and worked through more broken things in widely-used chips. People put up and work around a surprising amount of brokenness.

But yes, if the "basic console" demo works for you, that doesn't mean your bug is invalid! But maybe it gives us a point to start triangulating from to figure out the trigger conditions. Maybe we can come up with a minimal repro example, which will surely help get a fix faster.

moefear85, 2022-07-14:
@egnor 

it's premature to speak of silicon errata. The chance of you catching one here is small. Something like this would be reported by many users, this isn't something one can easily workaround, especially given all the tiny things people rush to the issues sections to complain about.

but obviously you're confused, since I'm not the one who reported the bug. read closely.

davidzuhn, 2022-07-14:
The small example I mentioned in the main ticket was indeed the basic console example from the IDF.   I have not been able to reproduce this in a small test case.   

Note that in the app I have, I am not only dropping lines from the 'help' command -- many operations will emit several lines of text (usually via ESP_LOGx), and that output is also irregular (subsequent examples will show differing results, sometimes even "correctly").   

The 'help' output is just the easiest and most obvious example.

@moefear85 : Just to be able to rule out certain systemic issues, I have seen this on all of my ESP32-S3 devices -- DevKitC, UM Feather-S3, Adafruit ESP32-S3 Feather.  I've used a variety of cables (as the DevKit C are USB-micro, and the other two are USB-C) using multiple upstream host ports (USB-C & USB-A) so I am fairly confident in ruling out a specific connector or cable failure.

It's also the case that any given line that does get output seems to always be correct.   The errors entail entire lines being dropped (some of which are just blank lines).   There are no garbled or missing characters within a line.

As far as drivers go, when I use my Pi4 as the development and USB host, I see the same problems as well.

egnor, 2022-07-14:
> The small example I mentioned in the main ticket was indeed the basic console example from the IDF.   I have not been able to reproduce this in a small test case.   

Ah, bummer. I know you posted a binary, is it possible to share source that reproduces the problem? Or is that too proprietary?

(Mostly I'm trying to rule out that something else in your app is messing with, say, interrupt handling in a way that's causing lost data.)

davidzuhn, 2022-07-14:
Using minicom and miniterm, I see similar results as when using 'idf.py monitor'.  

As for sharing the source to the app, I cannot distribute it in any public fashion.

pjsg, 2022-08-03:
I suspect that this is a problem in the monitor code. In particular, if the monitor reads "\r\n" from the (virtual) serial port, then it outputs "\r\n". However, if it reads "\r" and then "\n", it only outputs "\r". 

I *think* that the root cause of this is https://github.com/espressif/esp-idf/blob/release/v4.4/tools/idf_monitor_base/serial_handler.py#L85

However, getting rid of that line causes extra newlines in the output, but does fix the missing newlines.

I'm going to try and figure out what is going on.....

pjsg, 2022-08-04:
I think that this is the fix:

```
diff --git a/tools/idf_monitor_base/serial_handler.py b/tools/idf_monitor_base/serial_handler.py
index af267088f2..957ba9e13b 100644
--- a/tools/idf_monitor_base/serial_handler.py
+++ b/tools/idf_monitor_base/serial_handler.py
@@ -78,12 +78,8 @@ class SerialHandler:
             # add unprocessed part from previous "data" to the first line
             sp[0] = self._last_line_part + sp[0]
             self._last_line_part = b''
-        if sp[-1] != b'':
-            # last part is not a full line
-            self._last_line_part = sp.pop()
+        self._last_line_part = sp.pop()
         for line in sp:
-            if line == b'':
-                continue
             if self._serial_check_exit and line == console_parser.exit_key.encode('latin-1'):
                 raise SerialStopException()
             if gdb_helper:

```

davidzuhn, 2022-08-04:
While that may indeed be an issue, the problem that I see does not lie solely in the monitor.py program.   I see the same unusual results when using minicom as when I use monitor.py.   

I will also note that I have updated my development tree to ESP-IDF v5.0-dev-4547-gfde4afc67a and I see the same issues

When I have your patch applied (to that version of the esp-idf), I see different flawed output.  Using the same example, which has seen some changes since the initial report, but nothing in the area of stdio & console config, I now see some lines merged.   In the initial problem, lines (including blank lines) are sometimes dropped.   Now I see output like this:

```
throttle 
  throttle support
wifi 
  wifi supporttasks 
  Show tasks
```

Compared to the earlier "good text", the lines about the 'battery' command are no longer expected, but the 'wifi support' text is immediately followed by the 'tasks' command header.

pjsg, 2022-08-04:
Are you using the driver to write to the serial jtag or not? I.e. do you have

`esp_vfs_usb_serial_jtag_use_driver();`

or

`esp_vfs_usb_serial_jtag_use_nonblocking();`

It appears that you can overrun the output buffer if you are not using the driver and this can cause dropped characters.

davidzuhn, 2022-08-05:
My code calls ```esp_vfs_usb_serial_jtag_use_driver```.

pjsg, 2022-08-05:
I'm not sure that I can help then. My debugging approach was to run the `idf.py monitor` under linux `strace` to see what it was actually reading from the USB CDC device. That showed me that the data arriving on my linux box was correct. It sounds as though that isn't the case for you -- but I'd still verify it anyway.

Good luck tracking it down!

davidzuhn, 2022-08-05:
I hadn't thought about strace.   I ran minicom under strace (```strace -o minicom.log2 -s1024 minicom -D /dev/ttyACM0 ```) and then ran my help command.   I saw this:
```
BKT -1 #0>help
  data fs command
i2c 
  read/write/scan i2c values
adc 
  read ADC values

state 
  Display current connection state values

  throttle support

wifi 
  wifi support

tasks 
  Show tasks

event 
  Print event loop debug

reset 
  Reset the device (optionally to factory default)

version 
  Display version information

board 
  read board values

knob 
  View/set the knob position levels
help 
  Print the list of registered commands

battery 
  notify battery level
```

Here are some of the  corresponding lines from the strace output:
```
read(3, "fs command\r\r\ni2c \r\n  read/write/scan i2c values\r\r\nadc \r\n  read ADC values\r\n\r\nstate \r\n  Display c", 127) = 96
write(1, "fs command\r\ni2c \r\n  read/write/scan i2c values\r\nadc \r\n  read ADC values\r\n\nstate \r\n  Display c", 93) = 93
ioctl(3, TIOCMGET, [TIOCM_DTR|TIOCM_RTS|TIOCM_CTS]) = 0
ioctl(3, TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
pselect6(4, [0 3], NULL, NULL, {tv_sec=1, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=0, tv_nsec=999968242})
read(3, "urrent connection state values\r\n\r\nthrottle \r  throttle support\r\n\r\nwifi \r\n  wifi support\r\n\r\ntasks \r\n  Show tasks\r\n", 127) = 113
write(1, "urrent connection state values\r\n\nthrottle \r  throttle support\r\n\nwifi \r\n  wifi support\r\n\ntasks \r\n  Show tasks\r\n", 110) = 110
```

Notice the lack of a \n reading the line "throttle \r".   This is a line generated within the console help routine.  This is the missing header before the "  throttle support" text. 

I ran this cycle several times, and here's the output from three log files:
```
$ grep "event loop" minicom.log* | grep read
minicom.log:read(3, " tasks\r\n\r\nevent \r\n  Print event loop debug\r\n\r\nreset \r\n  Reset the device (optionally to factory default)\r\n\r\nversion \r  Display ", 127) = 127
minicom.log2:read(3, "\r\nevent \r\n  Print event loop debug\r\n\r\nreset \r\n  Reset the device (optionally to factory default)\r\n\r\nversion \r\n  Display version", 127) = 127
minicom.log3:read(3, "hrottle \r\n  throttle support\r\r\nwifi \r\n  wifi support\r\n\r\ntasks \r\n  Show tasks\r\n\r\nevent \r  Print event loop debug\r\r\nreset \r\n  Res", 127) = 127
```

Notice that 'event ' header lines are inconsistent.  Sometimes it has \r\n, and sometimes it's just the \r.   When the \n is missing, the text doesn't display properly  (pretty sure the first line is being output, but then is overwritten because of the \r without the \n).   

But the key thing is that the data coming in from the ESP32-S3 is inconsistent, even though the device has not been refreshed (or even restarted) between these runs.

pjsg, 2022-08-05:
The interesting thing about losing `\n` is that the output function handles `\n` specially. In particular in `usb_serial_jtag_write` it calls `usb_serial_jtag_ll_txfifo_flush();` immediately after adding the `\n` to the buffer. I'd try commenting out that line, and see if the problem changes. I made a whole bunch of changes to that function (I was also losing `\n` but that was due to the idf monitor) during my debugging. 

The inconsistency makes me think that it is related to the timing of the linux box polling the cdc device and precisely when characters get added to the output buffer. I imagine that the USB polling causes interrupts which happen at different times on each run. 

The other thing to try is that Wireshark supports tracing USB transactions via the `usbmon` kernel module. Tracing the USB CDC device might reveal whether the missing newline is at the start of a packet, in the middle or at the end.

davidzuhn, 2022-08-05:
When I disable the ``` usb_serial_jtag_ll_txfifo_flush``` call in ```usb_serial_jtag_write```, the entire console command line interface hangs, so I don't get my prompts, I don't get any further debug output, and I can't run any commands.   So disabling the call is very much a no-go situation.

I've done a trace under Wireshark, but I don't know how to read USB CDC traffic yet, so I haven't any further information on that front.  Capture file attached in case you already know what you're looking at.

[usb-cdc-monitor.zip](https://github.com/espressif/esp-idf/files/9264813/usb-cdc-monitor.zip)

This is a capture of connecting to the device, running several 'help' commands (during which time some of the lines were lost), and then disconnecting.   There are no other USB devices on this computer, so the capture contains only the ESP32-S3 traffic.

chipweinberger, 2023-01-02:
@davidzuhn 

> When I disable the  usb_serial_jtag_ll_txfifo_flush call in usb_serial_jtag_write, the entire console command line interface hangs

This PR *might* fix the hang. I'd need to look deeper to be sure.

https://github.com/espressif/esp-idf/pull/10208

chipweinberger, 2023-01-02:
I am also hitting the same issue when using python's serial console for reading.

Have we come to any conclusions?

chipweinberger, 2023-01-02:
And would also try increasing tx buffer size:

    usb_serial_jtag_driver_config_t config = {
        .tx_buffer_size = 4096,
        .rx_buffer_size = 256
    };

chipweinberger, 2023-01-02:
I'd also try replacing:

`esp_vfs_usb_serial_jtag_use_driver();`

with 

`esp_vfs_usb_serial_jtag_use_nonblocking();`

chipweinberger, 2023-01-02:
I'd also try getting rid of all calls to 

`usb_serial_jtag_ll_txfifo_flush()`

robert-alfaro, 2023-01-14:
Not sure if this is related to the "unreliability" ..  An issue we're facing with esp32s3 module and serial connection through usb serial/jtag peripheral (usb-c port to usb pins) and have noticed that when using idf.py monitor or minicom, at any point while  serial logs are printing on the host, if a keyboard press occurs (a letter, an arrow..whatever key) then all the logs cease and the only way to recover the host client is to unplug the usb-c connection with the hardware. Firmware continues to run fine. We are not using any uart input to firmware, only "stock" logging output.

chipweinberger, 2023-02-17:
Has any progress been made on this? 

Otherwise I'll be debugging it next week... begrudgingly.

egnor, 2023-02-20:
You have my moral support @chipweinberger ! Which isn't worth much but hey.

davidzuhn, 2023-02-21:
I have done no further work on this -- looking at the raw data (via strace) I'm pretty sure that all of the data was coming across the line *except* some of the newlines.

davidzuhn, 2023-02-21:
Chip, I'll buy you a beer[*] if you manage to find this one.

*: Or other tasty beverage of your choice.

egnor, 2023-02-21:
Make that two!

pjsg, 2023-02-22:
Are you using the serial monitor from the IDF? That does have a bug with
regard to newline handling (as I recall). This was the bug fix for me:

--- a/tools/idf_monitor_base/serial_handler.py
+++ b/tools/idf_monitor_base/serial_handler.py
@@ -78,12 +78,8 @@ class SerialHandler:
             # add unprocessed part from previous "data" to the first line
             sp[0] = self._last_line_part + sp[0]
             self._last_line_part = b''
-        if sp[-1] != b'':
-            # last part is not a full line
-            self._last_line_part = sp.pop()
+        self._last_line_part = sp.pop()
         for line in sp:
-            if line == b'':
-                continue
             if self._serial_check_exit and line ==
console_parser.exit_key.encode('latin-1'):
                 raise SerialStopException()
             if gdb_helper:

davidzuhn, 2023-02-22:
I have these issues when using the idf monitor, miniterm or minicom.

chipweinberger, 2023-02-24:
I can repro it!

I'm using this code to print sequential numbers, in various size chunks.

```
void app_main()
{
    // use driver
    pd_console_init_usb(error);

    printf("\nabcdefghijk");

    char* s = (char*) calloc(1024 * 1024, 1);

    while (true) {
        uint32_t m = 0;
        for(int i = 0; i < 1024 * 4; i++) {
            m += sprintf(s+m, "%u\n", i);
        }
        printf("%s", s);
        vTaskDelay(10);
    }
}
```

and then python code to check that the numbers are indeed sequential:

```
    jflashPort.determinePort(st)
    jflashReboot.reboot(st)
    jflashConsole.connect(st)

    i = 0
    while True:
        c = st.console.readline().decode('ascii', 'ignore')
        print("	  > " + c, end="")
        val = int(c.replace("\r\n",""))
        if val != i:
            print("fail " + c)
            break
        i = (i+1) % 4096

    return
```

And I hit this:

```
          > 1431
          > 1432
          > 1433
          > 1434
          > 1435
          > 1436
          > 3373
fail 3373
```

For some reason it jumped a huge amount of numbers!

chipweinberger, 2023-02-24:
Again. hit:

```
          > 486
          > 487
          > 488
          > 489
          > 2048
fail 2048
```

This is my sdkconfig:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10822109/sdkconfig.txt)

chipweinberger, 2023-02-24:
I can hit it event without using the driver:

```
          > 3658
          > 3659
          > 3660
          > 3661700
fail 3661700
```

chipweinberger, 2023-02-24:
I changed the code slightly. 1st number is the number of iterations. 2nd number is [0,10240] repeating.

**This is without the driver!**

```
          > 48,3114
          > 48,3115
          > 48,3116
          > 48,3117
          > 48,3118
          > 48,3368
fail 48,3368
```

```
        > 33,2230
          > 33,2231
          > 33,2232
          > 33,2233
          > 33,2234
          > 33,233,2335
fail 33,233,2335
```

**The different numbers of ways it can fail is pretty staggering.**

- It sometimes skips bytes
- It sometimes repeats bytes
- each failed line always has some "correct" bytes. But the wrong bytes seem like it can be anything.

chipweinberger, 2023-02-24:
Okay, things are getting strange. As a control, I plugged my computer into the UART port (no code changes), and I fail even faster.

```
          > 0,2375
          > 0,2376
          > 0,252
fail 0,252
```

```
          > 0,2375
          > 0,2376
          > 0,20,6556
fail 0,20,6556
```

```
          > 0,2375
          > 0,2376
          > 0,2238
fail 0,2238
```

```
          > 0,2374
          > 0,2375
          > 0,2376
          > 0,2
fail 0,2
```

chipweinberger, 2023-02-24:
I'm beginning to think this problem is out of our control, and even harder to fix than suspected. 

@igrr , please assist here. This is a big issue.

chipweinberger, 2023-02-24:
Okay, latest theory.

The problems only repro with `printf` & `putc`.  It does not repro with `uart_tx_char` directly. 

`printf` & `putc` are both _buffered_. They write to an internal buffer, and the flush when the buffer it full, or a newline is hit. 

My guess is that `_fflush_r` is buggy. It's not waiting long enough to actually flush the data, and therefor we are skipping bytes. 

fflush is implemented in ROM only. `PROVIDE( _fflush_r = 0x4000150c );` I don't have the source code for it. I think it is possible to overwrite this though, and not use the rom implementation. We are starting to get a bit out of my expertise.

chipweinberger, 2023-02-24:
I am at an impass. I am trying to reimplement `fflush` and `_fflush_r`, but my application freezes shortly after boot.

CMakeLists.txt
```
idf_component_register(SRCS  "../main.c")

target_link_libraries(${COMPONENT_LIB} INTERFACE "-Wl,--wrap=fflush")
target_link_libraries(${COMPONENT_LIB} INTERFACE "-Wl,--wrap=_fflush_r")
```

main.c 
```
IRAM_ATTR int __wrap_fflush(FILE* p)
{
    return fflush(p);
}

IRAM_ATTR int __wrap__fflush_r(struct _reent *ptr, FILE *stream) 
{
   return _fflush_r(ptr, stream);
}

// called on PRO-cpu
void app_main()
{
    printf("\nabcdefghijk");
    char* s = (char*) heap_caps_calloc(1, 1024 * 250, MALLOC_CAP_INTERNAL);
    int v = 0;
    while (true) {
        uint32_t m = 0;
        for(int i = 0; i < 1024 * 10; i++) {m += sprintf(s+m, "%u,%u\n", v, i);}
        for(int i = 0; i < m; i++) { putc(s[i], stdout);}
        vTaskDelay(1);
        v++;
    }
}

```

Logs: (freezes at last line)
```
I (2397) cpu_start: Pro cpu start user code
I (2398) cpu_start: cpu freq: 240000000
I (2398) cpu_start: Application information:
I (2398) cpu_start: Project name:     jJamApp
I (2400) cpu_start: App version:      v0.5
I (2401) cpu_start: Compile time:     Feb 24 2023 14:16:18
I (2402) cpu_start: ELF file SHA256:  edaed765041c71cc...
I (2404) cpu_start: ESP-IDF:          v4.4.3-515-g17e807a8ae
I (2405) heap_init: Initializing. RAM available for dynamic allocation:
I (2408) heap_init: At 3FC9A680 len 0004F090 (316 KiB): D/IRAM
I (2409) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (2411) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (2412) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (2414) spiram: Adding pool of 2048K of external SPI memory to heap allocator
```

pjsg, 2023-02-25:
I'm working on the lua nodemcu-firmware and I realize that my output setup
also has a

      fcntl(fileno(stdout), F_SETFL, 0);

Philip

pjsg, 2023-02-25:
I'm working on the lua nodemcu-firmware and I realize that my output setup
also has a

      `fcntl(fileno(stdout), F_SETFL, 0);`

The complete code is

```
      /* Minicom, screen, idf_monitor, esplorer send CR when ENTER key is pressed */
      esp_vfs_dev_usb_serial_jtag_set_rx_line_endings(ESP_LINE_ENDINGS_CR);
      /* Move the caret to the beginning of the next line on '\n' */
      esp_vfs_dev_usb_serial_jtag_set_tx_line_endings(ESP_LINE_ENDINGS_CRLF);

      /* Enable blocking mode on stdin and stdout */
      fcntl(fileno(stdout), F_SETFL, 0);
      fcntl(fileno(stdin), F_SETFL, 0);

      usb_serial_jtag_driver_config_t usb_serial_jtag_config;
      usb_serial_jtag_config.rx_buffer_size = 1024;
      usb_serial_jtag_config.tx_buffer_size = 1024;

      esp_err_t ret = ESP_OK;
      /* Install USB-SERIAL-JTAG driver for interrupt-driven reads and writes */
      ret = usb_serial_jtag_driver_install(&usb_serial_jtag_config);
      if (ret != ESP_OK)
      {
        return -1;
      }

      /* Tell vfs to use usb-serial-jtag driver */
      esp_vfs_usb_serial_jtag_use_driver();

```

chipweinberger, 2023-03-03:
@igrr @ESP-Marius @SoucheSouche 

Just trying to get any eyes on this. 

Newlib C fflush is not working correctly, causing printf to skip bytes. it's easy to repro.

chipweinberger, 2023-05-06:
I've opened up this PR: https://github.com/espressif/esp-idf/pull/11344/files

ermacv, 2023-07-20:
Hello! I just want to confirm that I also observe the same issue related to virtual UART discards or "eats" some characters while printing. I applied the patch @chipweinberger provided but there were still some "eaten" chars. And I noticed that the TX Ring buffer has only 256 bytes hardcoded into `USB_SERIAL_JTAG_DRIVER_CONFIG_DEFAULT` macro for `tx_buffer_size`. And I increased this value to 2048 bytes and everything became OK, no more skipped chars.

In my case I use custom ESP32-S3 based board and I use ESP_LOG and CLI printing into the same serial and CLI is used for automated tests where I found this issue first, some tests failed. So in my case buffer is filled quite fast and some chars are discarded and never added to ring buffer (so they are not even passed to HW).

jjsch-dev, 2023-08-22:
I have a smiliar problem, and upon examining the code, I noticed that the [usbjtag_tx_char_via_driver](https://github.com/espressif/esp-idf/blob/32472536715d674c160a2565795895e0273f8cde/components/vfs/vfs_usb_serial_jtag.c#L406) function contains a comparison that checks whether an immediate send operation succeeds. However, the return value from [usb_serial_jtag_write_bytes](https://github.com/espressif/esp-idf/blob/32472536715d674c160a2565795895e0273f8cde/components/driver/usb_serial_jtag/usb_serial_jtag.c#L158) is always 1, which prevents the intended logic to try blocking for 50 mS, but also a return that always skyp the bloking try.

**Proposed Solution:**
To address this issue, I suggest considering the following adjustments:

Modify the **usb_serial_jtag_write_bytes** function to return -1 on failure and 0 on success.
```c 
int usb_serial_jtag_write_bytes(const void* src, size_t size, TickType_t ticks_to_wait)
{
    ESP_RETURN_ON_FALSE(size != 0, ESP_ERR_INVALID_ARG, USB_SERIAL_JTAG_TAG, "size should be larger than 0");
    ESP_RETURN_ON_FALSE(src != NULL, ESP_ERR_INVALID_ARG, USB_SERIAL_JTAG_TAG, "Invalid buffer pointer.");
    ESP_RETURN_ON_FALSE(p_usb_serial_jtag_obj != NULL, ESP_ERR_INVALID_ARG, USB_SERIAL_JTAG_TAG, "The driver hasn't been initialized");

    const uint8_t *buff = (const uint8_t *)src;
    // Blocking method, Sending data to ringbuffer, and handle the data in ISR.
    BaseType_t result = xRingbufferSend(p_usb_serial_jtag_obj->tx_ring_buf, (void*) (buff), size, ticks_to_wait);

    // Now trigger the ISR to read data from the ring buffer.
    usb_serial_jtag_ll_ena_intr_mask(USB_SERIAL_JTAG_INTR_SERIAL_IN_EMPTY);
       
    // Return -1 on failure, 0 on success.
    return (result == pdFALSE) ? -1 : 0;
}
```
Update the **usbjtag_tx_char_via_driver** function to accommodate the adjusted return values and logic.
```c 
static void usbjtag_tx_char_via_driver(int fd, int c)
{
    char ch = (char) c;
    TickType_t ticks = (TX_FLUSH_TIMEOUT_US / 1000) / portTICK_PERIOD_MS;
    // Attempt to send the character immediately without blocking.
    if (usb_serial_jtag_write_bytes(&ch, 1, 0) != 0) {
        s_ctx.tx_tried_blocking = false;
    }

    // If immediate send fails, try blocking with a timeout.
    if (s_ctx.tx_tried_blocking == false) {
        if (usb_serial_jtag_write_bytes(&ch, 1, ticks) != 0) {
            return;
        } else {
            s_ctx.tx_tried_blocking = true;
        }
    } 
}
```

chipweinberger, 2023-08-22:
@jjsch-dev thats is a great observation!

Can you create a separate issue and open a Pr?

jjsch-dev, 2023-08-22:
@chipweinberger thanks for the comment, and I will do a new issue and open the PR.

closed 2023-09-04 by espressif-bot

davidzuhn, 2023-09-05:
I find that this problem is not fixed, and in fact is worse now with the latest version of the IDF (git commit 62ee4135e033cc85eb0d7572e5b5d147bcb4349e).

This is what I get when I run my help command (which should look like the content shown earlier in this bug report).

```
BKT-1 #0>BKT-1 #0>I (4918) wc.event: found ssid:Clearwater rssi:-16 known:noI (4918) wc.event: found ssid:Palouse rssi:-44 known:no
I (4918) wc.event: found ssid:Clearwater rssi:-61 known:no
I (4918) wc.event: found ssid:myownlittleidaho rssi:-74 known:no
I (4928) wc.event: found ssid:BarbraHaugen rssi:-91 known:noown:no
I (4928) wc.event: found ssid:CenturyLink1349 rssi:-96 known:no
BKT-1 #0>helpfs 
  data fs command
pilot 
  set pilot light on/off times
i2c 
  read/write/scan i2c valuadc 
  read ADC valustate 
  Display current connection state values
throttle 
  throttle support

wi  wifi support
heap 
  show heaptasks 
  Show tasks
me  Memory debug

event 
  Print event loop debug
reset 
  Reset the device (optionally to factory defaulversion 
  Display version information
board 
  read board valuknob 
  View/set the knob position levels
help  [<string  Print the summary of all registered commands if no arguments are given,
  otherwise print summary of given command.
      <string>  Name of commabattery 
  notify battery level
```

Instead of losing lines, I lose large chunks of the text in arbitrary places.

jjsch-dev, 2023-09-05:
Hi @davidzuhn, I think the master (as of 4th Sep) doesn't have this fix. Does your code use printf in more than one thread?

davidzuhn, 2023-09-05:
The tree I'm using has the fix that was claimed to close this issue (commit c92615308c96a2986678fdb0a9a33fd62637fffd of Aug 23).

My code uses a log of ESP_LOG across many threads, and a little bit of printf (which is almost all from the thread that handles the command line processing).

jjsch-dev, 2023-09-05:
Sorry with the fix date time, I made a mistake with the version. Although ESP_LOG is thread safe, printf is not, perhaps you can create a function that uses a mutex to serialize printf and if possible disable ESP_LOG.

chipweinberger, 2023-09-05:
@jjsch-dev perhaps we could add a warning? something like:

"printf is not thread safe. please only call printf from a single thread"

which would be logged if you call printf from multiple threads?

jjsch-dev, 2023-09-06:
@chipweinberger, You have right because while **printf()** is reentrant, meaning it can safely be called from multiple threads without corrupting its internal data structures, it doesn't inherently guarantee that output will be serialized correctly. This can result in interleaved or mixed output when multiple threads use **printf()** simultaneously.

I might be mistaken, but in these scenarios, I believe the issue of mixed output occurs when using a console to receive user commands. For instance, when utilizing **linenoise()**, which relies on **write()** and **read()** functions. If the formatted response is generated using **printf()** from a different thread than the one handling **linenoise()**, this problem can arise. It becomes even more challenging if you have **ESP_LOG** macros enabled. To resolve this issue, you would need to make changes to the **IDF** to establish a **mutex** that can be shared for **write()**, **printf(),** and **ESP_LOG**.

The straightforward solution is to disable the **ESP_LOG** macros and ensure that **printf** is called from the same thread that runs **linenoise**.

davidzuhn, 2023-09-06:
In my scenario, the printf output is not being interleaved between data from multiple threads, nor is there any interleave of output from ESP_LOG.   The block of text with the missing/corrupted data is generated and printed by the help command that comes from esp_console_register_help_command().

I completely understand that there is the chance of interleave between portions of the help text and log messages from ESP_LOG.   But that's not happening.   What's happening is that the blob of text from help is missing characters when being read from the host.  Earlier it was missing the carriage return, allowing the terminal to overwrite one line with the next giving the appearance of missing lines of text.   With the latest ESP-IDF, much more than that is going missing.

jjsch-dev, 2023-09-06:
@davidzuhn thanks for the firmware explanation, is the host program in python?

davidzuhn, 2023-09-06:
I get similar results when I use 'idf.py monitor' or when I use minicom.

jjsch-dev, 2023-09-06:
ok, can you share the source code? I have some time and maybe I can help you find the problem.

[5 more replies]