espressif/esp-idf #9319: docs: espcoredump.py now esp-coredump? (v5.0)

someburner, 2022-07-09:
With `release/v5.0` branch out I'm now developing against that. I saw it mentioned in the forum that API would likely not be changing much, but looks like there's still things trickling in.

Do not see anything about these methods changing in the docs yet:

- `espcoredump.py` - still in PATH but no longer executable, seems like you need to now execute `esp-coredump`?
- `spi_flash_erase_range` - removed, now must use `esp_flash_erase_region`?

If more comprehensive doc edits are on the way feel free to close, but just thought I'd document these.

igrr, 2022-07-09:
Hi @someburner,

Thanks for reporting the issue about espcoredump missing the executable flag! This was also noticed internally and the fix is on the way. We will also update the docs.

As for the spi_flash API removal, this is mentioned in the migration guide: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/peripherals.html#breaking-changes-in-legacy-apis. If you have noticed any other place in the docs which still references the old functions, please let us know and we'll fix that!

someburner, 2022-07-11:
I was going to mention that the json file in the IDF needed to be writeable which causes problems on linux if you have read-only access to the toolchain. But looks like you made a new issue https://github.com/espressif/esp-idf/issues/9329

closed 2022-07-19 by espressif-bot