espressif/esp-idf #9320: rst:0x1 (POWERON_RESET),boot:0x2 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))<CR><LF> waiting for download

harshalbhav, 2022-07-09:
## Environment

    - Development Kit: ESP32-Wrover-Kit and Other
    - Kit version (for WroverKit/PicoKit/DevKitC):
    - Module or chip used: ESP32-WROOM-32D
    - IDF version (run git describe --tags to find it):
    // v4.4
    - Build System: idf.py
    - Compiler version (run xtensa-esp32-elf-gcc --version to find it):
    // 1.22.0-80-g6c4433a
    - Operating System: Windows
    - (Windows only) environment type: Plain Command Prompt
    - Using an IDE?: No
    - Power Supply: USB and external 3.3

## Problem Description

   We are using ESP32E-WROOM-32UE controller for our application. We had enabled Secure Boot option with re-programmable option. We follow the following steps to do this.
   1. Burn the secure key using espefuse.py.
   2. Flash secure-bootloader at 0x0 location as mentioned in esp document
   3. Flash the partition binary and application binary accordingly.
   4. Start the device after flashing our application.

Some devices is working fine but for some of the devices pushing a logs "rst:0x1 (POWERON_RESET),boot:0x2 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))<CR><LF> waiting for download". We restart the device multiple times but still showing same logs. 

Please guide me to solve this issue.

### Expected Behavior

Device should be boot properly.

### Actual Behavior

Device showing the error log "rst:0x1 (POWERON_RESET),boot:0x2 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))<CR><LF> waiting for download"

09-Jul-22 18:11:49.752 [RX] - ets Jul 29 2019 12:21:46<CR><LF>
<CR><LF>
rst:0x1 (POWERON_RESET),boot:0x2 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))<CR><LF>
waiting for download<CR><LF>

negativekelvin, 2022-07-09:
https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html

dobairoland, 2022-07-11:
Hi @harshalbhav. Just to explain the above link, this is probably a hardware issue. The chip is entering bootloader mode on those board where you experience this issue. GPI0 is at GND so every reset will put the chip into bootloader mode.

closed 2022-07-11 by dobairoland