espressif/esp-idf #9322: FAT FS on External Flash example does not detect MXIC MX25L3206E and MX25L6406E

armwdev, 2022-07-09:
## Environment

- Development Kit:      ESP32-S2-DevKitM-1
- Kit version: v1
- Module or chip used:  [ESP32-S2FH4]
- IDF version ESP-IDF v4.4.1
- Build System:         [CMake]
- Compiler versionxtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

FAT FS on External Flash example detects the Winbond W25Q64FVIQ but does not detect the MXIC MX25L3206E serial flash chip. In the log it says:
```
I (5737) example: Initializing external SPI Flash
I (5737) example: Pin assignments:
I (5737) example: MOSI: 11   MISO: 13   SCLK: 12   CS: 10
I (5744) spi_flash_hal_common: spi_flash_hal_common_command!!!! host: 3ffe32c0 trans: 9f
I (5752) gspi_flash_ll.h: gpspi_flash_ll_get_buffer_data: ffffffff
I (5759) spi_flash_hal_common: trans->miso_data: ff
I (5765) memspi: raw_chip_id: FFFFFF
E (5769) memspi: no response!!!
I (5773) spi_flash: !!!esp_flash_init chip_id: 0, err: 108, 
E (5779) example: Failed to initialize external Flash: ESP_ERR_INVALID_RESPONSE (0x108)
```

The chip ID (RDID) should not be FFFFFF so not sure why it can't read the chip ID. Again, for Winbond chip, there is no problem.

Problem could be around "static esp_err_t IRAM_ATTR read_id_core(esp_flash_t* chip, uint32_t* out_id, bool sanity_check)"
but still checking.

### Expected Behavior

Since the MXIC is also in the supported chip list, it should be detected normally as the Winbond chip. Those chips use universal commands so it's not a vendor specific issue. I have also tried the MX25L6406E but again the same problem occurs. In the past, I could read and write flash chips without a problem but after IDF 3.0, ESP seems to have gone a bit backwards in many aspects and complexity. In the past there were legacy functions to read and write raw flash chips but all seems gone.

### Code to reproduce this issue

```cpp
static esp_flash_t* example_init_ext_flash(void)
{
    const spi_bus_config_t bus_config = {
        .mosi_io_num = FSPI_IOMUX_PIN_NUM_MOSI,
        .miso_io_num = FSPI_IOMUX_PIN_NUM_MISO,
        .sclk_io_num = FSPI_IOMUX_PIN_NUM_CLK,
        .quadhd_io_num = FSPI_IOMUX_PIN_NUM_HD,
        .quadwp_io_num = FSPI_IOMUX_PIN_NUM_WP,
    };

    const esp_flash_spi_device_config_t device_config = {
        .host_id = SPI3_HOST,
        .cs_id = 0,
        .cs_io_num = FSPI_IOMUX_PIN_NUM_CS,
        .io_mode = SPI_FLASH_SLOWRD,
        .speed = ESP_FLASH_20MHZ,
    };

    ESP_LOGI(TAG, "Initializing external SPI Flash");
    ESP_LOGI(TAG, "Pin assignments:");
    ESP_LOGI(TAG, "MOSI: %2d   MISO: %2d   SCLK: %2d   CS: %2d",
        bus_config.mosi_io_num, bus_config.miso_io_num,
        bus_config.sclk_io_num, device_config.cs_io_num
    );

    // Initialize the SPI bus
    ESP_ERROR_CHECK(spi_bus_initialize(SPI3_HOST, &bus_config, 2));

    // Add device to the SPI bus
    esp_flash_t* ext_flash;
    ESP_ERROR_CHECK(spi_bus_add_flash_device(&ext_flash, &device_config));

    // Probe the Flash chip and initialize it
    esp_err_t err = esp_flash_init(ext_flash);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize external Flash: %s (0x%x)", esp_err_to_name(err), err);
        return NULL;
    }

    // Print out the ID and size
    uint32_t id;
    ESP_ERROR_CHECK(esp_flash_read_id(ext_flash, &id));
    ESP_LOGI(TAG, "Initialized external Flash, size=%d KB, ID=0x%x", ext_flash->size / 1024, id);

    return ext_flash;
}
```

mythbuster5, 2022-07-12:
Not all MXIC chips are supported in IDF. And seems MX25L3206E and MX25L6406E are not recommended by flash vendor officially.

[image]

This is a little bit weird for these XMIC chip when I debugging, and I suppose the issue was about the `#hold` pin . What I'm going to suggest is that using `fastrd` mode and pull-up `#hold` pin directly (What I did is directly connect #hold to VCC directly on my chip).

This is a kind of workaround, perhaps will solve your issue ASAP. For the deep reason, it might take some time to dig out.

If this workaround can solve your issue, I think we can close this one.

armwdev, 2022-07-12:
That was it! Thanks a lot for this valuable info. I originally tried using gpio_set_level() and gpio_set_pull_mode() but haven't thought about physical pull up. Closing this for now.

closed 2022-07-12 by armwdev