espressif/esp-idf #9323: Memory leak in esp_bt_hidh_dev_report_write

selimrecep, 2022-07-10:
## Environment
    Development Kit: ESP32-DevKitC
    Kit version v4
    Module or chip used: ESP32-WROOM-32
    IDF version 4.4.1 (ESP_IDF_VERSION)
    Build System: idf.py (on platform.io)
    Compiler version: toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
    Operating System: Windows
    (Windows only) environment type: PowerShell
    Using an IDE?: Platform IO
    Power Supply: USB

## Problem Description

When I call `esp_hidh_dev_output_set` I detect heap memory leak, I think it is caused by here
https://github.com/espressif/esp-idf/blob/220590d599e134d7a5e7f1e683cc4550349ffbf8/components/esp_hid/src/bt_hidh.c#L796
when report_id is not NULL (and for most of the time it is not NULL) it allocates new memory block for the report to set the report id in the beginning of the report, then, it gets `osi_malloc` ed in `btc_transfer_context` which is (I believe) deallocated. But not the one allocated via malloc in `esp_bt_hidh_dev_report_write`

### Expected Behavior
The function shouldn't leak memory
### Actual Behavior
Function leaks memory
### Steps to reproduce

1. Call esp_hidh_dev_output_set with a non-zero report_id
2. It leaks memory

// If possible, attach a picture of your setup/wiring here.
Bare USB Wiring

### Code to reproduce this issue

```cpp
// I can't put the whole code here, but as an example
StructType aReport;

printf("\n-- %d ", esp_get_free_heap_size());
// The struct itself includes report id in the beginning so this sends address of the next uint8_t
esp_hidh_dev_output_set(dev, 0, 0x31, (uint8_t*)(&aReport.Data[1]), sizeof(aReport) - 1);
printf("\n-- %d ", esp_get_free_heap_size());
```

## Debug Logs

```
-- 128708 -- 128468 
-- 128620 -- 128380 
-- 128532 -- 128292 
-- 128444 -- 128204 
-- 128356 -- 128116 
-- 128268 -- 128028 
-- 128180 -- 127940 
```

## Other items if possible

- [X] [sdkconfig_renamed.txt](https://github.com/espressif/esp-idf/files/9078812/sdkconfig_renamed.txt)
- [ ] elf file in the ``build`` folder
- [ ] coredump

selimrecep, 2022-07-10:
Adding 
```cpp
if(report_id)
   free(p_data);
```

right after https://github.com/espressif/esp-idf/blob/220590d599e134d7a5e7f1e683cc4550349ffbf8/components/esp_hid/src/bt_hidh.c#L806
seems to fix it, but there could be side effects that ... don't know so not completely sure

boblane1, 2022-07-11:
@selimrecep 

Your fix is right, thanks!

closed 2022-07-19 by espressif-bot