espressif/esp-idf #9324: esp32s3 gdb does not work on Mac OS M1 (arm64) architecture

davidzuhn, 2022-07-11:
## Environment

- Development Kit:   ESP32-S3-DevKitC-1
- Kit version (for WroverKit/PicoKit/DevKitC): v1.0
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-4001-g495d35949d
- Build System:    CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32s3-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0

- Operating System:    Mac OS 12 (on Apple Silicon M1 - arm64)
- Using an IDE?:  No
- Power Supply:    USB

## Problem Description

Cross-development tools for Apple do not work properly in a multi-architecture environment.   The tools are built for x86_64, while the CPU is arm64.   Rosetta2 manages to run the compiler properly, and I can build for the target.  But running GDB in order to do JTAG debugging fails:

```
$ xtensa-esp32s3-elf-gdb -x gdbinit build/hello_world.elf 
dyld[62220]: Library not loaded: /opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/Python
  Referenced from: /Users/zoo/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220529/xtensa-esp-elf-gdb/bin/xtensa-esp-elf-gdb-3.9.HH6vTPr
  Reason: tried: '/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/Python' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file), '/System/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file), '/opt/homebrew/Cellar/python@3.9/3.9.13_1/Frameworks/Python.framework/Versions/3.9/Python' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file), '/System/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file)
/Users/zoo/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220529/xtensa-esp-elf-gdb/bin/xtensa-esp32s3-elf-gdb: line 45: 62220 Abort trap: 6           "${GDB_WRAPPER_TEMP_BINARY}" --mcpu=esp32s3 "$@"

```

### Expected Behavior

The debugger should be running.   

### Actual Behavior

See the error results above.

### Steps to reproduce

1. create gdbinit as describe in ESP-IDF Programming Guide
2. cat gdbinit 
```target remote :3333
set remote hardware-watchpoint-limit 2
mon reset halt
flushregs
thb app_main
c
```
3. (In one shell), run the following:
```
openocd -f board/esp32s3-builtin.cfg
```
4. (In a different shell, also with the esp-idf active via export.sh). Run the gdb command as above. 

The DevKit is connected only using a single cable from the Mac (USB-C) to the Micro-USB plug labelled 'USB'

### Code to reproduce this issue

Use any program (the hello_world example will do just fine).  This problem is not specific to the program we're attempting to run on the ESP32-S3.

## Debug Logs

## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
- [X] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

[hello-world.zip](https://github.com/espressif/esp-idf/files/9080573/hello-world.zip)

antmak, 2022-07-11:
@davidzuhn Hi! Thanks for reporting.

We have several wrappers to run GDB with different python versions, but unfortunately it does not work correctly on macos m1. (Python is arm64, but GDB is amd64).

Could you try the next command for running gdb without python support, please?

`xtensa-esp-elf-gdb-no-python --mcpu "esp32s3" -x gdbinit build/hello_world.elf`

davidzuhn, 2022-07-13:
The gdb-no-python command does work.

antmak, 2022-07-14:
Thank you!  We're updating the wrappers to have a fallback. Also we will release native builds for aarch64-darwin soon

Alvin1Zhang, 2022-07-25:
Thanks for reporting, fix is available https://github.com/espressif/esp-idf/commit/c0e94c6688b6769f8279e0c52eac58825d385611, feel free to reopen.

closed 2022-07-25 by Alvin1Zhang

davidzuhn, 2022-07-25:
Confirmed that gdb works as expected on the same M1 Mac as the initial report.   

Thank you!