espressif/esp-idf #9326: Create virtual network interface like in LINUX, 1 static, 1 DHCP

franz-ms-muc, 2022-07-11:
**Is your feature request related to a problem? Please describe.**

So, when you use a ESP32 device e.g. with Ethernet. 
it would be nice to have a Interface eth0 which is DHCP Enabled, 
so the DHCP Server gives the ESP32 a Address and Gateway. 
this is most easy on most Networks to get Internet Acces. 

then it would be nice to have a Second, Alias Interface eth0:0 with a STATIC IP Address. 
so when the Service Techican comes and want to connect to the Device he does not need to find out the DHCP Assigned IP, but he can use the fixed one. 

**Describe the solution you'd like**

make Alias Interfaces working, prefferrably for Wifi and Ethernet, 
but at least for Ehternet.

**Describe alternatives you've considered**

not yet done a lot research on it. 
a Google on "virtual network interfaces in ESP32" did not bring up any solution. 

**Additional context**

none.

NathanJPhillips, 2022-07-14:
This is probably a request that should be directed towards the LWIP repository. The ESP port of LWIP shouldn't be diverging from the main branch in ways this significant.

franz-ms-muc, 2022-07-14:
Do you know their issue Tracker?

Holen Sie sich Outlook für Android<https://aka.ms/AAb9ysg>

[https://www.meisterschulen-mchn.de/images/meisterschulen/allgemein/4-logos/logofc.jpg]

Meisterschulen am Ostbahnhof
Mühldorfstr. 6
81671 München

Tel.: 089 416002 - 0
Fax: 089 416002 - 111

Internet: www.meisterschulen-münchen.de<http://www.meisterschulen-mchn.de>

Alvin1Zhang, 2022-11-21:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/88525d1ec6cc69e22f61f32868910ff7bb817d08, feel free to reopen.

closed 2022-11-21 by Alvin1Zhang