espressif/esp-idf #9327: Version 5.0 doesn't work system-wide

stkw0, 2022-07-11:
## Environment

- Development Kit:      none
- Module or chip used: ESP32-WROOM-32E
- IDF version: release/5.0
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:  Linux
- Using an IDE?: Kdevelop
- Power Supply:  USB

## Problem Description

I have a package that installs esp-idf system-wide because it have numerous advantages (doesn't need a manual set-up, better integration with IDEs and other tools, easier to upgrade idf version, etc). With version 4.4.1 it worked correctly but if I try to use release/5.0 this approach doesn't work anymore. When running the same command, `idf build` It only shows the next error message and doesn't build the image.

```
ERROR: /home/droman/.espressif/python_env/idf5.0_py3.9_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

fatal: not a git repository: '/usr/share/esp-idf/.git'
WARNING: Git version unavailable, reading from source
ESP-IDF v5.0.0
``` 
I am aware that version 5.0 isn't released yet but we use a ksz8863 Ethernet chip and the official idf driver on [eth-drivers](https://github.com/espressif/esp-eth-drivers) requires version 5.0

### Expected Behavior

IDF is capable of building the image

### Actual Behavior

An error is shown

dobairoland, 2022-07-11:
Have you run the suggested command? I assume that you've had and used your environment from `idf4.4_py3.9_env` until now and after you've updated ESP-IDF, it tries to find and use `idf5.0_py3.9_env` but it cannot find because the installer hasn't been run yet for the new ESP-IDF version.

stkw0, 2022-07-11:
No, I never run the installer for version 4.4 (and it wasn't a problem for 4.x versions). Starting with 5.0 this breaks as can be seen in the error message, so I guess this is a new check that wasn't in place in previous versions

dobairoland, 2022-07-11:
All v4.x and v5.0 versions require to run `install.sh` at least once and consequently `. export.sh` in all new shells in order to set up the tools.

The above message is new for v5.0, but the user-side setup requirements have not changed.

Please see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html for more information.

dobairoland, 2022-07-11:
I think this could be useful for you as well: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html

closed 2022-07-11 by dobairoland

stkw0, 2022-07-11:
> All v4.x and v5.0 versions require to run `install.sh` at least once and consequently `. export.sh` in all new shells in order to set up the tools.
>
> The above message is new for v5.0, but the [...]

I don't think that's true.for v4.x. For example I have installed idf and never executed install.sh nor sourcing export.sh in any shells. The package that provides idf for v4.x creates a file in /etc/env.d/99esp-idf which exports `IDF_PATH=/usr/share/esp-idf` where idf is installed system-wide. Then the idf.py knows where to look for IDF without caring about exporting anything which is very useful when using an IDE. The IDE is capable of running idf builds without having to ". export.sh" (which, in some IDEs is impossible to do due the inability to configure the shell of the IDE). 

With IDF 5.0 this behavior breaks which means that IDF no longer works