espressif/esp-idf #9330: ESP32 S2/S3 Host USB HID, feature report read/write

AristarchosOfSamos, 2022-07-11:
Hi,

Seen the nice work that is currently underway for USB host drivers as CDC or MSC in new S3/S2 ESP32 variants.
A very nice though addition, that could open up more potential is the presence of a host USB HID example.

Well, for my purpose it could suffice an example that simply opens and interacts via 'feature reports' read and written to a simple (classless) HID device.
That could be a first easy approach to the host USB HID world for ESP32 S2/S3, whereas after that we could proceed to class devices eg keyboard or rmouse if they are a bit more complicated.
So, it could be a good idea if such a simple HID feature report driver could be implemented as it would give a better completeness in the host USB range of device drivers for ESP32 S2/S3 whereas the HID is now missing.

Is there any hint if this has been done already ?

Thanks in advance,

FanhuaCloud, 2022-07-12:
https://github.com/espressif/esp-idf/pull/9191
Looks like someone has already committed code for this, but it hasn't been merged

tore-espressif, 2022-07-12:
@AristarchosOfSamos Thank you for the feature request. We'll try to prioritize HID class in future development, unfortunately we don't have official support for HID yet. (I've seen some community solution, if that interests you)

@FanhuaCloud The PR you linked is for USB HID _device_

AristarchosOfSamos, 2022-07-13:
@tore-espressif thanks for replying to the post.

There are many little USB devices out there with sensors that are accessed via HID with simple 'feature reports' and not need class drivers.
That is why I asked if such a simple example can be easily given. No need for HID host class driver eg keyboard or mouse..
I think that could be a easy example to have in the USB host ESP32 S2/S3 examples.
I'll wait until something comes out officially.

AristarchosOfSamos, 2022-07-30:
Seen that in a latest commit, at least a HID device example is given
[https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_hid](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_hid)

I hope that now consideration  can be turned to where a similar example will be given from the side of a HID host.
A simple USB HID host example where plain 'feature reports' will be read and written. And of course both sides host & device could be implemented with ESP32-S2/S3 so having completeness.

mrweaver, 2022-10-15:
I'm also very keen to see a USB HID host example.  Is there any update on the progress for this?

tore-espressif, 2022-10-18:
> That is why I asked if such a simple example can be easily given. No need for HID host class driver eg keyboard or mouse..

Actually, implementing a mere example for keyboard/mouse is easier, because we don't have to implement all the features and flexibility that HID class defines.

I'd suggest that we provide a simple example for keyboard and mouse combo, and later we can provide a fully-featured HID driver. Does it make sense to you?

@mrweaver do you have a specific device in mind?

mrweaver, 2022-10-18:
This would be amazing! 

A keyboard /mouse combo is perfect I think. I'm looking at a Chinese wireless remote (G20s Airmouse), which is essentially a combination keyboard and mouse as I understand it.

relaxibus, 2022-11-10:
Host USB HID to read joysticks & Co would be amazing. Will the S2/S3 also support USB Hubs when in Host mode?

mrweaver, 2022-12-08:
I was just wondering if there are any updates on the development of this HID host implementation?

Dazza0, 2022-12-08:
@mrweaver The first version of the HID Host Class driver is already in review internally, so we should have something on master for you to test soon. However, the first version will only support HID Boot protocol (basically standard Keyboards an Mice) but we have plans to expand it to full HID support (see below).

@relaxibus @AristarchosOfSamos Joy sticks (and other non-boot protocol HID devices) will take longer to support due to the need to parse HID report descriptors. Currently, there is already some logic in the `esp_hid` component for HID report descriptors. But our long term plan is to tidy up `esp_hid` and integrate USB Host HID with `esp_hid` somehow so that HID applications it can share the same HID descriptor logic with BT/BTLE HID devices as well. This will allow users to...

- use a common  `esp_hid` (or similar) API for all HID devices, regardless of what protocol the device is running on (i.e., USB/BT/BTLE).
- used the provided HID report descriptor APIs to easily add their own drivers for less common/custom HID devices

As for USB Hubs in Host mode, that is also in our roadmap. But, unfortunately I can't offer a release date for that yet.

mrweaver, 2022-12-09:
@Dazza0, that's great news! Thanks for getting back to me so quickly - looking forward to testing it out :D

relaxibus, 2022-12-12:
@Dazza0 thats great news! Thanks for keeping up this USB Host HID topic. Let us know when there are new updated on this topic. :-)

closed 2022-12-14 by espressif-bot

shlomo-edu, 2023-06-15:
Hi to AristarchosOfSamos and tore-espressif
I  have an HID Serial device that I want to connect to a PC via the ESP32-S3 BLE.
I could not find any sample that will enable me to connect the ESP32 (as a host) to the HID device.
Did you solve this problem?
If yes, van you send me a sample code that does it?

relaxibus, 2023-06-15:
As a side question: is HOST USB HID now working for gamepads, joysticks etc?

roma-jam, 2023-06-15:
@relaxibus 

> As a side question: is HOST USB HID now working for gamepads, joysticks etc?

As it was said before by @Dazza0: 
- First version (which is in master already) supports only HID protocols Keyboard and Mouse. 
- Second version (that will be released as an idf-external-component from day to day) supports all HID compatible devices and provide an interface to get an input report raw data. Based on that, any additional support can be done on application level. 

Integration with `esp_hid` is still under a discussion.

relaxibus, 2023-06-15:
Thanks, keep us updated regarding "Integration with esp_hid is still under a discussion."

shlomo-edu, 2023-06-18:
Our software developer (Viki) compiled roma-jam's code, but the problem that it works only for Keyboard and Mouse.
We need the same HID driver that can work with raw data.
Please note, our device uses HID for communication with a PC app using our proprietary protocol.
How can we convert this example to work with raw data.
 https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/hid

roma-jam, 2023-06-19:
Hi @shlomo-edu, 
In the current version there is a verification of the USB device descriptor field such as bInterfaceSubClass. Device is available only if there is a HID_SUBCLASS_BOOT_INTERFACE. 

For disabling, it is possible to remove HID_SUBCLASS_BOOT_INTERFACE comparison statement in the file [hid_host.c, line 282](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/components/hid/hid_host.c#L282)

FYI, there is a pull request to the idf-extra-components with the version of HID driver which provides a better interface of the HID host. Also, there is a fully support HID host requests, that you can easily use them for the proprietary protocol implementation over the HID.
PR can be found here: [ [USB Host] HID driver component ](https://github.com/espressif/idf-extra-components/pull/175)

shlomo-edu, 2023-06-19:
Thanks.
I run the hid_host_Example, and I removed the line 282 and I still don’t see incoming data.
What I do to see the incoming data is:

I connected my HID device to the ESP232-S3 and run the example.
The device sends data in a loop.

In: app_main
In the “do” loop after line 468:  EventBits_t event = xEventGroupWaitBit
I added a log line: ESP_LOGI(TAG, "xEventGroupWaitBits %d",(int) event);
And what I see in the terminal: every second 1 line of:
I (525846) example: xEventGroupWaitBits 0
So I assume it didn’t see the any incoming data.

I looked in PR definition and I don’t think that I need all the functions there because I have only one device that I want to connect to the ESP232-S3
So please help me make this example work (for raw data protocol).

Thanks,
Shlomo.

roma-jam, 2023-06-19:
Hi @shlomo-edu, 

May I ask you to open a new ticket with the issue description and debug output log attached? 

If you want to change the example to be able to get raw data from the HID device, It is better to move that discussion to a new thread. 

Thanks.

shlomo-edu, 2023-06-19:
Hi Roma.
I submitted a new issue named:
Example for ESP32-S3 as HID Host communicate with HID device