espressif/esp-idf #9332: [ESP32-S3][v4.4.1] esp_partition_erase_range() sometimes never returns (& hits WDT)

chipweinberger, 2022-07-12:
**ESP32-S3**
**ESP-IDF v4.4.1**

**Description:**

About ~30% of the time, `esp_partition_erase_range()` never completes and trigger the WDT. 

**Note:** I'm only erasing 4096 bytes, at flash address 0x41000.

It appears to hang here, if the logs are to be believed:
`_DoubleExceptionVector at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:463`

**Expectation:**

`esp_partition_erase_range()` should return an error on failure, and should never trigger the WDT.

**sdkconfig:**
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9088603/sdkconfig.txt)

Logs:

```
V (155877) pd developer: write_partition()
V (155877) pd developer: calling parition json write
V (155887) pd partition json: jBootCmds: write json
I (155897) pd partition json: write jBootCmds, JSON33{"developerMode":0,"cmds":[333]}
V (155907) pd partition json: erase partition (address 0x41000, size 4096, encrypted NO)
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c3
0x403743c3: _DoubleExceptionVector at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:463

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd1400,len:0x265c
load:0x403b6000,len:0x1ca0
load:0x403ba000,len:0x2e90
entry 0x403b6474
W (28) boot.esp32s3: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
W (33) efuse: Loading virtual efuse blocks from real efuses
I (51) boot: ESP-IDF v4.4.1-241-g16b7ed8697-dirty 2nd stage bootloader
I (51) boot: compile time 17:38:07
I (51) boot: chip revision: 0
I (55) boot.esp32s3: Boot SPI Speed : 80MHz
I (60) boot.esp32s3: SPI Mode       : DIO
I (64) boot.esp32s3: SPI Flash Size : 8MB
```

```
    ESP_LOGI(TAG, "write %s, JSON%u%s", partitionName, len, jsonStr);

    PD_LOGV(TAG, "erase partition %p (address %p, size %u, encrypted %s)", partition, 
        partition->address,
        partition->size,
        partition->encrypted ? "YES" : "NO"
    );

    // must always erase entire sector first
    esp_err_t err = esp_partition_erase_range(partition, 0, partition->size);
    if (err != ESP_OK){
        pd_err_fail(error, TAG, "esp_partition_erase_range() %s", esp_err_to_name(err));
        goto end;
    }

    PD_LOGV(TAG, "write sentinel");
```

chipweinberger, 2022-07-12:
I've added some logs to `esp_flash_erase_region()` and now I am always hitting `Cache disabled but cached memory region accessed`, so perhaps this is related to https://github.com/espressif/esp-idf/issues/9129

`Code:`
```
        ESP_LOGI("erase", "4");

        err = rom_spiflash_api_funcs->start(chip);
        if (err != ESP_OK) {
            break;
        }

        ESP_LOGI("erase", "5");
```

**Logs:**
```
I (15961) pd partition json: write jBootCmds, JSON33{"developerMode":0,"cmds":[333]}
V (15971) pd partition json: erase partition 0x3fcf4338 (address 0x41000, size 4096, encrypted NO)
I (15971) erase: start
I (15981) erase: 1
I (15981) erase: 2
I (15981) erase: 3
I (15991) erase: 4
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x421485ca  PS      : 0x00060934  A0      : 0x8037d0ee  A1      : 0x3fcf7210  
0x421485ca: esp_log_writev at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/log/log.c:191

A2      : 0x00000003  A3      : 0x3c1ac778  A4      : 0x3c1ac810  A5      : 0x3fca6c7c  
A6      : 0x3fcf7210  A7      : 0x00000001  A8      : 0x8038d275  A9      : 0x3961e79b  
A10     : 0x3c1ac778  A11     : 0x3c1ac778  A12     : 0x3c1ac810  A13     : 0x3fcf7240  
A14     : 0x3fcf7210  A15     : 0x0000000c  SAR     : 0x0000001f  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x00000000  

Backtrace:0x421485c7:0x3fcf72100x4037d0eb:0x3fcf7260 0x420b3057:0x3fcf72a0 0x42033623:0x3fcf72d0 0x42034ff8:0x3fcf7320 0x42035300:0x3fcf7360 0x4203552e:0x3fcf7390 0x4205f55e:0x3fcf73c0 0x4200ff2d:0x3fcf7410 0x420c5bb1:0x3fcf7450 0x42010d90:0x3fcf7480 0x4200f10b:0x3fcf74b0 0x42148227:0x3fcf7570 0x40387c42:0x3fcf75a0 
0x421485c7: esp_log_level_set at ??:?

0x4037d0eb: esp_flash_erase_region at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/spi_flash/esp_flash_api.c:616 (discriminator 13)
```

chipweinberger, 2022-07-12:
Setting CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL = 92 = sizeof(StaticQueue_t) does not fix https://github.com/espressif/esp-idf/issues/9332#issuecomment-1181243560

ESP-Marius, 2022-07-12:
@chipweinberger I think those logs you added for debugging might be causing their own issues (`Cache disabled but ...`). While erasing the flash the function will disable the cache, so non-IRAM functions are not safe to call at that point.

You could try replacing them with e.g. `esp_rom_printf` (`esp_rom_sys.h`) instead.

chipweinberger, 2022-07-12:
Thanks for the suggestion. 

I tried replacing it `esp_rom_printf` with and it still repros the same way

**Logs:**
```
V (23782) pd partition json: jBootCmds: write json
I (23782) pd partition json: write jBootCmds, JSON77{"developerMode":0,"cmds":[2147483647,2147483647,2147483647,2147483647,333]}
V (23802) pd partition json: erase partition 0x3fcf4920 (address 0x41000, size 4096, encrypted NO)
erase starterase 1erase 2erase 3erase 4Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x40043daa  PS      : 0x00060934  A0      : 0x80044284  A1      : 0x3fcf7870  
A2      : 0x00000000  A3      : 0x3c1ac778  A4      : 0x00000000  A5      : 0x3fcf7870  
A6      : 0x00000004  A7      : 0x00060920  A8      : 0xb50cd406  A9      : 0x3fcf78b0  
A10     : 0x000000ad  A11     : 0xffffff7f  A12     : 0xffffffbf  A13     : 0xfffffc1f  
A14     : 0x00000000  A15     : 0x02ce0b34  SAR     : 0x0000001f  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace:0x40043da7:0x3fcf7870 |<-CORRUPTED

ELF file SHA256: 7a947324a5e1522b

I (5973) esp_core_dump_flash: Save core dump to flash...
I (5979) esp_core_dump_flash: Erase flash 49152 bytes @ 0xc000
erase starterase 1erase 2erase 3erase 4�
                                        к�
                                          к�
                                            к�
```

negativekelvin, 2022-07-12:
Try 

https://github.com/espressif/esp-idf/blob/5810ed1d043cbc54c9a7616db1e84d99f71f3ccb/components/log/include/esp_log.h#L444-L445

ESP-Marius, 2022-07-12:
The `spi_flash` driver should also yield during erasing to avoid WDT timeouts like this.

You can take a look at the related configs in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#os-functions (CONFIG_SPI_FLASH_ERASE_YIELD_X) and see if tweaking any of those help. 

Just make sure to test it without the additional logging.

chipweinberger, 2022-07-12:
@negativekelvin thanks for the suggestion. You were correct. `esp_rom_printf(DRAM_STR("erase 1"));` fixes my logging.

Using `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=94`, I see all the logs and `esp_paritition_erase_range()` works (verified with 4x invocations).

Using `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=72`, ~25% of the time I see the logs up to "erase 5", where it then freezes and hits WDT.

Which corresponds to this code freezing:

```
        esp_rom_printf(DRAM_STR("erase5"));

        bus_acquired = true;

#ifndef CONFIG_SPI_FLASH_BYPASS_BLOCK_ERASE
        // If possible erase an entire multi-sector block
        if (block_erase_size > 0 && len_remain >= block_erase_size && (erase_addr % block_erase_size) == 0) {
            err = chip->chip_drv->erase_block(chip, erase_addr);
            erase_addr += block_erase_size;
            len_remain -= block_erase_size;
        } else
#endif
        {
            // Otherwise erase individual sector only
            err = chip->chip_drv->erase_sector(chip, erase_addr);
            erase_addr += sector_size;
            len_remain -= sector_size;
        }

        esp_rom_printf(DRAM_STR("erase6"));
```

closed 2022-07-12 by chipweinberger