espressif/esp-idf #9333: IDF-4.3.3-ESP32 - Secure boot Start fail, boot write boot key and eufse ABS_DONE_0 fail

Freelifeww, 2022-07-12:
## Environment

- Module or chip used:  ESP32-WROOM-32E
- IDF version (run ``git describe --tags`` to find it):
  ESP-IDF Release v4.3.3
- Build System:     [Make|CMake|idf.py]

## Problem Description
We enable the safe boot and use the specified key compilation mode. At this time, the device starts up, and the randomly generated key writing efuse and the flag bit of the safe boot boot complete code writing (ABS_DONE_0) cannot be completed. The demo verification results are the same

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

/* Flash encryption Example
   This example code is in the Public Domain (or CC0 licensed, at your option.)
   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "soc/efuse_reg.h"
#include "esp_efuse.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "esp_partition.h"
#include "esp_flash_encrypt.h"
#include "esp_efuse_table.h"
#include "nvs_flash.h"

static void example_print_chip_info(void);
static void example_print_flash_encryption_status(void);
static void example_read_write_flash(void);

static const char* TAG = "example";

#if CONFIG_IDF_TARGET_ESP32
#define TARGET_CRYPT_CNT_EFUSE  ESP_EFUSE_FLASH_CRYPT_CNT
#define TARGET_CRYPT_CNT_WIDTH  7
#elif CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32C3
#define TARGET_CRYPT_CNT_EFUSE ESP_EFUSE_SPI_BOOT_CRYPT_CNT
#define TARGET_CRYPT_CNT_WIDTH  3
#endif

void app_main(void)
{
    printf("\nExample to check Flash Encryption status\n");

    example_print_chip_info();
    example_print_flash_encryption_status();
    example_read_write_flash();
    /* Initialize the default NVS partition */
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
}
[... 191 lines cut ...]
D (1468) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (1474) secure_boot_v1: Generating new secure boot key...
D (1481) efuse: coding scheme 0
D (1484) efuse: coding scheme 0
D (1487) efuse: coding scheme 0
D (1490) efuse: coding scheme 0
D (1493) efuse: coding scheme 0
I (1497) secure_boot_v1: Generating secure boot digest...
D (1565) bootloader_flash: mmu set paddr=00000000 count=1 size=ab00 src_addr=1000 src_addr_aligned=0
D (1570) secure_boot_v1: write iv+digest to flash
I (1571) secure_boot_v1: Digest generation complete.
I (1574) boot: Checking secure boot...
I (1578) secure_boot_v1: Read & write protecting new key...
I (1584) secure_boot_v1: blowing secure boot efuse...
D (1590) secure_boot_v1: before updating, EFUSE_BLK0_RDATA6 4
I (1596) secure_boot_v1: Disable JTAG...
I (1600) secure_boot_v1: Disable ROM BASIC interpreter fallback...
D (1607) efuse: coding scheme 0
D (1610) efuse: coding scheme 0
D (1614) efuse: coding scheme 0
D (1617) efuse: coding scheme 0
D (1620) secure_boot_v1: after updating, EFUSE_BLK0_RDATA6 4
E (1626) secure_boot_v1: secure boot not enabled for bootloader image, EFUSE_RD_ABS_DONE_0 is probably write protected!
E (1637) boot: FAILED TO ENABLE SECURE BOOT (259).
abort() was called at PC 0x40079883

mahavirj, 2022-07-12:
@Freelifeww Could you please check fix from https://github.com/espressif/esp-idf/issues/9290#issuecomment-1174952889? This fix is already merged in internal codebase, it shall be available on GH with next v4.3 branch sync.

Freelifeww, 2022-07-12:
@mahavirj Thanks, I‘m check it.

Freelifeww, 2022-07-12:
> @Freelifeww Could you please check fix from [#9290 (comment)](https://github.com/espressif/esp-idf/issues/9290#issuecomment-1174952889)? This fix is already merged in internal codebase, it shall be available on GH with next v4.3 branch sync.

Thank you. The test was successful

mahavirj, 2022-07-12:
@Freelifeww Thanks for confirming. I will close this issue and allow fix to be tracked through https://github.com/espressif/esp-idf/issues/9290

closed 2022-07-12 by mahavirj