espressif/esp-idf #9337: IDF-4.3.3-ESP32-NVS key is not encrypted and written to flash

Freelifeww, 2022-07-12:
## Environment

- Module or chip used:  ESP32-WROOM-32E
- IDF version (run ``git describe --tags`` to find it):
    V4.3.3
- Build System:         [Make|CMake|idf.py]
## Problem Description
I understand that the code generated key stored in the NVS key area needs to be encrypted by flash to complete the encryption write, but in this scheme, the encryption key in the NVS area is written in plain text, which is obviously not secure

### Code to reproduce this issue
static esp_err_t nvs_parting_secure_init(void)
{

    esp_err_t ret = ESP_OK;
    const esp_partition_t *key_part = esp_partition_find_first( ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS_KEYS, NULL);
    if(NULL == key_part) {
        printf("CONFIG_NVS_ENCRYPTION is enabled, but no partition with subtype,nvs_keys found in the partition table.\n");
        printf("nvs_flash_init.....\n");
        ret = nvs_flash_init();
        if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
            ret = nvs_flash_erase();
            if(ret != ESP_OK) {
                printf("nvs_flash_init ERR ,nvs_flash_erase is ERR\n");
                return ret;
            }
            ret = nvs_flash_init();
            if(ret != ESP_OK) {
                printf("nvs_flash_init ERR ,nvs_flash_erase is ok,nvs_flash_init err\n");
                return ret;
            }
        }
        return ret;
    }

    nvs_sec_cfg_t cfg;
    memset(&cfg,0,sizeof(nvs_sec_cfg_t));

    ret = nvs_flash_read_security_cfg(key_part, &cfg);
    if (ret != ESP_OK) {
    /* We shall not generate keys here as that must have been done in default
        * NVS partition initialization case */
        ret = nvs_flash_erase();
        if(ret != ESP_OK) {
            printf("nvs_flash_erase ...\n");
            return ret;
        }
        printf("Failed to read NVS security cfg: [0x%02X] (%s)\n",ret,esp_err_to_name(ret));
        ret = nvs_flash_generate_keys(key_part, &cfg);
        if (ret != ESP_OK) {
            printf("nvs_flash_generate_keys err:%d", ret);
            return ret;
        }
    }

    ret = nvs_flash_secure_init_partition(NVS_DEFAULT_PART_NAME, &cfg);
    if (ret == ESP_OK) {
        printf("---nvs partition nvs is encrypted----\n");
    }else {
        printf("---nvs partition nvs is encrypted err----\n");
    }

    return ret;
}

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = ESP_OK;
    extern OPERATE_RET tuya_start_mac_for_file_auth_set_dev(void); // user entry
    tuya_start_mac_for_file_auth_set_dev();
[... 1757 lines cut ...]
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options

https://docs.espressif.com/projects/esp-idf/en/v4.3.3/esp32/api-reference/storage/nvs_flash.html
[image]

mahavirj, 2022-07-12:
@Freelifeww 

Actual NVS encryption key is derived from flash contents through flash decryption scheme, please refer to:

https://github.com/espressif/esp-idf/blob/9b8c558e63d95b491355a22e69f247834e899b47/components/nvs_flash/src/nvs_api.cpp#L593-L597

Freelifeww, 2022-07-13:
> @Freelifeww
> 
> Actual NVS encryption key is derived from flash contents through flash decryption scheme, please refer to:
> 
> https://github.com/espressif/esp-idf/blob/9b8c558e63d95b491355a22e69f247834e899b47/components/nvs_flash/src/nvs_api.cpp#L593-L597

Thanks, I'm check read key data,Thank you again for your prompt answer,

Freelifeww, 2022-07-13:
> @Freelifeww
> 
> Actual NVS encryption key is derived from flash contents through flash decryption scheme, please refer to:
> 
> https://github.com/espressif/esp-idf/blob/9b8c558e63d95b491355a22e69f247834e899b47/components/nvs_flash/src/nvs_api.cpp#L593-L597

After verification, it is really as you instructed, thank you for your guidance

closed 2022-07-13 by Freelifeww