espressif/esp-idf #9339: ESP32-S3 IDF5.0 WIFI connect too slow

FanhuaCloud, 2022-07-12:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-S3-WROOM-1-N8R8
- IDF version (run ``git describe --tags`` to find it):
    // v5.0-dev-4037-g9b8c558e63
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32s3-elf-gcc.exe (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes VSCode
- Power Supply:         USB

## Problem Description

When I use esp_wifi_connect() it takes a long time to connect WIFI

### Steps to reproduce

1. use this example test
https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station

## Debug Logs

```
I (1338) net80211: net80211 rom version: e7ae62f
I (1348) wifi:wifi driver task: 3fce9138, prio:23, stack:6656, core=0
I (1348) system_api: Base MAC address is not set
I (1348) system_api: read default base MAC address from EFUSE
I (1348) wifi:wifi firmware version: 2a8b962
I (1348) wifi:wifi certification version: v7.0
I (1348) wifi:config NVS flash: enabled
I (1358) wifi:config nano formating: disabled
I (1358) wifi:Init data frame dynamic rx buffer num: 32
I (1358) wifi:Init management frame dynamic rx buffer num: 32
I (1358) wifi:Init management short buffer num: 32
I (1358) wifi:Init static tx buffer num: 16
I (1358) wifi:Init tx cache buffer num: 32
I (1358) wifi:Init static tx FG buffer num: 2
I (1358) wifi:Init static rx buffer size: 1600
I (1358) wifi:Init static rx buffer num: 10
I (1358) wifi:Init dynamic rx buffer num: 32
I (1358) wifi_init: rx ba win: 6
I (1358) wifi_init: tcpip mbox: 32
I (1358) wifi_init: udp mbox: 6
I (1358) wifi_init: tcp mbox: 6
I (1358) wifi_init: tcp tx win: 5744
I (1358) wifi_init: tcp rx win: 5744
I (1358) wifi_init: tcp mss: 1440
I (1358) wifi_init: WiFi/LWIP prefer SPIRAM
I (1358) wifi_init: WiFi IRAM OP enabled
I (1358) wifi_init: WiFi RX IRAM OP enabled
I (1368) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (1398) wifi:mode : sta (7c:df:a1:fd:f9:c0)
I (1398) wifi:enable tsf
I (1398) wifi:Set ps type: 1

I (1398) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1408) wifi:new:<6,2>, old:<1,0>, ap:<255,255>, sta:<6,2>, prof:1
I (1408) wifi:state: init -> auth (b0)
I (1418) wifi:state: auth -> assoc (0)
I (1448) wifi:state: assoc -> run (10)
```
and then,it use many many time

FanhuaCloud, 2022-07-12:
Further confirm this commit：
https://github.com/espressif/esp-idf/commit/a7ae72e1f5a420716a42e1a150586f7aef768c45
This is my sdkconfig
https://gist.github.com/FanhuaCloud/ef38030f293dfe4465d5da459607400a

MaxwellAlan, 2022-07-14:
Hi @FanhuaCloud 
Could you show me the full log?
And could you capture wireless packets? If yes, also upload the pkts.
What's the configs of your AP?
Thanks.

FanhuaCloud, 2022-07-14:
```
I (24) boot: ESP-IDF v5.0-dev-4037-g9b8c558e63 2nd stage bootloader
I (25) boot: compile time 21:18:52
I (25) boot: chip revision: V001
I (28) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (35) qio_mode: Enabling default flash chip QIO
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : QIO
I (50) boot.esp32s3: SPI Flash Size : 8MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 001f4000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=ee5e8h (976360) map
I (261) esp_image: segment 1: paddr=000fe610 vaddr=3fc9c000 size=01a08h (  6664) load
I (263) esp_image: segment 2: paddr=00100020 vaddr=42000020 size=d1958h (858456) map
I (397) esp_image: segment 3: paddr=001d1980 vaddr=3fc9da08 size=03314h ( 13076) load
I (400) esp_image: segment 4: paddr=001d4c9c vaddr=40378000 size=13f34h ( 81716) load
I (418) esp_image: segment 5: paddr=001e8bd8 vaddr=50000000 size=00010h (    16) load
I (427) boot: Loaded app from partition at offset 0x10000
I (428) boot: Disabling RNG early entropy source...
I (429) octal_psram: ECC is enabled
I (432) octal_psram: vendor id    : 0x0d (AP)
I (437) octal_psram: dev id       : 0x02 (generation 3)
I (443) octal_psram: density      : 0x03 (64 Mbit)
I (448) octal_psram: good-die     : 0x01 (Pass)
I (454) octal_psram: Latency      : 0x01 (Fixed)
I (459) octal_psram: VCC          : 0x01 (3V)
I (464) octal_psram: SRF          : 0x01 (Fast Refresh)
I (470) octal_psram: BurstType    : 0x00 ( Wrap)
I (475) octal_psram: BurstLen     : 0x03 (1024 Byte)
I (481) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (487) octal_psram: DriveStrength: 0x00 (1/1)
W (492) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (503) esp_psram: Found 8MB SPI RAM device
I (507) esp_psram: Speed: 80MHz
I (591) mmu_psram: Instructions copied and mapped to SPIRAM
I (668) mmu_psram: Read only data copied and mapped to SPIRAM
I (668) cpu_start: Pro cpu up.
I (669) cpu_start: Starting app cpu, entry point is 0x403796a8
0x403796a8: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-master/components/esp_system/port/cpu_start.c:147

I (0) cpu_start: App cpu up.
I (990) esp_psram: SPI SRAM memory test OK
I (1054) cpu_start: Pro cpu start user code
I (1054) cpu_start: cpu freq: 240000000 Hz
I (1054) cpu_start: Application information:
I (1054) cpu_start: Project name:     test-lvgl
I (1055) cpu_start: App version:      9937e99
I (1055) cpu_start: Compile time:     Jul 14 2022 21:18:47
I (1056) cpu_start: ELF file SHA256:  09491b0a72ea9c3b...
I (1057) cpu_start: ESP-IDF:          v5.0-dev-4037-g9b8c558e63
I (1058) heap_init: Initializing. RAM available for dynamic allocation:
I (1059) heap_init: At 3FCA28A0 len 0003D760 (245 KiB): D/IRAM
I (1060) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (1061) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (1062) esp_psram: Adding pool of 5812K of external SPI memory to heap allocator
I (1064) spi_flash: detected chip: generic
I (1064) spi_flash: flash io: qio
I (1065) sleep: Configure to isolate all GPIO pins in sleep state
I (1065) sleep: Enable automatic switching of GPIO sleep configuration
I (1067) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1067) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1087) lvgl_gui: Turn off LCD backlight
I (1087) lvgl_gui: Initialize Intel 8080 bus
[... 37 lines cut ...]
I (1417) wifi:mode : sta (MAC)
I (1417) wifi:enable tsf
I (1417) wifi:Set ps type: 1

I (1417) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1427) wifi:new:<6,2>, old:<1,0>, ap:<255,255>, sta:<6,2>, prof:1
I (1427) wifi:state: init -> auth (b0)
I (1427) wifi:state: auth -> assoc (0)
I (1437) wifi:state: assoc -> run (10)
I (11437) wifi:state: run -> init (cc00)
I (11437) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
I (13847) wifi:new:<6,2>, old:<6,0>, ap:<255,255>, sta:<6,2>, prof:1
I (13857) wifi:state: init -> auth (b0)
I (13857) wifi:state: auth -> assoc (0)
I (13877) wifi:state: assoc -> run (10)
W (13887) wifi:<ba-add>idx:0 (ifx:0, MAC), tid:5, ssn:0, winSize:64
I (13977) wifi:connected with WIFI_NAME, aid = 2, channel 6, 40D, bssid = MAC
I (13977) wifi:security: WPA2-PSK, phy: bgn, rssi: -74
I (13987) wifi:pm start, type: 1

I (13987) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (13987) wifi:BcnInt:102400, DTIM:1
I (14857) esp_netif_handlers: sta ip: 172.16.1.26, mask: 255.255.0.0, gw: 172.16.0.1
W (15707) wifi:<ba-add>idx:1 (ifx:0, MAC), tid:0, ssn:1, winSize:64
```

FanhuaCloud, 2022-07-14:
My AP is using 11bgn mixed mode, the band bandwidth is 40mhz/20mhz automatic

FanhuaCloud, 2022-07-14:
Sometimes when I first power it up it connects quickly

MaxwellAlan, 2022-07-20:
Hi @FanhuaCloud 
>I (11437) wifi:state: run -> init (cc00)

From your log reason code cc00 means that 4-way handshake timeout, there are some situation will cause this, one is wrong password, the other is that the wireless air is a little crowd. Maybe you can try disable modem sleep(call esp_wifi_set_ps(0); after esp_wifi_start();) and when you got IP then re-enable modem sleep or not depends on yourself.
Thanks.

closed 2022-07-24 by FanhuaCloud

rtheil-growlink, 2022-09-20:
I'm experiencing this exact same issue on IDF 4.4.2. A power cycle results in a fast connection. This issue does not exist in IDF 4.3.x. 

Adding esp_wifi_set_ps(WIFI_PS_NONE); after esp_wifi_start() does not fix the issue on first boot after flashing. On second and subsequent boots it's a fast connection. I've tried this in a couple different environments on a couple different access points, and the behavior is always the same.

FanhuaCloud, 2022-09-21:
> 

This problem doesn't exist when I use other routers

rtheil-growlink, 2022-09-21:
Thanks for that update. I will do some additional testing on other devices and see what the results are. But so far, once I set that ps level to 0, it seems to work fine.

AxelLin, 2024-06-14:
> My AP is using 11bgn mixed mode, the band bandwidth is 40mhz/20mhz automatic And,In the log I marked the location that took a long time. @MaxwellAlan

I have the same bad experience when test with esp32s3.
I test with tp-link deco, 11bgn mixed mode, the band bandwidth is 40mhz/20mhz automatic.
esp32 is stable in my test, it only takes 2 ~ 3 seconds to upload file to https server.
With esp32s3, it sometimes take more than 10 seconds to upload file to https server.
So I'm windering if this is a known issue for esp32s3?

MaxwellAlan, 2024-06-14:
Hi @AxelLin 

> I have the same bad experience when test with esp32s3. I test with tp-link deco, 11bgn mixed mode, the band bandwidth is 40mhz/20mhz automatic. esp32 is stable in my test, it only takes 2 ~ 3 [...]

Which IDF version/commit that you test with?

We do not observe related issue in our release test, could you provide some WiFi sniifer log for we to analysis?

MaxwellAlan, 2024-06-14:
Hi @AxelLin 

Did the WIFI dongle support Monitor mode? And check the dongle working channel and bandwith when sniffer .

AxelLin, 2024-06-19:
@MaxwellAlan 
Please take a look: https://github.com/espressif/esp-idf/issues/14008 .
I believe there are something wrong with esp32s3 wifi (because esp32 has no issue when test at the same time).