espressif/esp-idf #9344: Linking broken

francis2tm, 2022-07-13:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3
- Build System:         [Make|CMake|idf.py]

## Problem Description

I have exactly the same problem as https://github.com/espressif/esp-idf/issues/3741
NO one fixed there the problem.
I have multiple components in a folder called "examples". I tried:
`set(EXTRA_COMPONENT_DIRS "examples/")`
And a component located in the folder "components" says:
`undefined reference to `adcCallback'`
the function `adcCallback`is in on of the components located in "examples".

And yes, I specified all my sources and REQUIRES in each idf_component_register. I also already tried adding each external component with "idf_build_component" but it didn't work.

igrr, 2022-07-13:
Hi @francis2tm,
Could you please attach the full build log (after doing `idf.py fullclean` and `idf.py build`)? Could you also describe the layout of your project in a bit more detail, it possible? (Or upload your project or part of it, if possible)
This will help us give you some suggestions for troubleshooting this issue.

francis2tm, 2022-07-13:
Hello,
Here is the project without the sources. In the directories that there is "some_sources.c" it's because that's where the component's sources are supposed to be located.
[project.zip](https://github.com/espressif/esp-idf/files/9101424/project.zip)

Here is the full build log:
[build.log](https://github.com/espressif/esp-idf/files/9101466/build.log)

Essentially in components/scientisst-sense-firmware/scientisst-sense-firmware/main/adc.c it includes a function:
`extern void adcCallback(uint16_t sample);`

from examples/ecg-biometrics/preprocessing.c.

This is where the link get's broken, with an error:
`components/scientisst-sense-firmware/scientisst-sense-firmware/main/adc.c:181: undefined reference to `adcCallback'`

igrr, 2022-07-13:
@francis2tm Thank you for sharing the files!

I think the issue is that `BINEDGE_EXAMPLE` variable used in `main/CMakeLists.txt` is not set, therefore `ecg-biometrics` component is not added as a dependency to `main`, and therefore the function reference doesn't get resolved.

Please try replacing `${BINEDGE_EXAMPLE}` in `main/CMakeLists.txt` and see if it works.

When you do `set(BINEDGE_EXAMPLE "ecg-biometrics")` in the project CMakeLists.txt file, this variable is not propagated to the component CMakeLists files, since the component CMakeLists files are evaluated in different CMake scope.

You can try to use `idf_build_set_property` function in the project CMakeLists.txt file and then retrieve it using `idf_build_get_property` in the `main` component CMakeLists.txt file. (To debug, you can add `message(STATUS "BINEDGE_EXAMPLE: ${BINEDGE_EXAMPLE}")` in your component CMakeLists.txt file.)

francis2tm, 2022-07-13:
Hey @igrr ,
Thanks for the quick reply.
Are you sure? because when I print the variable `BINEDGE_EXAMPLE` in `main/CMakeLists.txt` it appears:
`-- BINEDGE_EXAMPLE: ecg-biometrics`

francis2tm, 2022-07-13:
eheh @igrr , I found the problem!
It was in `components/scientisst-sense-firmware/scientisst-sense-firmware/main/CMakeLists.txt`. The variable `BINEDGE_EXAMPLE` was not well set and therefore the component `ecg-biometrics` was not included in the REQUIRES

igrr, 2022-07-13:
Thanks for confirming this fixes the problem @francis2tm! Can we close this issue?

francis2tm, 2022-07-20:
Hello, the problem actually was not solved. It was only solved because I hardcoded the value of `BINEDGE_EXAMPLE` into the code, but now I really need that `BINEDGE_EXAMPLE` is a global variable.

In the root project's CMakeLists.txt:
`set_property(GLOBAL PROPERTY BINEDGE_EXAMPLE_PROPERTY "ecg-biometrics")`

But then in `components/scientisst-sense-firmware/scientisst-sense-firmware/main/CMakeLists.txt` if I do:

`idf_build_get_property(var BINEDGE_EXAMPLE)
message(STATUS "binedge example:${var}")`

It prints:
`binedge example:

igrr, 2022-07-20:
I see, thanks for this extra detail. I think I understand the problem.

Component CMakeLists.txt files are processed twice during the build:

- First time the component CMakeLists.txt file is evaluated in CMake scripting mode, only to obtain the values passed to `REQUIRES` and `PRIV_REQUIRES` arguments of `idf_component_regsiter`. This step is necessary to figure out component dependencies and decide which components to include into the build. This happens very early in the build process. For example, Kconfig options are not resolved yet at this stage. CMake properties defined in the parent project are not available in the component CMakeLists.txt file at this stage, since the file is evaluated in CMake scripting mode (in a separate CMake process).
- Second time, component CMakeLists.txt file is evaluated when the component is added to CMake project as a library. At this time, all Kconfig options and build properties are available.

Since you are trying to use the build property to configure component dependencies, and since the build properties of the main project are not available when evaluating the component CMakeLists.txt file in interpreter mode, the build property ends up being empty.

There isn't an easy way to resolve this limitation of the build system. However you can solve this problem in a different manner:

* in the project CMakeLists.txt file, ensure that the component you wish to add as a requirement (`ecg-biometrics`) is included into the build. (This is the default behavior, unless you override the `COMPONENTS` variable.)
* in the component CMakeLists.txt file, don't add `BINEDGE_EXAMPLE` to REQUIRES or PRIV_REQUIRES. Instead, after the `idf_component_register` call, add another line:
  ```cmake
  target_link_libraries(${COMPONENT_LIB} PRIVATE idf::${BINEDGE_EXAMPLE})
  ```

You can see an example of the similar approach here: https://github.com/espressif/esp-idf/blob/508661b03eec6e15af66f77050e699a7658b9793/components/nvs_flash/CMakeLists.txt#L32-L35

Hope this helps!