espressif/esp-idf #9345: panic'ed (LoadProhibited) if using pppos

AxelLin, 2022-07-13:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v5.0-dev-4037-g9b8c558e63 (master) and v5.0-dev-3954-g271f4e4c40 (5.0 branch)
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

My application hit below panic if build with current master tree or v5.0 branch.
It was working fine if build with v4.x branches. So this is a regression.

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40103900  PS      : 0x00060c30  A0      : 0x8012000e  A1      : 0x3ffc3500
0x40103900: esp_netif_internal_dhcpc_cb at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1120
 (inlined by) netif_callback_fn at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:117

A2      : 0x3ffd8ec0  A3      : 0x00000000  A4      : 0x3ffd903c  A5      : 0x00000000
A6      : 0x0a060a06  A7      : 0x00000000  A8      : 0x801299b1  A9      : 0x3ffc3510
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffc228c  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x3ffc2238  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff

Backtrace: 0x401038fd:0x3ffc3500 0x4012000b:0x3ffc3540 0x4012007e:0x3ffc3560 0x4012011e:0x3ffc35a0 0x401285d4:0x3ffc35d0 0x40128b9a:0x3ffc3610 0x4012877e:0x3ffc3630 0x4011eb29:0x3ffc3650
0x401038fd: esp_netif_internal_dhcpc_cb at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1117
 (inlined by) netif_callback_fn at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:117

0x4012000b: netif_invoke_ext_callback at /home/axel/esp/esp-idf/components/lwip/lwip/src/core/netif.c:1819

0x4012007e: netif_set_addr at /home/axel/esp/esp-idf/components/lwip/lwip/src/core/netif.c:733

0x4012011e: netif_add at /home/axel/esp/esp-idf/components/lwip/lwip/src/core/netif.c:376

0x401285d4: ppp_new at /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/ppp.c:704

0x40128b9a: pppos_create at /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/pppos.c:186

0x4012877e: pppapi_do_pppos_create at /home/axel/esp/esp-idf/components/lwip/lwip/src/netif/ppp/pppapi.c:132

0x4011eb29: tcpip_thread_handle_msg at /home/axel/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:166
 (inlined by) tcpip_thread at /home/axel/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:148

david-cermak, 2022-07-13:
Hi @AxelLin 

Thank you for the report and sorry for the regression.
The quick patch is to:
```patch
diff --git a/components/esp_netif/lwip/esp_netif_lwip.c b/components/esp_netif/lwip/esp_netif_lwip.c
--- a/components/esp_netif/lwip/esp_netif_lwip.c
+++ b/components/esp_netif/lwip/esp_netif_lwip.c
@@ -1107,6 +1107,9 @@ static void esp_netif_internal_dhcpc_cb(struct netif *netif)
         ESP_LOGD(TAG, "null netif=%p", netif);
         return;
     }
+    if (netif->name[0] == 'p' && netif->name[1] == 'p') {
+        return;
+    }
     ESP_LOGD(TAG, "%s lwip-netif:%p", __func__, netif);
 
     esp_netif_t *esp_netif = esp_netif_get_handle_from_netif_impl(netif);
@@ -1804,7 +1807,7 @@ esp_ip6_addr_type_t esp_netif_ip6_get_addr_type(esp_ip6_addr_t* ip6_addr)
 static void esp_netif_internal_nd6_cb(struct netif *p_netif, uint8_t ip_index)
 {
     ESP_LOGD(TAG, "%s lwip-netif:%p", __func__, p_netif);
-    if (!p_netif) {
+    if (!p_netif || !p_netif->state) {
         ESP_LOGD(TAG, "esp_netif_internal_nd6_cb called with null p_netif");
         return;
     }
```

Actual fix coming soon.

david-cermak, 2022-07-13:
Thanks for the correction!
Yes, the interface name is set (few lines) after setting the IP which invoked the callback.

And yes, I'll add more general fix.

closed 2022-07-29 by espressif-bot