espressif/esp-idf #9348: one of two esp32-wroom-32 not obtaining an IP address, same code used (wifi station example)

ivnny, 2022-07-13:
Different behavior for the same hardware, one of the two esp32 not obtaining an IP address using the same example code (wifi station example, menuconfig all default, custom SSID pass)

--- idf_monitor on /dev/cu.usbserial-0001 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14788
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (27) boot: compile time 16:38:42
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1658ch ( 91532) map
I (143) esp_image: segment 1: paddr=000265b4 vaddr=3ffb0000 size=038d8h ( 14552) load
I (149) esp_image: segment 2: paddr=00029e94 vaddr=40080000 size=06184h ( 24964) load
I (159) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6e5c0h (452032) map
I (323) esp_image: segment 4: paddr=0009e5e8 vaddr=40086184 size=0e28ch ( 57996) load
I (347) esp_image: segment 5: paddr=000ac87c vaddr=50000000 size=00010h (    16) load
I (357) boot: Loaded app from partition at offset 0x10000
I (357) boot: Disabling RNG early entropy source...
I (369) cpu_start: Pro cpu up.
D (370) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (370) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
D (376) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
D (383) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 2 bit
I (390) cpu_start: Starting app cpu, entry point is 0x400811a4
0x400811a4: call_start_cpu1 at /Users/usero/esp-idf-dev/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
D (404) clk: RTC_SLOW_CLK calibration value: 3659059
I (417) cpu_start: Pro cpu start user code
I (417) cpu_start: cpu freq: 160000000
I (417) cpu_start: Application information:
I (421) cpu_start: Project name:     wifi_station
I (427) cpu_start: App version:      1
I (431) cpu_start: Compile time:     Jul 13 2022 16:38:33
I (437) cpu_start: ELF file SHA256:  1bc36cbaf00c3a22...
I (443) cpu_start: ESP-IDF:          v4.4.1-dirty
V (449) memory_layout: reserved range is 0x3f416574 - 0x3f4165ac
D (455) memory_layout: Checking 7 reserved memory ranges:
D (460) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (467) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb7578
D (473) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (479) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (486) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (492) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /Users/usero/esp-idf-dev/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1736

D (499) memory_layout: Reserved memory range 0x40080000 - 0x40094410
0x40080000: _WindowOverflow4 at /Users/usero/esp-idf-dev/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1736

[... 498 lines cut ...]
D (2570) esp_netif_lwip: check: remote, if=0x3ffbdb30 fn=0x400dcf60
0x400dcf60: esp_netif_up_api at /Users/usero/esp-idf-dev/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1256

D (2570) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffbdb30
D (2580) esp_netif_lwip: check: local, if=0x3ffbdb30 fn=0x400dd64c
0x400dd64c: esp_netif_update_default_netif_lwip at /Users/usero/esp-idf-dev/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:188

D (2580) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbdb30
V (2590) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbdb30
D (2600) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (2600) esp_netif_lwip: check: remote, if=0x3ffbdb30 fn=0x400dcbcc
0x400dcbcc: esp_netif_dhcpc_start_api at /Users/usero/esp-idf-dev/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1049

D (2610) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3ffbdb30
D (2620) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbdb30
D (2620) esp_netif_lwip: if0x3ffbdb30 start ip lost tmr: no need start because netif=0x3ffbdbb0 interval=120 ip=0
D (2630) esp_netif_lwip: starting dhcp client
D (2640) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (2640) event: running post WIFI_EVENT:4 with handler 0x400d60dc and context 0x3ffc4e24 on loop 0x3ffbcc04
0x400d60dc: event_handler at /Users/usero/eps/Snake/DeviceD/esp_wifi_station_example/station/build/../main/station_example_main.c:46

I (2760) wifi:AP's beacon interval = 102400 us, DTIM period = 1
D (2760) wifi:set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
D (2760) wifi:sig_b=0, sig_g=0, sig_n=0, max_b=22, max_g=108, max_n=144
D (2770) wifi:update trc

oroberman, 2022-07-13:
more info from my side:
strange behavior 1
from the logs we see 
`wifi:connected with NEXUS_HOME`

however, if I'm changing the network name on my access point to NEXUS_HOME_2.4 I still see in the logs the same: `wifi:connected with NEXUS_HOME`

strange behavior 2
in logs i see
`connected with NEXUS_HOME, aid = 6, channel 1`

where my access point configured to channel 11

MaxwellAlan, 2022-07-18:
Hi @oroberman @ivnny 
From your describtion and log, it really wired. 
The log shows that WiFi already connect but lack of 4-way handshake log, can you enable "Print debug messages from WPA Supplicant"?
While the ssid and channel issue, it seems that there is an another AP which ssid is NEXUS_HOME and in channel 1, and when you change your AP ssid, you also need to change station SSID name in your code. Can you capture wireless pkts in channel 1 and channel 11, which can explain why this happened.
Thanks.

oroberman, 2022-07-18:
> Hi @oroberman @ivnny From your describtion and log, it really wired. The log shows that WiFi already connect but lack of 4-way handshake log, can you enable "Print debug messages from WPA [...]

@MaxwellAlan 
thank you for your remark.  I recalled that I have a TP-LINK wifi repeater that retranslates the same SSID and it works on channel **11**

The question now is why we always connect to the repeater and not to the router even if the router is with better RSSI
======================================================================
             SSID             |    RSSI    |           AUTH           
======================================================================
                **NEXUS_HOME**    |     -44    |        WIFI_AUTH_WPA2_PSK
                   Liornet    |     -56    |        WIFI_AUTH_WPA2_PSK
                **NEXUS_HOME**    |     -60    |    WIFI_AUTH_WPA_WPA2_PSK
                      Reut    |     -75    |    WIFI_AUTH_WPA_WPA2_PSK
                      

The first one is the router and the second is the TP-LINK repeater.

MaxwellAlan, 2022-07-19:
Hi @oroberman 
Seems like related with channel, can you try exchange both AP channel ? AP 1 channel 1 change to 11, AP2 channel 11 change to 1 and retest.
If related with channel, maybe you need to try all-channel scan, ref to: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#esp32-wi-fi-scan
Thanks.

ivnny, 2022-07-27:
It happened when connecting to the wifi repeater instead of the router, so when esp32 connects to the router, everything is ok with IP assigning.
Thanks!

MaxwellAlan, 2022-07-29:
> It happened when connecting to the wifi repeater instead of the router, so when esp32 connects to the router, everything is ok with IP assigning. Thanks!

So that maybe repeater issue. Or could you capture wireless pkts? Maybe you can upload some pkts for us to analyse.

ivnny, 2022-07-29:
Hi, we'll take a look for more details for the analysis
Thanks!

MaxwellAlan, 2022-08-04:
Hi @ivnny 

Could you share more details about wifi repeater (type, version etc.) ?

ivnny, 2022-08-04:
> Hi @ivnny
> 
> Could you share more details about wifi repeater (type, version etc.) ?

Hi, tp-link wifi av500

MaxwellAlan, 2022-08-09:
Hi @ivnny 

Did your test topology like this https://www.youtube.com/watch?v=XVGjix7nisw ? One Router with two AV500 powerline extender ?

ivnny, 2022-08-09:
> Hi @ivnny
> 
> Did your test topology like this https://www.youtube.com/watch?v=XVGjix7nisw ? One Router with two AV500 powerline extender ?

Hi, not yet, we’ll try
thanks

MaxwellAlan, 2022-08-09:
Hi @ivnny 

It's difficult for us to buy the same type av500, this tpye seems only for overseas. It will take some time for us to setup the same test environment. It's very appreciate if you could help to capture the get ip fail pkts for us to analyse.

Thanks.

closed 2023-02-17 by Alvin1Zhang