espressif/esp-idf #9351: TWAI Driver using FreeRTOS, Is there anyway to use it without RTOS?

bhardadinesh-eaton, 2022-07-14:
TWAI.c and TWAI.h driver files using FREERTOS. I want to use it without FREERTOS. Is there anyway or other files to use it without any RTOS.

Dazza0, 2022-07-14:
@bhardadinesh-eaton I don't think that is possible without major modifications to the TWAI driver. You can however write your own non-OS TWAI driver by calling the `twai_hal.h`. HAL APIs in ESP-IDF do not use any FreeRTOS APIs or primitives.

bhardadinesh-eaton, 2022-07-14:
@Dazza0 Thank you!. Yes, I can write my own driver for the TWAI using twai_hal.h. I wanted to use this twai.c file which is derived from the twai_hal.h for zephyr OS but it is giving error as there is no FreeRTOS is available to supports it's APIs.
I can write our own driver for zephyr OS using twai_hal.h.

sylvioalves, 2022-07-14:
@bhardadinesh-eaton, CAN driver is in progress. You can check initial draft in here: 
https://github.com/henrikbrixandersen/zephyr/commits/can_sja1000

bhardadinesh-eaton, 2022-07-14:
@sylvioalves Shall I know, What is pending for this CAN driver and when it will be available for use?

bhardadinesh-eaton, 2022-07-20:
@henrikbrixandersen I have gone through the initial draft. Can you please give some update what is pending for the CAN driver for ESP32?

henrikbrixandersen, 2022-07-20:
I am just now working on the bus recovery feature, which should be the one of the last features. I expect a draft PR to be submitted either today or tomorrow.

bhardadinesh-eaton, 2022-07-20:
@henrikbrixandersen Thank you! Once the PR is raised. Please let me know. So, I can use it for my development. Thank you!! once again.

henrikbrixandersen, 2022-07-21:
> @henrikbrixandersen Thank you! Once the PR is raised. Please let me know. So, I can use it for my development. Thank you!! once again.

https://github.com/zephyrproject-rtos/zephyr/pull/48124

bhardadinesh-eaton, 2022-07-21:
> > @henrikbrixandersen Thank you! Once the PR is raised. Please let me know. So, I can use it for my development. Thank you!! once again.
> 
> [zephyrproject-rtos/zephyr#48124](https://github.com/zephyrproject-rtos/zephyr/pull/48124)

I will go through and understand the same. Thank you!

bhardadinesh-eaton, 2022-07-23:
@henrikbrixandersen Sorry to bothering you again and again. I have understood the CAN drivers. Do we have example code to test the loopback for ESP32 of TWAI ( CAN ). I can see driver test code for timing available in tests folder of zephyr repo. Also, I have looked the example code. it is older one which are using FreeRTOS.

Dazza0, 2023-02-27:
@bhardadinesh-eaton Closing this issue as Zephyr will be adding CAN/TWAI support. Please feel free to reopen this issue if needed.

closed 2023-02-27 by Dazza0