espressif/esp-idf #9354: Master Modbus RTU transmission failure (GLITCH) with Wifi STA connection

JavierSVE, 2022-07-14:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for DevKitC): v4
- Module or chip used:  ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: Yes (Visual Studio 2019 plus Visual GDB plugin)
- Power Supply:         USB

## Problem Description

I migrated my application from version 4.2 to 4.4.1 and now errors appear in the modbus communication.
My application consists of a web server via a wifi access point with an embedded web page  and a connection with a slave device via modbus RTU.
My application poll the slave every 2 seconds with requests of the type, mbc_master_send_request() and randomly appear failed requests with ESP_ERR_TIMEOUT and ESP_ERR_INVALID_REQUEST.
I believe that there is some task that interrupts the diver that controls the uart and causes the glitchs.

### Expected Behavior

Request OK:
[image]

### Actual Behavior

Request with a GLITCH:
[image]

### Steps to reproduce

To reproduce the problem I have used the examples that come with the idf:
- C:\SysGCC\esp32\esp-idf\v4.4.1\examples\protocols\modbus\serial\mb_master
- C:\SysGCC\esp32\esp-idf\v4.4.1\examples\protocols\modbus\serial\mb_slave
- C:\SysGCC\esp32\esp-idf\v4.4.1\examples\protocols\http_server\file_serving
I use an ESP32-DevKitC to run the master and another ESP32-DevKitC to run the slave and connect them directly without using 485 transceivers.

1. When I run the master and slave examples as they come in the idf everything all works fine.
2. I add the example file_serving to the master so that it works in parallel with the polling of the modbus slave and I override the alarm_state so that the polling is infinite.
3. So the glitches appear in some modbus requests. 

### Code to reproduce this issue

[https://github.com/JavierSVE/modbus_glitches.git](url)

## Debug Logs

[2022-07-13 modbus_master ESP_ERR (0x107).txt](https://github.com/espressif/esp-idf/files/9111056/2022-07-13.modbus_master.ESP_ERR.0x107.txt)
[2022-07-13 modbus_slave.txt](https://github.com/espressif/esp-idf/files/9111057/2022-07-13.modbus_slave.txt)

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

alisitsyn, 2022-07-18:
Hello @JavierSVE,

Thank you for your issue and your submitted artifacts. Could you use the [esp-modbus component](https://components.espressif.com/component/espressif/esp-modbus) instead? This component can work with esp-idf starting from version v4.1 and is recommended for new projects. 
In order to use this component you need: 

1. [idf-component-manager](https://pypi.org/project/idf-component-manager/) installed. 
2. Add the manifest file: [idf_component.yml](https://github.com/espressif/esp-idf/files/9130829/idf_component.yml.log) to the `modbus_master/main` `modbus_slave/main` component folder of your applications. 
3. Add the line below in the CMakeLists.txt file of your project to exclude the freemodbus component from your build and use esp-modbus instead:
```
cmake_minimum_required(VERSION 3.5)

# this line is added to exclude the existing freemodbus component from the build
set(EXCLUDE_COMPONENTS freemodbus)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(modbus_master)
```
Please try to use the `esp-modbus` first and I will help you to fix the issues.
Let me know if you have any questions.
Thank you.

JavierSVE, 2022-07-18:
Thank you alisitsyn for your quick response.
I don't understand why I have to change the modbus component. I am already working with the official component of the Espressif esp_idf framework in its version v4.4.1 (stable).
If I have not misunderstood, what you are proposing is to migrate this component to a previous version of the framework, the v4.1.

alisitsyn, 2022-07-18:
> If I have not misunderstood, what you are proposing is to migrate this component to a previous version of the framework, the v4.1.

The freemodbus component has been moved to a separate esp-modbus repository. This component can be used with any version of esp-idf from v4.1 up to latest. This `esp-modbus` is the latest version of the modbus component that is recommended for future support.

alisitsyn, 2022-07-21:
@JavierSVE ,

Could you try to start and check the [updated project](https://github.com/alisitsyn/modbus_support/tree/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii) on your side?

JavierSVE, 2022-07-22:
Hello @alisitsyn

I have already tried the [updated project](https://github.com/alisitsyn/modbus_support/tree/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii) on my side  and now another type of error identified as ESP_ERR_TIMEOUT appears.

Now, there are no more glitches in the frames sent by the master, but randomly there are delays on the falling edge of the master's RTS that cause the warning "MB_MASTER_SERIAL: Drop short packet 1 byte(s)" and the error "ESP_ERR_TIMEOUT". On these occasions the responses of the slave reach the master, but the master interprets them as wrong.

[image]

A Master-DE (RTS) of 1.3788ms can be observed instead of 756.06us which is normal and the next request is delayed by 169.2124ms instead of 8ms.

Now all the frames reach the slave well and the slave sends all the responses to the master, but some are not interpreted correctly.

alisitsyn, 2022-07-22:
Hello @JavierSVE ,

Please send me your kconfig file. 
Thanks.

JavierSVE, 2022-07-26:
Hello @alisitsyn,

Here you have my kconfig and my sdkconfig:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9187318/sdkconfig.txt)
[Kconfig.txt](https://github.com/espressif/esp-idf/files/9187342/Kconfig.txt)

But I have made the changes you propose and the problems continue.
This is the new sdkconfig:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9187390/sdkconfig.txt)

JavierSVE, 2022-07-27:
Sorry @alisitsyn,

I have already found why your [updated project](https://github.com/alisitsyn/modbus_support/tree/modbus_support/fix_glitches_esp_modbus_master_rtu_ascii) did not work for me.
In the directory components\espressif__esp-modbus there is a dependency file (idf_component.yml) which points to version 1.0.4 of the IDF Component Manager.
This caused that when I rebuilt the project the component was updated and crushed your modifications.

I have removed the dependency file and rebuilt your project and now it works. I can now run the file web server and the modbus RTU master in parallel.

Thank you very much.

alisitsyn, 2022-07-27:
Hello @JavierSVE,

Thank you for your update and sorry for the delay with answer. 
> In the directory components\espressif__esp-modbus there is a dependency file (idf_component.yml) 

Yes, the file name was changed in my project to temporary prevent processing of it.

The project still preliminary and has some known issues. The reason for posting this is to try the approach on your side because on my side it was working pretty reliable and did not cause timeout errors when update the web page.  However there is still issues related to UART buffer reading and they still need some investigation. I need time to complete the remaining work and will update you as soon as I can.

alisitsyn, 2024-01-24:
Unfortunately, the freemodbus component which is part of esp-idf v4.4 will not be updated. It is recommended to use the latest esp-modbus  component and exclude the obsolete one as described in the documentation. The corresponded issue has been fixed in that component.
The issue will be closed. feel free to reopen if you have some comments.