espressif/esp-idf #9356: BT SPP Baud rate issue

naragururaj, 2022-07-14:
Hi,
 For one of our projects we are using BT SPP for communication between ESP and windows. While performing the data speed analysis we noticed that speed is significantly compromised. As per this  _https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-faq/en/latest/software-framework/ble-bt.html_ a max of 1900Kbps can be achieved.  But at my end I see that it is just around 50-75Kbps(**speed of data going out of ESP alone**).  

Firmware sends out 960 bytes of data every 15ms which means around 66pkts(1000/15) should be sent out for every second Theoretically it comes up to 0.5Mbps and like I mentioned above we are just getting 75Kbps.
Attaching the sample code for your reference.
[example_spp_acceptor_demo.zip](https://github.com/espressif/esp-idf/files/9113924/example_spp_acceptor_demo.zip)

naragururaj, 2022-07-14:
Would like to close this issue as I was able to achieve the expected rate by modifying certain configs on the client side. Before that can you please explain the data flow between the BT SPP acceptor and BT SPP Initiator. Does either side expect ack after sending data?

yiyangc18, 2022-08-13:
Hi，
we encountered the same problem as you. Cloud you explain exactly what configuration you modified? Thant you!

> Would like to close this issue as I was able to achieve the expected rate by modifying certain configs on the client side. Before that can you please explain the data flow between the BT SPP acceptor and BT SPP Initiator. Does either side expect ack after sending data at the stack level?

boblane1, 2022-08-16:
Hi @naragururaj,

Thanks for reporting the issue. The SPP throughput can be affected by many reasons:

1. The default MTU size of classic Bluetooth SPP on ESP32 is 990 bytes, and higher throughput can be achieved in the case that data chunck size is close to the MTU size or multiple of MTU size. For example, sending 100 bytes data per second is much better than sending 10 bytes every 100 milliseconds.
2. SPP uses credit based flow control. If the receiver can't give the credit back to the sender as soon as possible, it can also slow down the throughput.
3. Unnecessary logs in the SPP callback functions.
4. The coexistence of BLE or Wi-Fi on ESP32.
5. Complex electromagnetic environment.

yiyangc18, 2022-08-17:
Thanks for reply. I think I have solved this problem by increasing the frequency of sending and the amount of data per packet.

1. BT SPP speed is not affected by baud rate, because RFCOMM, the configuration file under SPP, will try to transmit data packets as fast as possible. You can change Virtue BT COM in the Device Manager, but it's useless for transmission speed. Because we are used to the design of hardware UART, the best way to increase the data rate is to change the baud rate, but this is not necessary when using Bluetooth SPP.
2. The default MTU size is 990 bytes, but you can send a larger MTU to get more bandwidth. When increasing the packet size is useless, you can increase the sending frequency. Don't config the send interval less than 10 ms, because the MAX ACK Fre of BT is about 100 Hz. Higher fre may lead to packet loss. In my Windows PC, the data speed can achieve 1200 kbps-150 kB/s with 20 send interval and 3 kB Per packet. 

> Hi @naragururaj,
>
> Thanks for reporting the issue. The SPP throughput can be affected by many reasons:
>
> 1. The default MTU size of classic Bluetooth SPP on ESP32 is 990 bytes, and higher throughput [...]

naragururaj, 2022-08-22:
Satisfied with the response. Please close the issue.

closed 2022-08-22 by naragururaj