espressif/esp-idf #9358: ESP32-C3 LEDC fade interrupt enabled when setting `intr_type` to `LEDC_INTR_DISABLE`

Kampi, 2022-07-15:
## Environment

- Development Kit: `ESP32-C3-32S`
- Module or chip used:  ESP32-C3
- IDF version (run ``git describe --tags`` to find it): `4.4.1`
- Build System:  `CMake`
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): `8.4.0+2021r2-patch3`
- Operating System: `Windows`
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: `VS Code` / `PlatformIo`
- Power Supply: `USB`

## Problem Description

I use the LEDC of the chip to handle an RGB LED. I also want to use the `LEDC_INTR_FADE_END` callbacks to detect the end of a fading. I have disabled the interrupts by setting `.intr_type`to `LEDC_INTR_DISABLE` in the channel configuration, but the callback gets triggered all the time when a fading is complete.

I´m not sure if this is a bug or if I use the driver wrong, but I understand the documentation in that way that I have to set `intr_type` to `LEDC_INTR_FADE_END` to enable the interrupt. 

### Expected Behavior

The callback is not triggered and the LED does not fade.

### Actual Behavior

The LED is fading red up and down all the time.

### Code to reproduce this issue

```cpp

static const ledc_timer_config_t _Timer = {
    .speed_mode = LEDC_LOW_SPEED_MODE,
    .duty_resolution = LEDC_TIMER_8_BIT,
    .timer_num = LEDC_TIMER_0,
    .freq_hz = LED_PWM_FREQUENCY,
    .clk_cfg = LEDC_USE_RTC8M_CLK
};

static const ledc_channel_config_t _LED_Red_Channel = {
    .gpio_num = LED_RED,
    .speed_mode = _Timer.speed_mode,
    .channel = LEDC_CHANNEL_0,
    .intr_type = LEDC_INTR_DISABLE,
    .timer_sel = _Timer.timer_num,
    .duty = 0,
    .hpoint = 0,
    .flags = { 0 }
};

static const ledc_channel_config_t _LED_Green_Channel = {
    .gpio_num = LED_GREEN,
    .speed_mode = _Timer.speed_mode,
    .channel = LEDC_CHANNEL_1,
    .intr_type = LEDC_INTR_DISABLE,
    .timer_sel = _Timer.timer_num,
    .duty = 0,
    .hpoint = 0,
    .flags = { 0 }
};

static const ledc_channel_config_t _LED_Blue_Channel = {
    .gpio_num = LED_BLUE,
    .speed_mode = _Timer.speed_mode,
    .channel = LEDC_CHANNEL_2,
    .intr_type = LEDC_INTR_DISABLE,
    .timer_sel = _Timer.timer_num,
    .duty = 0,
    .hpoint = 0,
    .flags = { 0 }
};

static bool _LED_FadeDone;
static const uint32_t _Limit = ((1UL << _Timer.duty_resolution) - 1);

static bool _on_LED_FadeComplete_Handler(const ledc_cb_param_t* p_Param, void* p_Args)
{
    portBASE_TYPE taskAwoken = pdFALSE;

    if(p_Param->event == LEDC_FADE_END_EVT)
    {
        _LED_FadeDone = true;
    }

    return (taskAwoken == pdTRUE);
}

void LED_Init(void)
{
    ledc_cbs_t Callbacks;

    _LED_FadeDone = true;

    Callbacks = {
        .fade_cb = _on_LED_FadeComplete_Handler
    };

    ledc_timer_config(&_Timer);
    ledc_channel_config(&_LED_Red_Channel);
    ledc_channel_config(&_LED_Green_Channel);
    ledc_channel_config(&_LED_Blue_Channel);
    ledc_fade_func_install(0);

    ledc_cb_register(_LED_Red_Channel.speed_mode, _LED_Red_Channel.channel, &Callbacks, NULL);
    ledc_cb_register(_LED_Green_Channel.speed_mode, _LED_Green_Channel.channel, &Callbacks, NULL);
    ledc_cb_register(_LED_Blue_Channel.speed_mode, _LED_Blue_Channel.channel, &Callbacks, NULL);
}

void LED_FadeUp(bool Red, bool Green, bool Blue)
{
    if(_LED_FadeDone == false)
    {
        return;
    }

    _LED_FadeDone = false;

    ledc_set_fade_with_time(_LED_Red_Channel.speed_mode, _LED_Red_Channel.channel, (Red == true) ? _Limit : 0, 1000);
    ledc_set_fade_with_time(_LED_Green_Channel.speed_mode, _LED_Green_Channel.channel, (Green == true) ? _Limit : 0, 1000);
    ledc_set_fade_with_time(_LED_Blue_Channel.speed_mode, _LED_Blue_Channel.channel, (Blue == true) ? _Limit : 0, 1000);

    ledc_fade_start(_LED_Red_Channel.speed_mode, _LED_Red_Channel.channel, LEDC_FADE_NO_WAIT);
    ledc_fade_start(_LED_Green_Channel.speed_mode, _LED_Green_Channel.channel, LEDC_FADE_NO_WAIT);
    ledc_fade_start(_LED_Blue_Channel.speed_mode, _LED_Blue_Channel.channel, LEDC_FADE_NO_WAIT);
}

void LED_FadeDown(bool Red, bool Green, bool Blue)
{
    if(_LED_FadeDone == false)
    {
        return;
    }

    _LED_FadeDone = false;

    ledc_set_fade_with_time(_LED_Red_Channel.speed_mode, _LED_Red_Channel.channel, 0, 1000);
    ledc_set_fade_with_time(_LED_Green_Channel.speed_mode, _LED_Green_Channel.channel, 0, 1000);
    ledc_set_fade_with_time(_LED_Blue_Channel.speed_mode, _LED_Blue_Channel.channel, 0, 1000);

    ledc_fade_start(_LED_Red_Channel.speed_mode, _LED_Red_Channel.channel, LEDC_FADE_NO_WAIT);
    ledc_fade_start(_LED_Green_Channel.speed_mode, _LED_Green_Channel.channel, LEDC_FADE_NO_WAIT);
    ledc_fade_start(_LED_Blue_Channel.speed_mode, _LED_Blue_Channel.channel, LEDC_FADE_NO_WAIT);
}

bool LED_isFadeDone(void)
{
    return _LED_FadeDone;
}

void LED_FadeUpAndDown(bool Red, bool Green, bool Blue)
{
    LED_FadeUp(Red, Green, Blue);
    while(LED_isFadeDone() == false)
    {
        vTaskDelay(20 / portTICK_PERIOD_MS);
    }
    LED_FadeDown(Red, Green, Blue);
    while(LED_isFadeDone() == false)
    {
        vTaskDelay(20 / portTICK_PERIOD_MS);
    }
}
```

```cpp
while(1)
{
    LED_FadeUpAndDown(true, false, false);
}
```

songruo, 2022-07-18:
Hi @Kampi,

This is because the `LEDC_INTR_FADE_END` interrupt will always get enabled when you call `ledc_fade_start`.  Can you explain your usage scenario and why you don't want LEDC to fade but calling `ledc_fade_start`?

Kampi, 2022-07-18:
Hi @songruo,
I don´t want to implement this scenario. I´m just wondering why the interrupt got triggered even if I haven´t enabled it.

But your answer is fine :)
Thank you!

closed 2022-07-18 by Kampi