espressif/esp-idf #9359: vTaskGetRunTimeStats error

zongshuwei, 2022-07-15:
Hi , there is a problem when i called vTaskGetRunTimeStats in idf 4.4.1.
some codes are as follows,
			char *pbuffer = (char *)calloc(1, 2048);
			printf("--------------- heap:%u ---------------------\r\n", esp_get_free_heap_size());
			vTaskGetRunTimeStats(pbuffer);
			printf("%s", pbuffer);
			printf("----------------------------------------------\r\n");
			free(pbuffer);

it's about 2 hours，it prints

                      [17-03-22.251]--------------- heap:1359183 ---------------------
                      [17-03-22.251]state machin    146607611               24446%
                      [17-03-22.251]IDLE            3511941474              585616%
                      [17-03-22.261]IDLE            3852805379              642455%
                      [17-03-22.271]tcp client      62321713                10392%
                      [17-03-22.271]note task       15467737                2579%
                      [17-03-22.271]button scan     3087959         514%
                      [17-03-22.281]ble rev task    4765265         794%
                      [17-03-22.281]wifi rev task   228317          38%
                      [17-03-22.281]tiT             13231518                2206%
                      [17-03-22.291]ext io scan     50130642                8359%
                      [17-03-22.291]wifi_work       14726363                2455%
                      [17-03-22.291]wifi send task  786136          131%
                      [17-03-22.291]gui             3387910         564%
                      [17-03-22.301]ble send task   415719          69%
                      [17-03-22.301]wifi_manager_ta 678802          113%
                      [17-03-22.301]bat task        12957218                2160%
                      [17-03-22.301]ble host manage 1449074         241%
                      [17-03-22.311]ipc1            54175           9%
                      [17-03-22.311]ipc0            29800           4%
                      [17-03-22.311]sys_evt         18359           3%
                      [17-03-22.311]wifi            92175911                15370%
                      [17-03-22.311]esp_timer       11404173                1901%
                      [17-03-22.321]BTU_TASK        77146850                12864%
                      [17-03-22.321]BTC_TASK        17092908                2850%
                      [17-03-22.321]hciT            158677549               26459%
                      [17-03-22.332]btController    429732883               71657%
                      [17-03-22.332]udp_server      43411           7%
                      [17-03-22.332]Tmr Svc         17              <1%
                      [17-03-22.342]----------------------------------------------
   So how do I get the correct adaptation of threads? 
   Thanks.

igrr, 2022-07-15:
FreeRTOS runtime counters feature unfortunately doesn't handle overflow of the counters, which leads to this issue. The issue is not specific to ESP-IDF and affects FreeRTOS in general. You can read https://www.freertos.org/FreeRTOS_Support_Forum_Archive/May_2019/freertos_How_to_reset_vTaskGetRunTimeStats_425f1134aaj.html for some ideas about working around it.

Dazza0, 2023-03-01:
@zongshuwei as mentioned above, the [FreeRTOS run time stats](https://www.freertos.org/rtos-run-time-stats.html) feature measures CPU usage of each task over the life time of the scheduler, thus it does not support counter overflows. Some alternatives you can consider are:

- Using a slower run time stats counter to delay the counter overflow (see the [run time stats clock configuration](https://docs.espressif.com/projects/esp-idf/en/v5.0.1/esp32/api-reference/kconfig.html#config-freertos-run-time-stats-clk))
- We have an example of how in ESP-IDF on how to use the FreeRTOS run time stats functions to measure CPU usage across a given period of time (instead of the entire scheduler's life time). Please see the [real time stats example](https://github.com/espressif/esp-idf/tree/v5.0.1/examples/system/freertos/real_time_stats) in ESP-IDF.

closed 2023-03-01 by Dazza0