espressif/esp-idf #9362: TDM4 Slave Incorrect alignment between Clocks ( BCK + WS ) and Data

CesareVin, 2022-07-15:
## Environment

- Development Kit: ESP32-S3-DevKitC-1
- Kit version : v1
- IDF version : v5.0-dev-4001-g495d35949d
- Build System:   idf.py
- Compiler version : (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     Linux and macOS
- Using an IDE?: NO
- Power Supply:         USB

## Problem Description

If I2S is configured as TDM4 Slave , BCK and WS is not correctly aligned DATA.

### Expected Behavior

I2S slave should have data aligned to slave clock.

### Actual Behavior

- 1 esp32s3 is Master Clock.
- 1 esp32s3 is Slave Clock.

Both ESP send the same pattern that identify channels.

- CH1 : 0xAAAAAAAA
- CH2 : 0xBBBBBBBB
- CH3 : 0xCCCCCCCC
- CH4 : 0xDDDDDDDD

Master and Slave esp are both configured in the same standard ( tried all configuration, assume I2S_TDM_PHILIP ).
Slave data is formally correct but not correctly aligned to the clock.
At 48 Khz the misalignment is less than half BCK period and data seem fine.
[image]

[image]

At 96 Khz the misalignment is >= then half BCK period and data seem corrupted.
[image]

[image]

Screen Shot Legend : 
- TDM_DATA_IN : master output
- TDM_OUT : slave output
- TDM_BCK : master Tdm Bit Clock
- SLAVE_WRITE_TIME : master TDM Word Clock ( sorry wrong name on analizer channel ) 

### Steps to reproduce

1.  Download the attached project
2. Connect master and slave following pinout described in project README
3. Switch between 48 KHz and 96 KHz with `#define EXAMPLE_SAMPLE_RATE` in the project. 
4. Flash 2 esp32s3 devkitC

### Code to reproduce this issue

[tdm_test_idf5.zip](https://github.com/espressif/esp-idf/files/9122692/tdm_test_idf5.zip)

@L-KAYA I mention you because i see your contribution on https://github.com/espressif/esp-idf/issues/8339 and you seem the most expert of esp32s3 I2S driver.

L-KAYA, 2022-07-18:
Sorry for the late reply, the I2S slave mode on ESP32-C3/S3 is not same as ESP32, there is indeed some latency on the TX data signal, the latency is related to the mclk period, you can try to increase the 'mclk' frequency by using a bigger `mclk_multiple`. Hope it can help~

CesareVin, 2022-07-19:
@L-KAYA thanks for reply.
Trying to change `I2S_MCLK_MULTIPLE_256` to `I2S_MCLK_MULTIPLE_384` on the slave don't change anything, also the debug log is the same : 
`i2s_tdm: Clock division info: [sclk] 160000000 Hz [mdiv] 1 [mclk] 98304000 Hz [bdiv] 8 [bclk] 12288000 Hz`

Another very crazy behavior of the TDM4 slave is that the peripheral seem not use the frame clock to understand channels order.
Sometimes it starts with CH1/CH2 - CH3/CH4 (correct behavior ) ,  sometimes CH3/CH4 - CH1/CH2.
Am I doing something wrong with the configuration ?
Or is it a peripheral limit in some way ?
Thanks in advance for the support.

CesareVin, 2022-08-25:
discard the previous message, i miss understand you suggestion.
As you suggest, using faster clock ( 240 MHz ) in slave mode with the smallest integer divider ( 2 ) reduce the BCK delay.

closed 2022-08-25 by CesareVin

L-KAYA, 2025-02-24:
@CesareVin Ah, my mistake, for the slave role, you need to increase `bclk_div` to get a larger `mclk`, not `mclk_multiple`.

Post here if anyone else met the same issue~