espressif/esp-idf #9364: [ESP32-S3] USB CDC Console Example hangs & freezes shortly after boot

chipweinberger, 2022-07-15:
**ESP32-S3 Dev Kit C**
**ESP-IDF v4.4.1**

**Code:** https://github.com/espressif/esp-idf/tree/v4.4.1/examples/system/console_usb

**Problem:**
After flashing the CDC Console Example to my ESP32-S3, I then plug the ESP32-S3's USB Native port into my computer and call `idf.py monitor`. The device reboots, logs a little bit, then promptly hangs, then reboots, repeat forever...

**Logs:**
```
charlesweinberger@Chips-Macbook-Pro-2 console_usb % idf.py set-target esp32s3  
...
-- Build files have been written to: /Volumes/User/MBP-Google-Drive/jamcorder/firmware/my-examples/console_usb/build

charlesweinberger@Chips-Macbook-Pro-2 console_usb % idf.py build                          
...
Project build complete. To flash, run this command:

charlesweinberger@Chips-Macbook-Pro-2 console_usb % idf.py -p /dev/cu.SLAB_USBtoUART flash
Executing action: flash
..
Hard resetting via RTS pin...
Done

//
// NOW I PHYSICALLY DISCONNECT THE USB CABLE FROM THE ESP32S3 UART-USB 
// AND MOVE IT TO THE USB-NATIVE PORT
//

charlesweinberger@Chips-Macbook-Pro-2 console_usb % idf.py monitor                        
Executing action: monitor
Serial port /dev/cu.usbmodem2101
Connecting...
Detecting chip type... ESP32-S3
Running idf_monitor in directory /Volumes/User/MBP-Google-Drive/jamcorder/firmware/my-examples/console_usb
Executing "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf4.4_py3.9_env/bin/python /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem2101 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /Volumes/User/MBP-Google-Drive/jamcorder/firmware/my-examples/console_usb/build/console_usb.elf -m '/Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/python_env/idf4.4_py3.9_env/bin/python' '/Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.usbmodem2101 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048dfe
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd01c8,len:0x1158
load:0x403b6000,len:0xa4c
load:0x403ba000,len:0x2e10
entry 0x403b61f0
I (0) cpu_start: App cpu up.
I (102) cpu_start: Pro cpu start user code
I (102) cpu_start: cpu freq: 160000000
I (102) cpu_start: Application information:
I (102) cpu_start: Project name:     console_usb
I (102) cpu_start: App version:      1
I (103) cpu_start: Compile time:     Jul 15 2022 13:28:53
I (103) cpu_start: ELF file SHA256:  cb8077a39e36173f...
I (103) cpu_start: ESP-IDF:          v4.4.1-241-g16b7ed8697-dirty
I (104) heap_init: Initializing. RAM available for dynamic allocation:
I (104) heap_init: At 3FC95AB0 len 0004A550 (297 KiB): D/IRAM
I (104) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (105) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (105) heap_init: At 600FE028 len 00001FD8 (7 KiB): RTCRAM // THE MONITOR JUST HANGS HERE FOR AWHILE
read failed: [Errno 6] Device not configured
Waiting for the device to reconnect
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40052ce3
```

igrr, 2022-07-16:
Hi @chipweinberger, the usb_console example is only supported on ESP32-S2, not on ESP32-S3. Please check the "supported targets" table at the top of the readme.

USB CDC console via the USB_OTG peripheral is not implemented on ESP32-S3 yet. There are a couple open issues about this: https://github.com/espressif/esp-idf/issues/8879, https://github.com/espressif/esp-idf/issues/8738.

chipweinberger, 2022-07-16:
Thank you for your response Ivan.

That makes more sense. I saw that it was "S2 Only" but thought it was an out-of-date configuration. 

Happily waiting for it =)

closed 2023-04-18 by espressif-bot