espressif/esp-idf #9366: [ESP32-S3] How to detect if USB CDC Console is connected to a USB Host? (USB OTG Peripheral)

chipweinberger, 2022-07-15:
**ESP32-S3**

I'd like to know when a USB Host computer is connected to the ESP32-S3's USB CDC Console (which uses the USB OTG Peripheral).

I want to alter the behavior of my Usb application code if a CDC Host is connected within the first 3 seconds of boot.

Specifically:
- (A) Under normal operation, my ESP32-S3 uses Usb Host to communicate with external devices
- (B) However, for the first 3 seconds after boot I want the ESP32-S3 to listen in USB CDC Console mode, and if a computer connects I skip step (A) and remain in CDC Console mode

igrr, 2022-07-16:
In case your device isn't connected to the host PC (and your program initializes USB host itself), where do you need the console output (logs) to go? Do you still want to be able to access them over the UART?

If you plan to retarget the console output to UART when going into the USB host mode, it might actually be easier to start with UART console in the first place, and only initialize the USB CDC console if the device is connected to the host PC. If the host is not connected, then initialize USB host and carry on with the UART console.

chipweinberger, 2022-07-16:
Thanks for the suggestions =)

> In case your device isn't connected to the host PC (and your program initializes USB host itself), where do you need the console output (logs) to go? Do you still want to be able to access them over the UART?

Ideally yes. Or possibly USB Serial/JTAG. Is there any example code for "retargeting" log output?

>  might actually be easier to start with UART console in the first place, and only initialize the USB CDC console if the device is connected to the host PC

Can you elaborate? AFAICT:

Boot w/ Uart logs Advantages:
- more reliable for early logs & panic handler
- no need to "switch" log outputs when turning on USB Host mode, so an attached uart monitor will have access to all logs

Boot w/ CDC logs Advantages:
- my device has an external USB port, so more logs will be available at boot time without opening the device

I think you are right that booting w/ uart advantages outweighs the cons.

chipweinberger, 2022-07-16:
Also, are there any examples of external USB Transceivers working with ESP32-S3? My device has 2 external usb ports, and I want add a transceiver for USB Serial/JTAG access on the 2nd port but I've not been able to get it to work: https://github.com/espressif/esp-idf/issues/6408

suda-morris, 2022-07-16:
> Also, are there any examples of external USB Transceivers working with ESP32-S3? My device has 2 external usb ports, and I want add a transceiver for USB Serial/JTAG access on the 2nd port but I've not been able to get it to work: #6408

IDF doesn't provide such example yet. But I think @Lsitar has made great progress in working with external USB PHY+ ESP32S3 , after this PR got merged (https://github.com/espressif/esp-idf/pull/9302)

chipweinberger, 2022-07-16:
Thanks @suda-morris. Do you know what chip he is using? and schematic? I'm pulling my hair out trying to get this to work. Would love to find a "known working" implementation.

Lsitar, 2022-07-21:
Hello @chipweinberger . Sorry for the late reply, I was away for a few days.
I'm using built-in PHY for debugging, and the external PHY (TUSB1106PWR) is connected same as You mentioned in #6408.
[image]
In init struct, .external_phy = true must be set.
```
	tinyusb_config_t tusb_cfg =
	{
		.descriptor = &usbDescriptor,
		.string_descriptor = NULL,		//Uses default string specified in Menuconfig
		.external_phy = true,
	};

	if ((ret = tinyusb_driver_install(&tusb_cfg)) != ESP_OK)
		return ret;
```

Then I based on some example to start CDC mode:
```
	tinyusb_config_cdcacm_t amc_cfg =
	{
		.usb_dev = TINYUSB_USBDEV_0,
		.cdc_port = TINYUSB_CDC_ACM_0,
		.rx_unread_buf_sz = 64,
		.callback_rx = tinyusb_cdc_rx_callback,
		.callback_rx_wanted_char = NULL,
		.callback_line_state_changed = tinyusb_cdc_line_state_changed_callback,
		.callback_line_coding_changed = NULL
	};

	tusb_cdc_acm_init(&amc_cfg);
```
After PR #9302 and #9264 it works ok. Sometines host PC has problem to read descriptor, but it may be due to poor PCB layout, I did't debugged it yet. 
I tested it as echo: tinyusb_cdc_rx_callback() received data from PC host and immediately re-send with tinyusb_cdcacm_write_queue() and tinyusb_cdcacm_write_flush().
My device will always be CDC, I do not activated OTG.

chipweinberger, 2022-07-22:
Many thanks for taking the time to respond. That is helpful.

chipweinberger, 2022-11-09:
To answer my own question, there are 2 ways to detect if a USB Console is connected. 

For the USB Serial/JTAG driver, the easiest way to send a terminela probe request:

`fprintf(stdout, "\x1b[5n");`

and then poll this function for a couple seconds waiting for a response:

`static inline int usb_serial_jtag_ll_rxfifo_data_available(void)`

and for CDC you could use:

```
esp_err_t esp_usb_console_init(void);
ssize_t esp_usb_console_available_for_read(void);
```

There is also a PR to add a flag to the USB Serial/JTAG Driver: https://github.com/espressif/esp-idf/pull/10097

closed 2022-11-09 by chipweinberger

dongrigorio, 2023-01-19:
For esp32-c3: 

bool is_usb_connected(void){
    //Control OUT endpoint status information
    uint32_t *qq = USB_SERIAL_JTAG_OUT_EP0_ST_REG;
    return (bool)*qq;
}

chipweinberger, 2023-01-19:
@dongrigorio, how did you find that? I don't see it used anywhere.

dongrigorio, 2023-01-19:
I came up with this trick because I couldn't find anything on the internet :) I checked, it works.

ozarchie, 2023-01-19:
Maybe ?
When USB_SERIAL_JTAG_SERIAL_OUT_RECV_PKT_INT is detected, there are
USB_SERIAL_JTAG_OUT_EP0_WR_ADDR-2 bytes data in OUT EP0.

chipweinberger, 2023-04-18:
`bool usb_serial_jtag_is_connected(void)` was added to master on Februrary 26, 2023. I think it's landing in v5.1. https://github.com/espressif/esp-idf/commit/1a66459b4454f0b45b1910b2bf260c77aef119f4 

I'm going to close this issue. 

Note: The original issue was about USB OTG Peripheral, but that was before I understood I actually wanted this for USB-SERIAL-JTAG peripheral.